nordicsemi/Android-DFU-Library #96: startForeground fail after update to Android 8.1

kinber95, 2018-02-22:
My App crash after I update my Nexus 5x to Android 8.1, the reasons is that you need for a Foreground-Service a own NotificationChannel. 

Can I do it on my own, or do you have to update the lib ? 

Cheers

philips77, 2018-02-22:
You may create your own, but it needs to have the id = [NOTIFICATION_CHANNEL_DFU](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L99). You may use this method:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L588
but also may create the channel on your own.

kinber95, 2018-02-22:
Thanks for the fast answer.

Do I have to create the channel in my NotificationActivity?

philips77, 2018-02-22:
We usually do it in the Application class in onCreate, so it's done once the app starts. But I have no really clue where it should be done. Doc says it just needs to be done :)

kinber95, 2018-02-22:
Okay I read the docs again and found at the bottom the method : `DfuServiceInitiator.createDfuNotificationChannel(context);
`

I add it to my firmware update code, know is all good :)

Thanks for the fast help 👍

philips77, 2018-02-22:
Sure:)

closed 2018-02-22 by philips77