nordicsemi/Android-DFU-Library #97: NullPointerException in ArchiveInputStream.read

skrabacz-michal, 2018-03-07:
`Fatal Exception: java.lang.NullPointerException: Attempt to get length of null array
       at no.nordicsemi.android.dfu.internal.ArchiveInputStream.read(ArchiveInputStream.java:338)
       at no.nordicsemi.android.dfu.BaseCustomDfuImpl.uploadFirmwareImage(BaseCustomDfuImpl.java:351)
       at no.nordicsemi.android.dfu.SecureDfuImpl.sendFirmware(SecureDfuImpl.java:523)
       at no.nordicsemi.android.dfu.SecureDfuImpl.performDfu(SecureDfuImpl.java:220)
       at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1173)
       at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:67)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:154)
       at android.os.HandlerThread.run(HandlerThread.java:61)`

philips77, 2018-03-07:
Could you post more information what were you doing?
The logs from LogCat? Does it happen 100% of times? Does it happen with any ZIP or just your ZIP file, etc.
Thanks!

skrabacz-michal, 2018-03-07:
The exception caught by one of our error monitors so I don't have any additional logs but it is only one so far. It happened during updating the firmware on one of Suunto Movesense devices (we have custom firmware package).

jasonim, 2018-07-25:
I have the same problem, killing the app in dfu background

biserakav, 2018-09-13:
Same issue has happened with our application when dfu process was interrupted by killing the app. I have managed to reproduce the issue with nrfConnect app too (see logs below). 
I am testing on Google Pixel device, Android version 8.0.0. 
In the app we are using version 1.6.0 of the library.
 
```09-13 15:32:41.134 14338-14393/? E/bt_btif: bta_gattc_mark_bg_conn unable to find the bg connection mask for: e4:70:5a:43:41:6e
09-13 15:32:41.135 14896-14907/? D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=E4:70:5A:43:41:6E
09-13 15:32:41.142 483-483/? D/SurfaceFlinger: duplicate layer name: changing com.google.android.apps.nexuslauncher/com.google.android.apps.nexuslauncher.NexusLauncherActivity to com.google.android.apps.nexuslauncher/com.google.android.apps.nexuslauncher.NexusLauncherActivity#1
09-13 15:32:41.143 28869-28869/? W/SearchServiceClient: start new client, handoverId [1] client id [126968741489434558]
    com.google.android.apps.gsa.shared.util.common.f: SearchServiceClient: start new client, handoverId [1] client id [126968741489434558]
        at com.google.android.apps.gsa.shared.util.common.e.a(SourceFile:43)
        at com.google.android.apps.gsa.shared.util.common.e.f(SourceFile:6)
        at com.google.android.apps.gsa.search.shared.service.SearchServiceClient.a(SourceFile:116)
        at com.google.android.apps.gsa.search.shared.service.SearchServiceClient.startWithNewSession(SourceFile:103)
        at com.google.android.apps.gsa.search.shared.overlay.a.b.aTX(SourceFile:486)
        at com.google.android.apps.gsa.search.shared.overlay.a.b.onResume(SourceFile:531)
        at com.google.android.apps.gsa.nowoverlayservice.u.onResume(SourceFile:295)
        at com.google.android.libraries.gsa.e.b.g.PU(SourceFile:29)
        at com.google.android.libraries.gsa.e.b.q.handleMessage(SourceFile:123)
        at com.google.android.libraries.gsa.e.b.r.handleMessage(SourceFile:9)
        at android.os.Handler.dispatchMessage(Handler.java:101)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6541)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
09-13 15:32:41.143 28869-28869/? W/SessionLifecycleManager: Handover failed. Creating new session controller.
09-13 15:32:41.151 28869-21930/? I/RunnerModule: Event Bus thread is disabled
09-13 15:32:41.167 14896-14907/? D/BluetoothGatt: close()
    unregisterApp() - mClientIf=7
09-13 15:32:41.186 28869-20917/? W/LocationOracle: No location history returned by ContextManager
    
    --------- beginning of crash
09-13 15:32:41.222 14896-22064/? E/AndroidRuntime: FATAL EXCEPTION: IntentService[DfuBaseService]
    Process: no.nordicsemi.android.mcp, PID: 14896
    java.lang.NullPointerException: Attempt to get length of null array
        at no.nordicsemi.android.dfu.internal.ArchiveInputStream.read(Unknown Source:338)
        at no.nordicsemi.android.dfu.BaseCustomDfuImpl.uploadFirmwareImage(Unknown Source:351)
        at no.nordicsemi.android.dfu.SecureDfuImpl.sendFirmware(Unknown Source:524)
        at no.nordicsemi.android.dfu.SecureDfuImpl.performDfu(Unknown Source:221)
        at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(Unknown Source:1186)
        at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:68)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
09-13 15:32:41.228 905-936/? I/ActivityManager: Showing crash dialog for package no.nordicsemi.android.mcp u0```

PR #126 (2018-09-26): fix upload not aborted on service destroy
fixes #97 , #108  
When the app is killed by user and the service with it, it closes the file stream, but doesn't abort the DFU process, which then tries to read next chunk to send to the device and crashes.

Not sure if this is the best way to fix it, but at least it doesn't crash anymore.
closed 2018-09-26 by philips77