nordicsemi/Android-DFU-Library #99: Crash on DfuBaseService class (line #1578)

datainvent, 2018-03-13:
I updated the sdk from 1.4.2 to 1.6.1 and now whenever i start dfu process my app crashes with the error "No Method found for NotificationCompact.Build in base apk" 
What could be the cause?

philips77, 2018-03-13:
Looks like a proguard issue.

datainvent, 2018-03-13:
I am using gradle @philips77

philips77, 2018-03-13:
Did you enable proguard in your gradle file?

datainvent, 2018-03-13:
No. Does it have anything to do with the compiled sdk version? I am using 27 btw. @philips77

philips77, 2018-03-13:
Did you also switch support lib to 27.0.2 or 27.1.0 in dependencies?

philips77, 2018-03-13:
https://stackoverflow.com/questions/41242832/fatal-exception-java-lang-nosuchmethoderror-no-interface-method-build-no-int

philips77, 2018-03-14:
Hi, did you manage to solve the issue?

closed 2018-10-24 by philips77