nordicsemi/Android-DFU-Library #101: Problem inferring file type from file path in newer Android versions

nfdz, 2018-03-26:
Hi all,

I got problems updating only bootloader zip files whereas there is no problem with application zip files. My error was related with Manifest file validation _"Manifest file must specify at least one file."_. 

I have dived deep in library code and I have found this logic:

``
int fileType = intent.getIntExtra(EXTRA_FILE_TYPE, TYPE_AUTO);
if (filePath != null && fileType == TYPE_AUTO)
	fileType = filePath.toLowerCase(Locale.US).endsWith("zip") ? TYPE_AUTO : TYPE_APPLICATION;
``
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L917

This logic is wrong because in last versions of Android when you does not require storege permission and does not manage file picking by yourself you got and URI and path like this: 
_"/document/1906"_

I does not understand why there is this line of code in the initialization of the DFU service but I think that is not the proper way nowadays.

Current workaround would be manage file picking by yourself or does not use file type auto and manage the scope manually.

Regards,
Noé

philips77, 2018-03-26:
Hi, 
But if EXTRA_FILE_TYPE is set to auto, and your filePath is null (you're saying you're using fileUri) then if will not go into this if.
This code is for legacy reasons as some apps may not set fileType (there were times without the dfuServiceInitiator).

nfdz, 2018-03-27:
Ok, you are totally right. However, I don't feel intuitive at all and someone could loose time looking for strange errors. Maybe it is time to deprected more methods in the initializator, write a big warning/disclaimer or annotate with @Nullable at least.
```java
/**
 * Sets the URI or path of the ZIP file. If the URI and path are not null the URI will be used.
 * @param uri the URI of the file
 * @param path the path of the file
 * @return the builder
 */
public DfuServiceInitiator setZip(final Uri uri, final String path) {
	return init(uri, path, 0, DfuBaseService.TYPE_AUTO, DfuBaseService.MIME_TYPE_ZIP);
}
```
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L392
Now I know that I should use only the one with URI parameter.
Thanks

closed 2018-10-24 by philips77