nordicsemi/Android-DFU-Library #102: The Dfu firmware upgrade implements the callback method onProgressChanged (), but an exception is thrown when the percent=99%.

jingmo0319, 2018-04-18:
The Dfu firmware upgrade implements the callback method onProgressChanged (), but an exception is thrown when the percent=99%.Then the onError( ) method was executed, error=4096 errorType= message=DFU DEVICE DISCONNECTED.
Abnormal logs are as follows:
04-18 17:02:19.730 15417-15460/? E/bt_att: HAL bt_gatt_callbacks
04-18 17:02:20.173 1038-1060/? E/MFI-Conn:  (     mfia_task_bootstrap,  587) mfia_task_bootstrap(): retrying to connect to mal-mfi error 13!! ====
04-18 17:02:21.173 1038-1060/? E/MFI-Conn:  (     mfia_task_bootstrap,  587) mfia_task_bootstrap(): retrying to connect to mal-mfi error 13!! ====
04-18 17:02:22.174 1038-1060/? E/MFI-Conn:  (     mfia_task_bootstrap,  587) mfia_task_bootstrap(): retrying to connect to mal-mfi error 13!! ====
04-18 17:02:22.472 1213-1642/? E/WifiStateMachine: calculateWifiScore not update currentConfiguration visibility
04-18 17:02:23.175 1038-1060/? E/MFI-Conn:  (     mfia_task_bootstrap,  587) mfia_task_bootstrap(): retrying to connect to mal-mfi error 13!! ====
04-18 17:02:24.175 1038-1060/? E/MFI-Conn:  (     mfia_task_bootstrap,  587) mfia_task_bootstrap(): retrying to connect to mal-mfi error 13!! ====
04-18 17:02:24.438 15417-15448/? E/BluetoothRemoteDevices: state12newState1
04-18 17:02:24.438 10973-11127/com.risoo.app E/DfuImpl: Disconnected while sending data
04-18 17:02:24.438 10973-11127/com.risoo.app E/DfuBaseService: Uploading Firmware Image failed: device disconnected

philips77, 2018-04-18:
Please, write more details:
1. What phone(s) are you using, what Android version,
2. What SDK is your fw based on,
3. Does it happen every time or how often,
4. Paste more logs. Now only the last 2 lines are relevant. Filter by "dfu" in the logcat.

philips77, 2018-10-24:
Hi, did you manage to solve the problem?

Corical, 2019-06-07:
Goodness gracious, for the sake of anyone else finding this, can you at least post a resolution, even if it wasn't resolved.