nordicsemi/Android-DFU-Library #103: DfuBaseService: Connection state change error: 8 newState: 0

jingmo0319, 2018-04-23:
When I upgraded the firmware to 100 percent, I thought the upgrade was successful, but it threw an exception. Who knows what caused it?My diary is as follows, please help me!
04-23 15:48:00.117 4995-5065/com.risoo.app E/DfuBaseService: Connection state change error: 8 newState: 0
04-23 15:48:00.118 4995-5158/com.risoo.app E/DfuImpl: Disconnected while sending data
04-23 15:48:00.118 4995-5158/com.risoo.app E/DfuBaseService: Uploading Firmware Image failed: device disconnected
04-23 15:48:00.125 4995-4995/com.risoo.app E/bm: onDeviceDisconnected
04-23 15:48:00.127 4995-4995/com.risoo.app E/bm: onError=4096     errorType=0   message=DFU DEVICE DISCONNECTED

philips77, 2018-04-23:
Could you paste a bit more log? What happens before that.

jingmo0319, 2018-04-23:
04-23 16:59:30.666 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onDfuProcessStarting-deviceAddress=CE:99:42:70:BA:CF
04-23 16:59:32.936 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onEnablingDfuMode-deviceAddress=CE:99:42:70:BA:CF
04-23 16:59:35.274 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onDeviceConnecting()-deviceAddress=CE:99:42:70:BA:CF
04-23 16:59:36.475 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onDfuProcessStarting-deviceAddress=CE:99:42:70:BA:CF
04-23 16:59:41.949 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=0
04-23 16:59:42.016 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=1
04-23 16:59:42.045 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=2
04-23 16:59:42.073 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=3
04-23 16:59:42.095 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=4
04-23 16:59:42.124 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=5
04-23 16:59:42.149 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=6
04-23 16:59:42.180 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=7
04-23 16:59:42.205 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=8
04-23 16:59:42.249 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=9
04-23 16:59:42.278 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=10
04-23 16:59:42.308 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=11
04-23 16:59:42.349 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=12
04-23 16:59:42.372 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=13
04-23 16:59:42.397 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=14
04-23 16:59:42.419 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=15
04-23 16:59:42.445 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=16
04-23 16:59:42.471 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=17
04-23 16:59:42.500 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=18
04-23 16:59:42.522 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=19
04-23 16:59:45.650 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=20
04-23 16:59:45.742 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=21
04-23 16:59:45.770 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=22
04-23 16:59:45.790 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=23
04-23 16:59:45.815 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=24
04-23 16:59:45.861 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=25
04-23 16:59:45.892 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=26
04-23 16:59:45.912 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=27
04-23 16:59:45.940 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=28
04-23 16:59:45.966 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=29
04-23 16:59:49.121 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=30
04-23 16:59:49.217 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=31
04-23 16:59:49.246 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=32
04-23 16:59:49.269 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=33
04-23 16:59:49.297 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=34
04-23 16:59:49.319 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=35
04-23 16:59:49.343 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=36
04-23 16:59:49.363 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=37
[... cut ...]
04-23 17:00:12.827 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=99
04-23 17:00:12.860 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=100
04-23 17:00:17.048 11901-12390/no.nordicsemi.android.nrftoolbox E/DfuBaseService: Unable to write Op Code: device disconnected
04-23 17:00:17.059 11901-11901/no.nordicsemi.android.nrftoolbox E/bm: onError-deviceAddress=CE:99:42:70:BA:CF error=4096 errorType=0 message=DFU DEVICE DISCONNECTED

philips77, 2018-04-23:
Could you enable debug log in the lib? Return true from isDebug in your DfuService class. And then paste the log, but just from around 98%, the ending.

jingmo0319, 2018-04-23:
I hava enabled debug log in the lib:
public class DfuService extends DfuBaseService {

    @Override
    protected Class<? extends Activity> getNotificationTarget() {
        /*
		 * As a target activity the NotificationActivity is returned, not the MainActivity. This is because the notification must create a new task:
		 * 
		 * intent.addFlags(Intent.FLAG_ACTIVITY_NEW_TASK);
		 * 
		 * when user press it. Using NotificationActivity we can check whether the new activity is a root activity (that means no other activity was open before)
		 * or that there is other activity already open. In the later case the notificationActivity will just be closed. System will restore the previous activity.
		 * However if the application has been closed during upload and user click the notification a NotificationActivity will be launched as a root activity.
		 * It will create and start the main activity and terminate itself.
		 * 
		 * This method may be used to restore the target activity in case the application was closed or is open. It may also be used to recreate an activity
		 * history (see NotificationActivity).
		 */
        return NotificationActivity.class;
    }

    @Override
    protected boolean isDebug() {
        return true;
    }
}

But I only get the following logs:
04-23 17:26:26.595 18898-18898/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=98
04-23 17:26:26.615 18898-18898/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=99
04-23 17:26:26.643 18898-18898/no.nordicsemi.android.nrftoolbox E/bm: onProgressChanged-deviceAddress=CE:99:42:70:BA:CF percent=100
04-23 17:26:31.187 18898-19295/no.nordicsemi.android.nrftoolbox E/DfuBaseService: Unable to write Op Code: device disconnected
04-23 17:26:31.215 18898-18898/no.nordicsemi.android.nrftoolbox E/bm: onError-deviceAddress=CE:99:42:70:BA:CF error=4096 errorType=0 message=DFU DEVICE DISCONNECTED
How do I get more logs?

philips77, 2018-04-23:
Are you using Legacy or Secure DFU?
Looks like you have filtered only to Error logs in your LogCat.

philips77, 2018-10-24:
Did you manage to solve the issue?

closed 2019-01-29 by philips77