nordicsemi/Android-DFU-Library #104: Packet transfer stop while sending data.

tusharSensovium, 2018-04-30:
I also face this issue in Redmi 3s prime(xiaomi). some times dfu stopped at 20% - 25% . And same code works on others mobiles like samsung, sony.

    ```
Executing data object (Op Code = 4)
04-30 19:21:06.119 24753-25649/focus.radiance.com.wbdfu I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (2/7)
04-30 19:21:06.270 24753-25649/focus.radiance.com.wbdfu I/DfuImpl: Uploading firmware...
04-30 19:21:08.880 24753-25649/focus.radiance.com.wbdfu I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
04-30 19:21:08.939 24753-25649/focus.radiance.com.wbdfu I/DfuImpl: Checksum received (Offset = 8192, CRC = A0E2CE9D)
04-30 19:21:08.940 24753-25649/focus.radiance.com.wbdfu I/DfuImpl: Executing data object (Op Code = 4)
04-30 19:21:08.999 24753-25649/focus.radiance.com.wbdfu I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (3/7)
04-30 19:21:09.149 24753-25649/focus.radiance.com.wbdfu I/DfuImpl: Uploading `firmware..
```

**application stucks here.**

philips77, 2018-04-30:
Did you try to enable PRNs in the initiator?
The log you have looks correct. Nothing wrong. Please debug where it stacks.

tusharSensovium, 2018-04-30:
I m using your latest library 1.6.1 so, by default PRNs is 12.

tusharSensovium, 2018-04-30:
> It's by default also disabled on Android 6+. You may force enable it with
> setPRPEnabled(..).

i also tried this. but still not work.

 service.setPacketsReceiptNotificationsEnabled(true);

-- 

Thanks.

*Tushar Joshi*Android Developer
Sensovium PVT LTD

M:  +91 9824206579

ABOUT THIS E-MAIL: The information transmitted is intended only for the
person or entity to which it is addressed and may contain confidential
and/or privileged material. Any review, re-transmission, dissemination or
other use of, or taking of any action in reliance upon, this information by
persons or entities other than the intended recipient is prohibited. If you
received this in error, please contact the sender and delete the material
from any computer.

philips77, 2018-04-30:
Then you have to debug it more on your own. Check whether it stops and why. There may be number of reasons on the device or Android side... I can't conclude from the logs you posted.

tusharSensovium, 2018-04-30:
But on same on nrfConnect, nrf Toolbox and my other DFU application(with same library and same your code) are working. I m totally confused what happening with this device and code. By the way thanks for quick response. Right now i m not closing this issue. When i solved it then close this one. Again thank you very much for your important time and response.

tusharSensovium, 2018-05-02:
Whenever DFU stuck my i got this logs every time
05-03 00:34:38.988 28641-28655/focus.radiance.com.wbdfu D/BluetoothGatt: onClientConnParamsChanged() - Device=C2:57:00:74:D5:45 interval=6 status=0
05-03 00:34:39.218 28641-28663/focus.radiance.com.wbdfu D/BluetoothGatt: onClientConnParamsChanged() - Device=C2:57:00:74:D5:45 interval=23 status=0
05-03 00:34:39.708 28641-28652/focus.radiance.com.wbdfu D/BluetoothGatt: onClientConnParamsChanged() - Device=C2:57:00:74:D5:45 interval=23 status=0

tusharSensovium, 2018-10-24:
> Hi, did you manage to solve the issue?

Not yet. I didn't find any solution. Have you any suggestions?

-- 

Thanks.

*Tushar Joshi*Android Developer
Sensovium PVT LTD

M:  +91 9824206579

ABOUT THIS E-MAIL: The information transmitted is intended only for the
person or entity to which it is addressed and may contain confidential
and/or privileged material. Any review, re-transmission, dissemination or
other use of, or taking of any action in reliance upon, this information by
persons or entities other than the intended recipient is prohibited. If you
received this in error, please contact the sender and delete the material
from any computer.

philips77, 2018-10-24:
Well.. I would have to see the Bluetooth sniffer logs. The upload for some reason stopped and the lib is waiting for some callback. I can't find the reason based on the logs you gave. We need more info, for example HCI logs, sniffer logs, or be able to reproduce it on our side.
How often does the issue happen?

tusharSensovium, 2018-10-24:
Most frequently

-- 

Thanks.

*Tushar Joshi*Android Developer
Sensovium PVT LTD

M:  +91 9824206579

ABOUT THIS E-MAIL: The information transmitted is intended only for the
person or entity to which it is addressed and may contain confidential
and/or privileged material. Any review, re-transmission, dissemination or
other use of, or taking of any action in reliance upon, this information by
persons or entities other than the intended recipient is prohibited. If you
received this in error, please contact the sender and delete the material
from any computer.

jeffhuangtw, 2019-05-16:
In my case, sometimes firmware already partial uploaded, let's say 25% and app is force closed.
Then the next time app will upload firmware from 25% and resume uploading.
This can cause a CRC mismatch in my case, so I just disable the resume upload firmware and everything can work normally.

service.disableResume();

papfonseca, 2019-08-26:
Hi, 

I'm facing the same problem. The app just stops uploading. This not happens all the time, but i don't know how to reproduce:

```08 01 - 12:40:09 DFUChildFragment -> Start DFU - Recover: false
08 01 - 12:40:09 DFUChildFragment -> [DFU] DFU service started
08 01 - 12:40:09 DFUChildFragment -> [DFU] Opening file...
08 01 - 12:40:09 DFUChildFragment -> [DFU] Firmware file opened successfully
08 01 - 12:40:09 DFUChildFragment -> [DFU] Connecting to DFU target...
08 01 - 12:40:09 DFUChildFragment -> [DFU] gatt = device.connectGatt(autoConnect = false)
08 01 - 12:40:09 DFUChildFragment -> [DFU] Connected to E9:13:C9:4E:64:F3
08 01 - 12:40:09 DFUChildFragment -> [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
08 01 - 12:40:09 DFUChildFragment -> [DFU] Discovering services...
08 01 - 12:40:09 DFUChildFragment -> [DFU] gatt.discoverServices()
08 01 - 12:40:10 DFUChildFragment -> [DFU] Services discovered
08 01 - 12:40:10 DFUChildFragment -> DFU Device Connected: E9:13:C9:4E:64:F3
08 01 - 12:40:10 DFUChildFragment -> DFU Proccess Starting: E9:13:C9:4E:64:F3
08 01 - 12:40:10 DFUChildFragment -> [DFU] wait(1000)
08 01 - 12:40:11 DFUChildFragment -> [DFU] Requesting new MTU...
08 01 - 12:40:11 DFUChildFragment -> [DFU] gatt.requestMtu(517)
08 01 - 12:40:11 DFUChildFragment -> [DFU] MTU changed to: 23
08 01 - 12:40:11 DFUChildFragment -> [DFU] Enabling notifications for 8ec90001-f315-4f60-9fb8-838830daea50
08 01 - 12:40:11 DFUChildFragment -> [DFU] gatt.setCharacteristicNotification(8ec90001-f315-4f60-9fb8-838830daea50, true)
08 01 - 12:40:11 DFUChildFragment -> [DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
08 01 - 12:40:11 DFUChildFragment -> [DFU] Data written to descr.8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-00
08 01 - 12:40:11 DFUChildFragment -> [DFU] Notifications enabled for 8ec90001-f315-4f60-9fb8-838830daea50
08 01 - 12:40:11 DFUChildFragment -> [DFU] Notifications enabled
08 01 - 12:40:11 DFUChildFragment -> [DFU] wait(1000)
08 01 - 12:40:12 DFUChildFragment -> [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
08 01 - 12:40:12 DFUChildFragment -> [DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
08 01 - 12:40:12 DFUChildFragment -> [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 06-01
08 01 - 12:40:12 DFUChildFragment -> [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-06-01-00-01-00-00-00-00-00-00-00-00-00-00
08 01 - 12:40:12 DFUChildFragment -> [DFU] Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
08 01 - 12:40:12 DFUChildFragment -> [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
08 01 - 12:40:13 DFUChildFragment -> [DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
08 01 - 12:40:13 DFUChildFragment -> [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 02-00-00
08 01 - 12:40:13 DFUChildFragment -> [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-02-01
08 01 - 12:40:13 DFUChildFragment -> [DFU] Packet Receipt Notif disabled (Op Code = 2, Value = 0)
08 01 - 12:40:13 DFUChildFragment -> [DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
08 01 - 12:40:13 DFUChildFragment -> [DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
08 01 - 12:40:13 DFUChildFragment -> [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-01-87-00-00-00
08 01 - 12:40:13 DFUChildFragment -> [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
08 01 - 12:40:13 DFUChildFragment -> [DFU] Command object created
08 01 - 12:40:13 DFUChildFragment -> [DFU] Writing to characteristic 8ec90002-f315-4f60-9fb8-838830daea50
08 01 - 12:40:13 DFUChildFragment -> [DFU] gatt.writeCharacteristic(8ec90002-f315-4f60-9fb8-838830daea50)
08 01 - 12:40:13 DFUChildFragment -> [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 12-84-01-0A-3E-08-01-12-3A-08-01-10-34-1A-02-8C-01-20-00-28
08 01 - 12:40:13 DFUChildFragment -> [DFU] Writing to characteristic 8ec90002-f315-4f60-9fb8-838830daea50
08 01 - 12:40:13 DFUChildFragment -> [DFU] gatt.writeCharacteristic(8ec90002-f315-4f60-9fb8-838830daea50)
08 01 - 12:40:13 DFUChildFragment -> [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 00-30-00-38-C0-BA-0C-42-24-08-03-12-20-70-E3-14-CB-75-01-93
08 01 - 12:40:13 DFUChildFragment -> [DFU] Writing to characteristic 8ec90002-f315-4f60-9fb8-838830daea50
08 01 - 12:40:13 DFUChildFragment -> [DFU] gatt.writeCharacteristic(8ec90002-f315-4f60-9fb8-838830daea50)
08 01 - 12:40:13 DFUChildFragment -> [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): C8-BE-66-50-94-8B-92-A8-B9-62-F3-AF-51-19-81-24-F8-5B-0C-B0
08 01 - 12:40:13 DFUChildFragment -> [DFU] Writing to characteristic 8ec90002-f315-4f60-9fb8-838830daea50
08 01 - 12:40:13 DFUChildFragment -> [DFU] gatt.writeCharacteristic(8ec90002-f315-4f60-9fb8-838830daea50)
08 01 - 12:40:13 DFUChildFragment -> [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 70-52-1C-12-BD-48-00-10-00-1A-40-6D-E5-C3-58-6C-77-6C-38-01
08 01 - 12:40:13 DFUChildFragment -> [DFU] Writing to characteristic 8ec90002-f315-4f60-9fb8-838830daea50
08 01 - 12:40:13 DFUChildFragment -> [DFU] gatt.writeCharacteristic(8ec90002-f315-4f60-9fb8-838830daea50)
[... cut ...]
I'm using 1.9.0 sdk version.

``` 
DfuServiceInitiator dfuStarter = new DfuServiceInitiator(mNewDeviceAddress)
                .setDeviceName(ResourceUtils.getString(R.string.dfu_device_name))
                .setForeground(true)
                .setDisableNotification(true)
                .setKeepBond(true)
                .setPacketsReceiptNotificationsEnabled(true);

        if (VersionUtils.isAtLeastO()) {
            DfuServiceInitiator.createDfuNotificationChannel(mActivity);
        }
```

Thank You!