nordicsemi/Android-DFU-Library #105: I close Bluetooth when dfu uploading,onConnectionStateChange didn't be called

liuwenzheng, 2018-05-16:
android 8.1.0
Piexl
Another phone's onConnectionStateChange  will be called and status is STATE_DISCONNECTED,but Piexl not.
When I use dfu again,it didn't work,onDeviceConnecting didn't be called,the progress stuck.
I use nRF Connect app can recurrent this.

philips77, 2018-07-03:
Hi,
Do you still have this issue? Could you post here some logs/more details how to reproduce it?

liuwenzheng, 2018-07-10:
Yes,this issue still exist.
My app name is fitpolo,I test dfu upgrade with different phone and log the **DfuProgressListenerAdapter**.
After the dfu started,I close the phone's  Bluetooth,The log is:

Piexl:

07-10 17:24:32.883 19772-19772/com.moko.fitpolo W/fitpolo: onDeviceConnecting...
07-10 17:24:32.886 19772-21340/com.moko.fitpolo D/BluetoothGatt: connect() - device: F1:FB:DD:ED:92:2D, auto: false
07-10 17:24:32.886 19772-21340/com.moko.fitpolo D/BluetoothGatt: registerApp()
07-10 17:24:32.887 19772-21340/com.moko.fitpolo D/BluetoothGatt: registerApp() - UUID=5a6ebc68-c434-47f8-baa0-75a5d1cd0d10
07-10 17:24:32.891 19772-19794/com.moko.fitpolo D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
07-10 17:24:32.894 19772-19794/com.moko.fitpolo D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=F1:FB:DD:ED:92:2D
07-10 17:24:32.896 19772-19794/com.moko.fitpolo D/BluetoothGatt: discoverServices() - device: F1:FB:DD:ED:92:2D
07-10 17:24:32.903 19772-19794/com.moko.fitpolo D/BluetoothGatt: onSearchComplete() = Device=F1:FB:DD:ED:92:2D Status=0
07-10 17:24:33.907 19772-19772/com.moko.fitpolo W/fitpolo: DfuProcessStarting...
07-10 17:24:33.979 19772-21340/com.moko.fitpolo D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
07-10 17:24:36.140 19772-19772/com.moko.fitpolo I/fitpolo: onProgressChanged...0
07-10 17:24:37.377 19772-19772/com.moko.fitpolo I/fitpolo: onProgressChanged...1
07-10 17:24:38.783 19772-19772/com.moko.fitpolo I/fitpolo: onProgressChanged...2
07-10 17:24:40.130 19772-19772/com.moko.fitpolo I/fitpolo: onProgressChanged...3
07-10 17:24:41.468 19772-19772/com.moko.fitpolo I/fitpolo: onProgressChanged...4
07-10 17:24:42.798 19772-19772/com.moko.fitpolo I/fitpolo: onProgressChanged...5
07-10 17:24:44.144 19772-19772/com.moko.fitpolo I/fitpolo: onProgressChanged...6
07-10 17:24:46.533 19772-19786/com.moko.fitpolo D/BluetoothAdapter: onBluetoothServiceDown: android.bluetooth.IBluetooth$Stub$Proxy@907d330

**result:the onConnectionStateChange didn't be called!**

The other phone's log:

07-10 17:19:22.949 6774-6774/com.moko.fitpolo W/fitpolo: onDeviceConnecting...
07-10 17:19:22.969 6774-8701/com.moko.fitpolo D/BluetoothGatt: connect() - device: F1:FB:DD:ED:92:2D, auto: false
07-10 17:19:22.969 6774-8701/com.moko.fitpolo D/BluetoothGatt: registerApp()
07-10 17:19:22.969 6774-8701/com.moko.fitpolo D/BluetoothGatt: registerApp() - UUID=a37291e0-f8bd-4476-934b-2aab7c2f1997
07-10 17:19:23.019 6774-8353/com.moko.fitpolo D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
07-10 17:19:23.039 6774-7745/com.moko.fitpolo D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=F1:FB:DD:ED:92:2D
07-10 17:19:23.049 6774-7745/com.moko.fitpolo D/BluetoothGatt: discoverServices() - device: F1:FB:DD:ED:92:2D
07-10 17:19:23.369 6774-6786/com.moko.fitpolo D/BluetoothGatt: onClientConnParamsChanged() - Device=F1:FB:DD:ED:92:2D interval=6 status=0
07-10 17:19:23.889 6774-6786/com.moko.fitpolo D/BluetoothGatt: onSearchComplete() = Device=F1:FB:DD:ED:92:2D Status=0
07-10 17:19:23.919 6774-7745/com.moko.fitpolo D/BluetoothGatt: onClientConnParamsChanged() - Device=F1:FB:DD:ED:92:2D interval=39 status=0
07-10 17:19:24.899 6774-6774/com.moko.fitpolo W/fitpolo: DfuProcessStarting...
07-10 17:19:24.989 6774-8701/com.moko.fitpolo D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
07-10 17:19:29.279 6774-6774/com.moko.fitpolo I/fitpolo: onProgressChanged...0
07-10 17:19:30.539 6774-6774/com.moko.fitpolo I/fitpolo: onProgressChanged...1
07-10 17:19:32.099 6774-6774/com.moko.fitpolo I/fitpolo: onProgressChanged...2
07-10 17:19:33.529 6774-6774/com.moko.fitpolo I/fitpolo: onProgressChanged...3
07-10 17:19:35.059 6774-6774/com.moko.fitpolo I/fitpolo: onProgressChanged...4
07-10 17:19:35.649 6774-6786/com.moko.fitpolo D/BluetoothAdapter: onBluetoothStateChange: up=false
07-10 17:19:35.649 6774-6786/com.moko.fitpolo D/BluetoothAdapter: Bluetooth is turned off, stop adv
07-10 17:19:35.649 6774-6786/com.moko.fitpolo D/BluetoothAdapter: There are no active google scan apps, stop scan
07-10 17:19:35.649 6774-6786/com.moko.fitpolo D/BluetoothLeScanner: stopAllScan standalone boolean is value is = false
07-10 17:19:35.649 6774-6786/com.moko.fitpolo D/BluetoothLeScanner: Exiting stopAllScan
07-10 17:19:35.649 6774-6785/com.moko.fitpolo D/BluetoothGatt: onBluetoothStateChange: up=false
07-10 17:19:35.649 6774-6785/com.moko.fitpolo D/BluetoothGatt: Bluetooth is turned off, disconnect all client connections
07-10 17:19:35.649 6774-6785/com.moko.fitpolo D/BluetoothGatt: cancelOpen() - device: F1:FB:DD:ED:92:2D
07-10 17:19:35.659 6774-7745/com.moko.fitpolo D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=F1:FB:DD:ED:92:2D
07-10 17:19:35.659 6774-7745/com.moko.fitpolo E/fitpolo: onConnectionStateChange
07-10 17:19:35.659 6774-7745/com.moko.fitpolo E/fitpolo: status : 0
07-10 17:19:35.659 6774-7745/com.moko.fitpolo E/fitpolo: newState : 0

**result:the onConnectionStateChange be called!**

Why?

philips77, 2018-10-24:
I don't know why, but as a workaround, I've added a broadcast receiver that will listen to Bluetooth state changes. If Bluetooth goes OFF, the receiver will stop the service.

Thanks for creating the issue!

PR #139 (2018-10-24): Handling Bluetooth off during upload
This should fix #105. On some phones the `onConnectionStateChanged` is not called when Bluetooth goes off. This PR adds a broadcast received that will close the DFU service in such case, as if the callback was received.
closed 2018-10-24 by philips77