nordicsemi/Android-DFU-Library #106: Bad notification for startForeground

sim1camporeale, 2018-05-24:
On Huawei P20 Pro With Android 8.1.0 I had to disable startForeground in order to let nordic dfu updater works, otherwise app crashes with this error:

```
05-24 16:09:08.024 12022-12022/bike.woolf.woolfupdater E/AndroidRuntime: FATAL EXCEPTION: main
                                                                         Process: com.example.dfuupdater, PID: 12022
                                                                         android.app.RemoteServiceException: Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=dfu pri=-1 contentView=null vibrate=null sound=null defaults=0x0 flags=0x42 color=0xff888888 vis=PRIVATE)
                                                                             at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2095)
                                                                             at android.os.Handler.dispatchMessage(Handler.java:109)
                                                                             at android.os.Looper.loop(Looper.java:166)
                                                                             at android.app.ActivityThread.main(ActivityThread.java:7367)
                                                                             at java.lang.reflect.Method.invoke(Native Method)
                                                                             at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:469)
                                                                             at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:963)
```
So i replaced this statement 
```
final DfuServiceInitiator starter = new DfuServiceInitiator(mSelectedDevice.getAddress())
                .setDeviceName(mSelectedDevice.getName())
                .setKeepBond(keepBond);
```
with
```
        final DfuServiceInitiator starter = new DfuServiceInitiator(mSelectedDevice.getAddress())
                .setForeground(false)                // Foreground doesn't work correctly for android 8.1
                // See https://stackoverflow.com/questions/47531742/startforeground-fail-after-upgrade-to-android-8-1
                .setDeviceName(mSelectedDevice.getName())
                .setKeepBond(keepBond);
```
Instead I think that startForeground should be implemented according to this issue notified on stackoverflow:
https://stackoverflow.com/questions/47531742/startforeground-fail-after-upgrade-to-android-8-1

philips77, 2018-05-24:
On Android 8.1 you have to create a notification channel. Check some last helper method in dfuServiceInitiator.

sim1camporeale, 2018-05-24:
Ok, thanks for your response.

So I added this in onCreate
```
        // Enable Notification Channel for Android OREO
        if(Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            Context ctx = getApplicationContext();
            DfuServiceInitiator.createDfuNotificationChannel(ctx);
        }
```

closed 2018-06-15 by philips77

nijat2018, 2018-11-14:
I have the same issue on my Android Things Board and solved by
`starter.setForeground(false);
            starter.setDisableNotification(true);`
Anyway I don't need them.

1ALo, 2019-03-27:
> I have the same issue on my Android Things Board and solved by
> `starter.setForeground(false);`
> `starter.setDisableNotification(true);`
> Anyway I don't need them.

what is starter in your code ?

sandeep8601, 2019-05-10:
Just shifted from android 7.1.1 to android 7.0.0 and it works for me

philips77, 2019-05-10:
@1ALo 

> what is starter in your code ?

starter is the DfuServiceInitiator, I guess.