nordicsemi/Android-DFU-Library #108: java.lang.NullPointerException

zhouzhijin0908, 2018-06-15:
SUPPORTED_64_BIT_ABIS=[Ljava.lang.String;@7c2da6f
versionCode=200
BOARD=universal7870
BOOTLOADER=A320FXXU2BQI3
IS_TRANSLATION_ASSISTANT_ENABLED=false
TYPE=user
ID=NRD90M
TIME=1506083097000
BRAND=samsung
TAG=Build
HARDWARE=samsungexynos7870
SERIAL=5200e2be5e95b41f
SUPPORTED_ABIS=[Ljava.lang.String;@d5f037c
CPU_ABI=armeabi-v7a
IS_DEBUGGABLE=false
RADIO=unknown
MANUFACTURER=samsung
IS_EMULATOR=false
SUPPORTED_32_BIT_ABIS=[Ljava.lang.String;@712b94e
TAGS=release-keys
CPU_ABI2=armeabi
IS_SYSTEM_SECURE=false
UNKNOWN=unknown
PERMISSIONS_REVIEW_REQUIRED=false
USER=dpi
FINGERPRINT=samsung/a3y17ltexx/a3y17lte:7.0/NRD90M/A320FXXU2BQI3:user/release-keys
HOST=SWDG4716
versionName=1.2.1
PRODUCT=a3y17ltexx
DISPLAY=NRD90M.A320FXXU2BQI3
MODEL=SM-A320F
DEVICE=a3y17lte
IS_SECURE=false
java.lang.NullPointerException: Attempt to get length of null array
	at no.nordicsemi.android.dfu.internal.ArchiveInputStream.read(ArchiveInputStream.java:338)
	at no.nordicsemi.android.dfu.BaseCustomDfuImpl.uploadFirmwareImage(BaseCustomDfuImpl.java:351)
	at no.nordicsemi.android.dfu.LegacyDfuImpl.performDfu(LegacyDfuImpl.java:467)
	at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1173)
	at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:67)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.os.HandlerThread.run(HandlerThread.java:61)

zhouzhijin0908, 2018-07-07:
can you handle it?

jasonim, 2018-07-25:
i also the same question

lachtanek, 2018-08-02:
happens when app is killed during DFU

closed 2018-10-24 by philips77