nordicsemi/Android-DFU-Library #109: E/DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED

zhouzhijin0908, 2018-06-15:
E/DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED

what's means? why?

philips77, 2018-06-15:
Hi,
If you're using the DFU bootloader from SDK 13, there may be many reasons for error 6.
All are caused by an invalid ZIP file with the firmware. It may be wrong signature, incorrect required SoftDevice ID, invalid app version  (must be >= to the current app verison), invalid bootloader version (if you update the bootloader (must be > then the current bl version).
Ensure that your ZIP is created correctly and is signed with correct private key (that matches the public key in the bootloader code), that it was created with --sd-id matching the current SD, etc.

zhouzhijin0908, 2018-06-15:
Same bracelet, same upgrade firmware package;
Upgrade 100 times and have a 2% chance of this error on some phones.
But on some phones (sm-a320f, android7.0) there is a 60% chance of this error

philips77, 2018-06-15:
Hmm... Could you post relevant log here? 
It may be caused by the device sending too quickly and flash saving is not fast enough, or some other reason. Did you try enabling PRNs on those devices with [this method](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L165)?

philips77, 2018-10-17:
Hi, did you manage to solve your issue?

philips77, 2018-10-24:
I'll close the issue due to inactivity. If you still face the problem, please open a new issue and give more details.

closed 2018-10-24 by philips77

feiyangjiayuan, 2018-12-14:
[image]

philips77, 2018-12-14:
Hi, what SDK is your DFU bootloader from?