nordicsemi/Android-DFU-Library #110: DFU  error on samsung s8/s9

nexmtushar, 2018-06-20:
DFU service created. Version: 1.6.1
Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
Connecting to the device...
Action received: android.bluetooth.device.action.ACL_CONNECTED
Connected to GATT server
Attempting to start service discovery... succeed
Services discovered
transfer file size = 0
transfer file size = 0
Disconnecting from the device...
Disconnected from GATT server
Refreshing result: true
Cleaning up...
Action received: android.bluetooth.device.action.ACL_DISCONNECTED
onerror = 262msg = OPERATION FAILED

philips77, 2018-07-03:
Hi,
Do you know the reason why the device gets disconnected? Could you post logs you get in the LocGat when you return true from isDebug() method in your DfuService class?

nexmtushar, 2018-07-05:
hi, 
these logs are  with  DfuService class . It should help 

lenght of file = 67.0
starting dfu process
DFU service created. Version: 1.6.1
Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
An exception occurred while calculating file sizejava.util.zip.ZipException: unexpected EOF
onerror = 4098msg = DFU FILE ERROR
DFU service destroyed
length of file = 67.0
starting dfu process
DFU service created. Version: 1.6.1
Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
Connecting to the device...
Action received: android.bluetooth.device.action.ACL_CONNECTED
Connected to GATT server
Attempting to start service discovery... succeed
Services discovered
Reading DFU version number...
Version number read: 0.8 -> Bootloader from SDK 9.0 or newer. Signature supported
Legacy DFU bootloader found
Enabling notifications...
Sending Start DFU command (Op Code = 1, Upload Mode = 4)
Sending image size array to DFU Packet (0b, 0b, 66952b)
Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
Sending 14 bytes of init packet
Sending init packet (Value = 01-00-01-00-FF-FF-FF-FF-01-00-FE-FF-37-71)
Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
Sending Receive Firmware Image request (Op Code = 3)
Uploading firmware...
Response received (Op Code = 16, Req Op Code = 3, Status = 6)
Sending Reset command (Op Code = 6)
Disconnecting from the device...
Disconnected from GATT server
Refreshing result: true
Cleaning up...
DFU service destroyed
length of file = 67.0
starting dfu process
DFU service created. Version: 1.6.1
Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
Connecting to the device...
Action received: android.bluetooth.device.action.ACL_CONNECTED
Connected to GATT server
Attempting to start service discovery... succeed
Services discovered
Reading DFU version number...
Version number read: 0.8 -> Bootloader from SDK 9.0 or newer. Signature supported
Legacy DFU bootloader found
Enabling notifications...
length of file = 67.0
starting dfu process
Sending Start DFU command (Op Code = 1, Upload Mode = 4)
Sending image size array to DFU Packet (0b, 0b, 66952b)
Sending Reset command (Op Code = 6)
Target device disconnected with status: 19
Refreshing result: true
Cleaning up...
Action received: android.bluetooth.device.action.ACL_DISCONNECTED
Restarting the service
Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
device connecting 
Connecting to the device...
Connection state change error: 133 newState: 0
Device not reachable. Check if the device with address C3:E3:6F:E3:95:92 is in range, is advertising and is connectable
Refreshing result: true
Cleaning up...
onerror = 4096msg = DFU DEVICE DISCONNECTED
Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
Connecting to the device...
Connection state change error: 133 newState: 0
Device not reachable. Check if the device with address C3:E3:6F:E3:95:92 is in range, is advertising and is connectable
Refreshing result: true
Cleaning up...
onerror = 4096msg = DFU DEVICE DISCONNECTED
DFU service destroyed
length of file = 67.0
length of file = 67.0
length of file = 67.0
length of file = 67.0
starting dfu process
DFU service created. Version: 1.6.1
Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
Connecting to the device...
Action received: android.bluetooth.device.action.ACL_CONNECTED
Connected to GATT server
Attempting to start service discovery... succeed
Services discovered
Reading DFU version number...
Version number read: 0.8 -> Bootloader from SDK 9.0 or newer. Signature supported
Legacy DFU bootloader found
Enabling notifications...
Sending Start DFU command (Op Code = 1, Upload Mode = 4)
Sending image size array to DFU Packet (0b, 0b, 66952b)
Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
Sending 14 bytes of init packet
Sending init packet (Value = 01-00-01-00-FF-FF-FF-FF-01-00-FE-FF-37-71)
Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
Sending Receive Firmware Image request (Op Code = 3)
Uploading firmware...
transfer file size = 0
transfer file size = 0
transfer file size = 1
transfer file size = 2
transfer file size = 2
transfer file size = 3
transfer file size = 4
transfer file size = 4
transfer file size = 5
Response received (Op Code = 16, Req Op Code = 3, Status = 6)
Sending Reset command (Op Code = 6)
Disconnecting from the device...
Disconnected from GATT server
Refreshing result: true
Cleaning up...
Action received: android.bluetooth.device.action.ACL_DISCONNECTED
onerror = 262msg = OPERATION FAILED

philips77, 2018-07-05:
Try enabling PRNs. Call setPackageR..Notif...Enabled in the dfuServiceInitiator.

philips77, 2018-10-24:
Hi, did the PRNs work for you?

Tush1995, 2018-11-26:
Not working, Dfu work fine for all devices excepts samsung note s8, galaxy s9, galaxy s9 plus

philips77, 2018-11-26:
Hi, we have s9 here, I'll check it.

eduardo-rta, 2019-01-31:
I was having the same issue, I've just changed the Target and Compile to 28 and everything is working now (1.8.1)

Min 21
Target 28
Compile 28
Build tool 28.0.3

I'm using the options: 
setZip
setDeviceName
setKeepBond(false)
setForeground(false)

Tested on: 
Samsung J1 (Android 6.0.1)
Galaxy S5 (Android 6.0.1), 
Galaxy S8 (Android 8.0.0), 
Galaxy S9+ (Android 8.0.0)
OnePlus 6T (Android 9)
Pixel 3 (Android 9)

setForeground = True in android 9 is causing the app to crash

philips77, 2019-01-31:
You need to add FOREGROUND_SERVICE permission on Android 9.

closed 2020-11-30 by roshanrajaratnam