nordicsemi/Android-DFU-Library #118: Program type already present: no.nordicsemi.android.dfu.BaseButtonlessDfuImpl$ButtonlessBluetoothCallback

Basavaraj1991, 2018-07-24:
Hi 

I'm using DFU while generating apk i'm getting the below issue in both debug and release build

philips77, 2018-07-24:
Looks like you have added dfu lib dependency twice?

Basavaraj1991, 2018-07-24:
Hi I'm using below thing as well
    api 'no.nordicsemi.android.support.v18:scanner:1.0.0'

philips77, 2018-07-24:
This should not matter. Could you paste your Gradle file here?

Basavaraj1991, 2018-07-24:
api 'com.google.android.gms:play-services-auth:15.0.1'
    api 'com.google.android.gms:play-services-location:15.0.1'
    api 'com.google.android.gms:play-services-auth:15.0.1'
    api 'com.google.android.gms:play-services-location:15.0.1'
    api 'com.android.support:appcompat-v7:27.1.1'
    api 'com.android.support:cardview-v7:27.1.1'
    api 'com.android.support:recyclerview-v7:27.1.1'
    api 'com.github.bumptech.glide:glide:3.7.0'
    api 'com.android.support:support-v4:27.1.1'
    api 'cz.msebera.android:httpclient:4.4.1.2'
    api 'com.google.code.gson:gson:2.8.2'
    api 'com.github.lzyzsd:circleprogress:1.2.1'
    api 'com.googlecode.libphonenumber:libphonenumber:7.0.6'
    api 'com.squareup:otto:1.3.8'
/*
    implementation 'com.facebook.android:facebook-android-sdk:[4,5)'
*/
    api 'com.facebook.android:facebook-android-sdk:4.26.0'
    api 'com.google.android.gms:play-services-plus:15.0.1'
    api 'com.google.android.gms:play-services-analytics:16.0.1'
    api 'com.google.android.gms:play-services-maps:15.0.1'
    api 'com.google.firebase:firebase-core:16.0.1'
    api 'com.google.firebase:firebase-messaging:17.0.0'
    api 'com.google.firebase:firebase-crash:16.0.1'
    api 'com.theartofdev.edmodo:android-image-cropper:2.3.+'
    api 'com.google.android.gms:play-services-fitness:15.0.1'
    api 'com.google.android.gms:play-services-awareness:15.0.1'
    api 'com.github.JakeWharton:ViewPagerIndicator:2.4.1'
    api 'com.mostafagazar:customshapeimageview:1.0.4'
    api 'no.nordicsemi.android.support.v18:scanner:1.0.0'
    api 'com.android.support:design:28.0.0-alpha1'
    api 'com.google.android.gms:play-services-gcm:15.0.1'
    api 'com.squareup.retrofit2:retrofit:2.4.0'
    api 'com.squareup.retrofit2:converter-gson:2.4.0'
    api 'org.greenrobot:eventbus:3.1.1'
    api 'com.github.warkiz.widget:indicatorseekbar:2.0.7'
    api 'com.google.android.gms:play-services-places:15.0.1'
    api 'com.android.support.constraint:constraint-layout:1.1.2'
    api 'com.google.maps.android:android-maps-utils:0.5'
    api "org.jetbrains.kotlin:kotlin-stdlib-jre7:$kotlin_version"
    api 'com.otaliastudios:cameraview:1.5.1'
    api 'cn.aigestudio.wheelpicker:WheelPicker:1.1.2'
    api 'no.nordicsemi.android:dfu:1.7.0'

philips77, 2018-07-24:
Seems ok. Any reason why you use 'api' instead of 'implementation' everywhere?

Basavaraj1991, 2018-07-24:
Thank you its resolved coz i was using dfu library module as well

closed 2018-07-24 by philips77