nordicsemi/Android-DFU-Library #124: Handling zip containg bl+sd+app with BL already up-to-date

NitroG42, 2018-09-17:
Hello, 
I'm having a hard time handling a case where our device needs to update its bootloader and his application, but the bootloader is already at the latest version.
To give some context the user use our phone app to pair with our device, it will try to update it to the latest version. For some reason, if the bootloader is updated but not the app, we won't be able to know (because updating the bootloader first make our phone unable to read the BL version, so it will try to update it again). The issue is that when we try to update the BL, we get a FW version error, because the bootloader is already up to date. It's normal, the issue is that we can't update the app right after, because the device is in bootloader mode and the DFU library won't find the device.

My thinking was to send a zip containing all (SD, BL, and app, before we were separating between two zip, one containing BL and the other one the APP), and that the library will be able to take care of this annoying use case.
Unfortunatelly, it throw the same error, (FW version error, bootloader is already up to date), and then stop, instead of continuing and flashing the app.
Don't you think the library should continue ?
Do you have a workaround for this issue ?

Right now the problem for the Android App to handle that is when we are pairing to the device, as it won't be able to keep the paired state (bc the update failed), and we shouldn't keep data in the database that the device has already the latest bootloader and should go right to the firmware.

Thank you !

philips77, 2018-09-17:
Hello!
So first of all, I would recommend using a single ZIP file. If you need to update only the app, you may use scopes:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/5cac2ce0e6f8fbf80f581862066ed2f56f58e3ec/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L269-L288
Coming back to the main topic. After you pair, do you have version numbers exposed?
Also, how about first updating the app? Why can't you just start updating the app after the first update fails? The device will switch back to normal more after a timeout, but you could scan for the device +1 address.

I think the lib could continue updating the app. The app's init packet contains the required SD ID, so in case the the app requires updated SD the update would still fail, there is no harm.

philips77, 2018-10-17:
I'll close the issue then. I opened a new one with your request.

closed 2018-10-17 by philips77

philips77, 2018-10-17:
Hi, have a look at the last changes. Secure DFU will now automatically try to send App when SD+BL failed, in the same connection.