nordicsemi/Android-DFU-Library #127: Firmware update crashes on LineageOS

josephtoles, 2018-10-10:
# TL;DR

Firmware update on LineageOS fails with a `java.lang.SecurityException`.

# Issue

I run a company that sells consumer devices using nRF52 chips. Users configure our devices via their cell phones. I'm trying to incorporate the Android-DFU-Library into our Android app so users can update firmware on the nRF52s running our products. I'm testing my app on three different phones.
- Samsung Galaxy S5 running Verizon's version of Samsung's stock ROM (Android 6.0.1)
- Samsung Galaxy S5 running LineageOS version 15.1 (Android 8.1.0 )
- Moto X Pure running LineageOS version 14.1 (Android 7.1.2)

Besides the ROMs, everything on these phones is identical. They are running exactly the same Android app. I am using the newest Android-DFU-Library, 1.7.0. The zip file I am attempting to update contains an application only. The zip file does not contain a SoftDevice nor a bootloader. I am performing a "buttonless" DFU.

On the Verizon phone, everything works great. The Android-DFU-Library udpates firmware as expected. `DfuProgressListener` calls `onDfuCompleted`, as expected.

On the two LineageOS phones, the Android-DFU-Library doesn't update firmware as expected. Instead, `DfuProgressListener` calls ` onError`, returning error code 4096. At the same time, the Android-DFU-Library writes a bunch of stuff to the logs that weren't present on the successful Verison DFU.

For the sake of completeness, here is the entire log.

```
10-09 19:10:32.545  9451  9676 I DfuImpl : Enabling indications...
10-09 19:10:32.546  9451  9676 D BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90003-f315-4f60-9fb8-838830daea50 enable: true
10-09 19:10:33.022   896   911 I ActivityManager: Waited long enough for: ServiceRecord{116353 u0 org.thoughtcrime.securesms/.service.KeyCachingService}
10-09 19:10:33.900  9451  9676 I DfuImpl : Sending Enter Bootloader (Op Code = 1)
10-09 19:10:33.936  4138  4296 D StreamDBQueue: Promised revision #45047 to queue a new op
10-09 19:10:33.936  4138  4296 D BaseStreamManager: Promised set @ r#45047 for (com.slightlyrobot.seabiscuit,tag=null,id=283,notifKey=0|com.slightlyrobot.seabiscuit|283|null|10122)
10-09 19:10:33.938  4138  4317 D DynamicBucketRankerImpl: StreamItemIdAndRevision{notifKey=0|com.slightlyrobot.seabiscuit|283|null|10122, packageName=com.slightlyrobot.seabiscuit, id=283, revision=45047, originalRevision=45045, updateCount=2} in bucket #3 (Ongoing)
10-09 19:10:33.938  4138  4317 D WatchStreamDB: com.google.android.clockwork.common.stream.TopLevelStreamItem@f18b85b ranked at index 0
10-09 19:10:33.939  4138  4317 D NewStreamAdapter: Publishing revision #45047 to adapter clients
10-09 19:10:34.252  9451  9676 I DfuImpl : Response received (Op Code = 1, Status = 1)
10-09 19:10:36.516   896  2508 D WifiStateMachine: updateCapabilities for config:SubEthafalse,false
10-09 19:10:38.442  2617  3074 W bt_btif : bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
10-09 19:10:38.442  2617  3074 W bt_btif : bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0008
10-09 19:10:38.442  2617  3074 W bt_btif : bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0008
10-09 19:10:38.443  2617  3074 W bt_btif : bta_gattc_conn_cback() - cif=6 connected=0 conn_id=6 reason=0x0008
10-09 19:10:38.443  2617  3074 W bt_btif : bta_gattc_conn_cback() - cif=7 connected=0 conn_id=7 reason=0x0008
10-09 19:10:38.443  2617  3074 W bt_btif : bta_gattc_conn_cback() - cif=8 connected=0 conn_id=8 reason=0x0008
10-09 19:10:38.443  2617  3074 W bt_btif : bta_gattc_conn_cback() - cif=9 connected=0 conn_id=9 reason=0x0008
10-09 19:10:38.443  2617  3074 I bt_btm_sec: btm_sec_disconnected clearing pending flag handle:7 reason:8
10-09 19:10:38.443  2617  3074 W bt_l2cap: L2CA_SetDesireRole() new:x0, disallow_switch:0
10-09 19:10:38.445  9451  9473 D BluetoothGatt: onClientConnectionState() - status=8 clientIf=5 device=FA:72:66:E5:BA:DA
10-09 19:10:38.445  9451  9472 D BluetoothGatt: onClientConnectionState() - status=8 clientIf=6 device=FA:72:66:E5:BA:DA
10-09 19:10:38.445  9451  9473 D sr_mobile:FirSer: Disconnected from GATT server.
10-09 19:10:38.445  9451  9464 D BluetoothGatt: onClientConnectionState() - status=8 clientIf=7 device=FA:72:66:E5:BA:DA
10-09 19:10:38.445  9451  9473 D BluetoothGatt: onClientConnectionState() - status=8 clientIf=9 device=FA:72:66:E5:BA:DA
10-09 19:10:38.446  9451  9473 W DfuBaseService: Target device disconnected with status: 8
10-09 19:10:38.448  2617  2715 E BluetoothRemoteDevices: state12newState1
10-09 19:10:38.451  9451  9676 I DfuBaseService: Cleaning up...
10-09 19:10:38.451  9451  9676 D BluetoothGatt: close()
10-09 19:10:38.451  9451  9676 D BluetoothGatt: unregisterApp() - mClientIf=9
10-09 19:10:38.455  2617  2617 D AvrcpBipRsp: onReceive
10-09 19:10:38.455  2617  2617 D AvrcpBipRsp: onReceive: android.bluetooth.device.action.ACL_DISCONNECTED
10-09 19:10:38.457  9451  9451 I DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
10-09 19:10:38.458  2617  2617 D BluetoothMapService: onReceive
10-09 19:10:38.458  2617  2617 D BluetoothMapService: onReceive: android.bluetooth.device.action.ACL_DISCONNECTED
10-09 19:10:38.459  2617  2617 E BluetoothMapService: Unexpected error!
10-09 19:10:38.460  9451  9676 I DfuImpl : Restarting to bootloader mode
10-09 19:10:38.470  2617  2617 D BluetoothPbapReceiver: PbapReceiver onReceive action = android.bluetooth.device.action.ACL_DISCONNECTED
10-09 19:10:38.472  2798  2798 I TrustAgent.Tracker: [BluetoothConnectionTracker] Bluetooth disconnect broadast for Slightly_Robot FA:72:66:E5:BA:DA
10-09 19:10:38.480  2617  2617 D BluetoothPbapReceiver: Calling start service with action = null
10-09 19:10:38.486  2617  2617 I BluetoothPbapReceiver: Exit - onReceive for intent:android.bluetooth.device.action.ACL_DISCONNECTED
10-09 19:10:38.487  2617  2617 D BluetoothPbapService: Enter - onStartCommand for service PBAP
10-09 19:10:38.487  2617  2617 D BluetoothPbapService: action: android.bluetooth.device.action.ACL_DISCONNECTED
10-09 19:10:38.487  2617  2617 D BluetoothPbapService: Exit - onStartCommand for service PBAP
10-09 19:10:38.489  9451  9473 D BluetoothLeScanner: onClientRegistered() - status=0 clientIf=9 mClientIf=0
10-09 19:10:38.493  9451  9473 W Binder  : Binder call failed.
10-09 19:10:38.493  9451  9473 W Binder  : java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.os.Parcel.readException(Parcel.java:1684)
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.os.Parcel.readException(Parcel.java:1637)
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:678)
[... 141 lines cut ...]
10-09 19:10:38.493  9451  9473 W Binder  : java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.os.Parcel.readException(Parcel.java:1684)
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.os.Parcel.readException(Parcel.java:1637)
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:678)
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:367)
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
10-09 19:10:38.493  9451  9473 W Binder  : 	at android.os.Binder.execTransact(Binder.java:565)
10-09 19:10:38.512  4334  4334 W SearchServiceCore: Abort, client detached.
```

Apparently the binder doesn't receive the appropriate `ACCESS_COARSE_LOCATION` or `ACCESS_FINE_LOCATION` required for a Bluetooth scan. This is unusual because my app includes both of these permissions in its `AndroidManifest.xml`. It looks as if Android-DFU-Library's binder isn't getting the appropriate permissions on LineageOS, despite my app having these permissions. I hope there's something simple I'm doing wrong and this isn't an bug with the Android-DFU-Library itself.

# Further Research

If I understand correctly, the nRF Connect Android app uses the Android-DFU-Library. So what happens when I try to do a firmware update with the nRF Connect Android app instead?

[... cut ...]  device reverts to its original applicatoin.
- **Moto X Pure running LineageOS version 14.1 (Android 7.1.2)**
Pressing "DFU" gets to "Connecting..." and then opens up a new tab labeled DFUTARG. This new tab updates firmware just fine.

Is this all intended behavior? Is there a workaround I can use to get the Android-DFU-Library working on LineageOS? If there is no workaround, is there any way we can tell in advance whether the Android-DFU-Library will work properly on a particular Android phone?

philips77, 2018-10-10:
Location permissions are a Runtime permission since Android 6. Do you request it from the user in your app? Try enabling them manually is Android settings > Applications > your app > Permissions > Enable location.

philips77, 2018-10-10:
Restarting Bluetooth on the 2nd phone was due to a hard error on Android. It tries to recover Bluetooth adapter by restarting it.

philips77, 2018-10-10:
The new tab will open if you have "Show incoming connections" enabled in nRF Connect settings > Connectivity. That's because the bootloader has a different address, so from app's view is a new device. But on the old tab you should see the dfu progress, if you switch back to it. You may close the new tab.

josephtoles, 2018-10-15:
Thank you for your help. I had indeed failed to enable the runtime location permission on my newer phones. However, now I'm running into what I think is a different error.

Ordinarily I wouldn't bother you, but I've been working full-time for a week to get Android DFU into my app. I've tried to solve the problem in lots of different ways. I could use any ideas you have to offer. The new issue is a 4109 error followed by a 4096 error. I can open the following problem as a different Issue on GitHub if you'd prefer.

# Samsung Galaxy S5 with Verizon's stock ROM

Once again, the Samsung Galaxy S5 running Verizon's version of Samsung's stock ROM (Android 6.0.1) 
updates firmware just fine.

The setup begins with the nRF52 unbonded, unconnected and advertising. The phone does not have any connections saved. Then I perform a BLE scan programmatically in my Android app's Java code. My programmatic BLE scan detects advertising nRF52 chip. Then I call `DfuServiceInitiator.start(...)`. The Android-DFU-Library updates firmware just fine. `DfuProgressListener` calls the following callbacks.

```
onDeviceConnected
onDfuProcessStarting
onEnablingDfuMode
onDeviceConnecting
onDeviceConnected (again)
onDfuProcessStarting (again)
onDfuProcessStarted
onProgressChanged 0-100
onDeviceDisconnecting
onDeviceDisconnected
onDfuCompleted
```

# Moto X running Lineage OS

I'm running exactly the same mobile app on the Moto X. I've made sure to grant the runtime location permission you mentioned 5 days ago via the phone's settings. However, when I perform a scan, I get a different result.

```
onDeviceConnected
onDfuProcessStarting
onEnablingDfuMode
onDeviceConnecting
onDeviceConnected (again)
onDfuProcessStarting (again)
onDeviceDisconnecting
onDeviceDisconnected
onError 4109
onDeviceDisconnected
onError 4096
```

# Analysis

The two logs begin to differ after `onDfuProcessStarting` is called for the second time. Does anything differ in the tracebacks here? Yes. In particular, the CRC Checksums don't match.

## Samsung Checksum

Here's a snippet from the Verizon Samsung phone's logs.

```10-15 00:41:39.402  6569  6579 I DfuBaseService: Services discovered
10-15 00:41:39.402  6569 12562 W DfuImpl : Secure DFU bootloader found
10-15 00:41:39.402  6569  6569 D sr_mobile:MainActivity: DFU onDfuProcessStarting
10-15 00:41:40.422  6569 12562 I DfuImpl : Requesting MTU = 517
10-15 00:41:40.462  6569  6579 I DfuImpl : MTU changed to: 247
10-15 00:41:40.462  6569 12562 I DfuImpl : Enabling notifications...
10-15 00:41:41.492  6569 12562 I DfuImpl : Setting object to Command (Op Code = 6, Type = 1)
10-15 00:41:41.562  6569 12562 I DfuImpl : Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
10-15 00:41:41.562  6569 12562 I DfuImpl : Sending the number of packets before notifications (Op Code = 2, Value = 0)
10-15 00:41:41.582  6569 12562 I DfuImpl : Creating Init packet object (Op Code = 1, Type = 1, Size = 136)
10-15 00:41:41.622  6569 12562 I DfuImpl : Sending 136 bytes of init packet...
10-15 00:41:41.622  6569 12562 I DfuImpl : Sending init packet (Value = 12-85-01-0A-3F-08-01-12-3B-08-FF-01-10-34-1A-02-A8-01-20-00-28-00-30-00-38-B4-A3-04-42-24-08-03-12-20-5A-F4-27-A6-5D-36-4E-D6-23-59-87-72-40-F8-F3-29-8E-86-AF-F9-70-63-58-01-DE-7E-0E-56-2E-8B-46-4E-48-00-10-00-1A-40-07-BD-96-2B-FC-D5-1C-7D-8A-F3-EA-D2-67-D7-5E-92-DE-A0-6D-AB-07-19-79-92-C3-5D-FF-0A-20-66-30-9F-35-C1-FC-1D-6C-9C-59-C5-7C-DC-F6-26-5B-8C-F5-94-7D-E8-CA-3C-72-AD-E7-DB-82-76-BB-BF-6D-9D-35-45)
10-15 00:41:41.622  6569 12562 I DfuImpl : Sending Calculate Checksum command (Op Code = 3)
10-15 00:41:41.662  6569 12562 I DfuImpl : Checksum received (Offset = 136, CRC = 6D55674D)
10-15 00:41:41.662  6569 12562 I DfuImpl : Executing init packet (Op Code = 4)
10-15 00:41:41.822  6569 12562 I DfuImpl : Setting object to Data (Op Code = 6, Type = 2)
10-15 00:41:41.852  6569 12562 I DfuImpl : Data object info received (Max size = 4096, Offset = 0, CRC = 00000000)
10-15 00:41:41.862  6569  6569 D sr_mobile:MainActivity: DFU onDfuProcessStarted
10-15 00:41:41.882  6569 12562 I DfuImpl : Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/18)
[... 47 lines cut ...]
10-15 00:33:37.600  9222  9270 I DfuImpl : Sending 136 bytes of init packet...
10-15 00:33:37.600  9222  9270 I DfuImpl : Sending init packet (Value = 12-85-01-0A-3F-08-01-12-3B-08-FF-01-10-34-1A-02-A8-01-20-00-28-00-30-00-38-B4-A3-04-42-24-08-03-12-20-5A-F4-27-A6-5D-36-4E-D6-23-59-87-72-40-F8-F3-29-8E-86-AF-F9-70-63-58-01-DE-7E-0E-56-2E-8B-46-4E-48-00-10-00-1A-40-07-BD-96-2B-FC-D5-1C-7D-8A-F3-EA-D2-67-D7-5E-92-DE-A0-6D-AB-07-19-79-92-C3-5D-FF-0A-20-66-30-9F-35-C1-FC-1D-6C-9C-59-C5-7C-DC-F6-26-5B-8C-F5-94-7D-E8-CA-3C-72-AD-E7-DB-82-76-BB-BF-6D-9D-35-45)
10-15 00:33:37.606  9222  9270 I DfuImpl : Sending Calculate Checksum command (Op Code = 3)
10-15 00:33:37.630  9222  9270 I DfuImpl : Checksum received (Offset = 0, CRC = 00000000)
10-15 00:33:37.630  9222  9270 I DfuImpl : CRC does not match! Retrying...(3/3)
10-15 00:33:37.630  9222  9270 I DfuImpl : Creating Init packet object (Op Code = 1, Type = 1, Size = 136)
10-15 00:33:37.659  9222  9270 I DfuImpl : Sending 136 bytes of init packet...
10-15 00:33:37.659  9222  9270 I DfuImpl : Sending init packet (Value = 12-85-01-0A-3F-08-01-12-3B-08-FF-01-10-34-1A-02-A8-01-20-00-28-00-30-00-38-B4-A3-04-42-24-08-03-12-20-5A-F4-27-A6-5D-36-4E-D6-23-59-87-72-40-F8-F3-29-8E-86-AF-F9-70-63-58-01-DE-7E-0E-56-2E-8B-46-4E-48-00-10-00-1A-40-07-BD-96-2B-F [... cut ...]
10-15 00:33:38.363  9222  9222 I DfuBaseService: DFU service destroyed
10-15 00:33:38.744  2794  2794 I TrustAgent.Tracker: [BluetoothConnectionTracker] Bluetooth disconnect broadast for DfuTarg FA:72:66:E5:BA:DB
```

## Checksum Analysis

As you can see, the Checksum succeeds on the Verizon Samsung phone but fails on the LineageOS Moto X phone. Do you know why this might be? Both phones are running exactly the same Android mobile app and I haven't done anything unusual to configure either of their Bluetooth systems.

josephtoles, 2018-10-15:
There's only one thing weird I'm doing. My nRF52 firmware application always erases the device's bond information when it starts up. (Our hardware doesn't have any user-facing buttons including power buttons so I used this as a workaround to allow the device to be paired to new devices.) I don't think the bond-erasing weirdness should be causing my CRC error because this DFU doesn't involve bonding. It is possible there is an interaction I am overlooking.

closed 2018-10-15 by philips77