nordicsemi/Android-DFU-Library #129: NullPointerException: BluetoothGattDescriptor is null when BaseDfulImpl.enableCCCD() is called

alixwar, 2018-10-15:
We've seen crash reports for our app that bundles `no.nordicsemi.android:dfu:1.6.1`

Details from the crash report:
Brand: asus
Model: MeMO Pad 7 (ME572C)
Orientation: Portrait
RAM free: 656.18 MB
Disk free: 7.15 GB

Operating System
Version: 5.0.1
Orientation: Portrait
Rooted: No

```
java.lang.NullPointerException
Attempt to invoke virtual method 'byte[] android.bluetooth.BluetoothGattDescriptor.getValue()' on a null object reference
no.nordicsemi.android.dfu.BaseDfuImpl.enableCCCD (BaseDfuImpl.java:351)
no.nordicsemi.android.dfu.ButtonlessDfuImpl.performDfu (ButtonlessDfuImpl.java:70)
no.nordicsemi.android.dfu.ButtonlessDfuWithoutBondSharingImpl.performDfu (ButtonlessDfuWithoutBondSharingImpl.java:76)
no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent (DfuBaseService.java:1173)
android.app.IntentService$ServiceHandler.handleMessage (IntentService.java:65)
```

philips77, 2018-10-17:
Will be released in v1.7.1.

closed 2018-10-17 by philips77

philips77, 2018-10-29:
Released in 1.8.0.