nordicsemi/Android-DFU-Library #143: Fatal Exception: java.lang.NullPointerException Attempt to get length of null array

alixwar, 2018-10-25:
A crash was reported in our app that bundles version 1.7.0 of the DFU library. Details below:

Device
Brand: samsung
Model: SM-G950F
Orientation: Portrait
RAM free: 1.28 GB
Disk free: 47.78 GB

Operating System
Version: 8.0.0
Orientation: Portrait
Rooted: No

```
no.nordicsemi.android.dfu.internal.ArchiveInputStream.read (ArchiveInputStream.java:338)
no.nordicsemi.android.dfu.SecureDfuImpl.sendFirmware (SecureDfuImpl.java:461)
no.nordicsemi.android.dfu.SecureDfuImpl.performDfu (SecureDfuImpl.java:221)
no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent (DfuBaseService.java:1186)
android.app.IntentService$ServiceHandler.handleMessage (IntentService.java:68)
android.os.Handler.dispatchMessage (Handler.java:105)
android.os.Looper.loop (Looper.java:164)
android.os.HandlerThread.run (HandlerThread.java:65)
```

philips77, 2018-10-25:
Duplicate of #97. It has been fixed by #126 and will be released in 1.8.0.

philips77, 2018-10-29:
Released.

LoPoBo, 2018-10-29:
The stacktrace of 1.8.0:

    java.lang.NullPointerException: Attempt to get length of null array
        at no.nordicsemi.android.dfu.internal.ArchiveInputStream.read(ArchiveInputStream.java:338)
        at no.nordicsemi.android.dfu.SecureDfuImpl.sendFirmware(SecureDfuImpl.java:499)
        at no.nordicsemi.android.dfu.SecureDfuImpl.performDfu(SecureDfuImpl.java:257)
        at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1236)
        at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:68)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)

philips77, 2018-10-29:
Could you paste some LogCat logs from before the crash occurs?

LoPoBo, 2018-10-29:
Sure. Here it is:

    10-29 14:20:26.299 2380-2380/my.app I/DfuBaseService: DFU service created. Version: 1.8.0
    10-29 14:20:26.408 2380-2380/my.app I/DfuService: [DFU] Firmware file opened successfully
    10-29 14:20:26.410 2380-2531/my.app I/DfuBaseService: Connecting to the device...
    10-29 14:20:26.767 2380-2380/my.app I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
    10-29 14:20:26.773 2380-2393/my.app I/DfuBaseService: Connected to GATT server
    10-29 14:20:26.777 2380-2380/my.app I/DfuService: [DFU] Connected to F1:77:EB:BC:F8:48
    10-29 14:20:26.788 2380-2393/my.app I/DfuBaseService: Attempting to start service discovery... succeed
    10-29 14:20:27.784 2380-2393/my.app I/DfuBaseService: Services discovered
    10-29 14:20:27.789 2380-2380/my.app I/DfuService: [DFU] Services discovered
    10-29 14:20:27.793 2380-2531/my.app W/DfuImpl: Secure DFU bootloader found
    10-29 14:20:28.814 2380-2531/my.app I/DfuImpl: Requesting MTU = 517
    10-29 14:20:28.859 2380-2393/my.app I/DfuImpl: MTU changed to: 23
    10-29 14:20:28.860 2380-2531/my.app I/DfuImpl: Enabling notifications...
    10-29 14:20:28.861 2380-2380/my.app I/DfuService: [DFU] MTU changed to: 23
    10-29 14:20:28.916 2380-2380/my.app I/DfuService: [DFU] Data written to descr.8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-00
    10-29 14:20:28.919 2380-2380/my.app I/DfuService: [DFU] Notifications enabled
    10-29 14:20:29.918 2380-2531/my.app I/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
    10-29 14:20:29.966 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 06-01
    10-29 14:20:29.967 2380-2531/my.app I/DfuImpl: Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
    10-29 14:20:29.967 2380-2380/my.app I/DfuService: [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-06-01-00-01-00-00-00-00-00-00-00-00-00-00
    10-29 14:20:29.969 2380-2531/my.app I/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 0)
    10-29 14:20:29.969 2380-2380/my.app I/DfuService: [DFU] Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
    10-29 14:20:30.004 2380-2380/my.app I/DfuService: [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-02-01
    10-29 14:20:30.005 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-02-01
    10-29 14:20:30.009 2380-2531/my.app I/DfuImpl: Creating Init packet object (Op Code = 1, Type = 1, Size = 136)
    10-29 14:20:30.010 2380-2380/my.app I/DfuService: [DFU] Packet Receipt Notif disabled (Op Code = 2, Value = 0)
    10-29 14:20:30.059 2380-2380/my.app I/DfuService: [DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
    10-29 14:20:30.060 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
    10-29 14:20:30.068 2380-2531/my.app I/DfuImpl: Sending 136 bytes of init packet...
        Sending init packet (Value = 12-85-01-0A-3F-08-01-12-3B-08-8A-0A-10-34-1A-02-98-01-20-00)
    10-29 14:20:30.068 2380-2380/my.app I/DfuService: [DFU] Command object created
    10-29 14:20:30.078 2380-2531/my.app I/DfuImpl: Sending init packet (Value = 28-00-30-00-38-E4-D3-0B-42-24-08-03-12-20-CF-08-18-D6-8A-DF)
    10-29 14:20:30.079 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 12-85-01-0A-3F-08-01-12-3B-08-8A-0A-10-34-1A-02-98-01-20-00
    10-29 14:20:30.084 2380-2531/my.app I/DfuImpl: Sending init packet (Value = 6B-C7-8E-1F-6E-58-9F-76-21-41-09-56-63-CE-6A-4E-35-FF-11-95)
    10-29 14:20:30.084 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 28-00-30-00-38-E4-D3-0B-42-24-08-03-12-20-CF-08-18-D6-8A-DF
    10-29 14:20:30.090 2380-2531/my.app I/DfuImpl: Sending init packet (Value = AE-3E-12-95-DA-82-48-00-10-00-1A-40-2A-C3-91-F9-1D-03-AC-84)
    10-29 14:20:30.091 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 6B-C7-8E-1F-6E-58-9F-76-21-41-09-56-63-CE-6A-4E-35-FF-11-95
    10-29 14:20:30.101 2380-2531/my.app I/DfuImpl: Sending init packet (Value = 8A-2F-9E-79-B4-6B-9C-84-BE-61-70-D8-AF-6B-C7-FE-7B-35-BE-2F)
    10-29 14:20:30.102 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): AE-3E-12-95-DA-82-48-00-10-00-1A-40-2A-C3-91-F9-1D-03-AC-84
    10-29 14:20:30.109 2380-2531/my.app I/DfuImpl: Sending init packet (Value = 02-FC-0B-9E-D6-58-1E-62-46-26-B1-7C-2C-FD-11-CF-EF-66-75-71)
    10-29 14:20:30.115 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 8A-2F-9E-79-B4-6B-9C-84-BE-61-70-D8-AF-6B-C7-FE-7B-35-BE-2F
    10-29 14:20:30.123 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 02-FC-0B-9E-D6-58-1E-62-46-26-B1-7C-2C-FD-11-CF-EF-66-75-71
    10-29 14:20:30.129 2380-2531/my.app I/DfuImpl: Sending init packet (Value = 21-5A-6C-CA-2A-2C-DE-B6-C4-52-01-E7-80-CC-7F-37)
    10-29 14:20:30.136 2380-2380/my.app I/DfuService: [DFU] Data written to 8ec90002-f315-4f60-9fb8-838830daea50, value (0x): 21-5A-6C-CA-2A-2C-DE-B6-C4-52-01-E7-80-CC-7F-37
[... cut ...]
            at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:68)
            at android.os.Handler.dispatchMessage(Handler.java:105)
            at android.os.Looper.loop(Looper.java:164)
            at android.os.HandlerThread.run(HandlerThread.java:65)
    10-29 14:21:45.040 2380-2419/my.app I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
    10-29 14:21:45.041 2380-2419/my.app I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
    10-29 14:22:40.478 2380-2380/my.app I/DfuBaseService: DFU service destroyed

LoPoBo, 2018-10-30:
I will investigate it

LoPoBo, 2018-11-05:
The problem is that when `SecureDfuImpl.sendFirmware(BluetoothGatt)` tries to resume it tries to read all the executed bytes (L495) from the stream but since the stream is reused (and not reset to the beginning of the file) it instead reads the rest of the file (and thus setting `currentSource` to `null`). On the next read (L499) the NPE occurs.

I'm not sure there is a simple solution to this.

There also seem to be some confusion/mismatch in how the `mark`/`reset` methods are used in the code. In some places it seems like it is assumed that `reset` resets the stream to the beginning.

LoPoBo, 2018-11-05:
In several places in `SecureDfuImpl` the field `mFirmwareStream` is cast to `ArchiveInputStream` with no apparent type check. Can it be assumed that the `mFirmwareStream` is always an instance of `ArchiveInputStream` in `SecureDfuImpl`? If so, I could add a "full reset" method in `ArchiveInputStream` which could be called before the read on L495.

philips77, 2018-12-04:
Hello @LoPoBo,
Sorry for the very long delay. 
First of all, indeed the Secure DFU Impl assumes that the ZIP file and `AchiveInputStream` are used. Other methods are deprecated and I don't think I'll change that, as it's much easier just to create the ZIP file then modify the Secure DFU Impl to accept *bin* and *dat* files separately. But thank you for pointing that out. This should be documented.
Secondly, restarting the service is problematic. As you've noticed, the service `resets()` the stream before uploading, but as it is reusing the same `InputStream` object (in that case `ArchiveInputStream`), and in the previous upload it put some marks there, the reset won't do "full reset". I'll try to fix it now.

philips77, 2018-12-04:
@LoPoBo, the commit added `fullReset()` method in the `ArchiveInputStream`. It is called every time the DFU is started. This resets the byte counter to 0 for the current stream, so if the disconnection happened during App update, after SD+BL were already updated, the App stream will be reset to 0, not the SB+BL. However, I couldn't test it properly, so I'll wait with publishing until you test it. Thank you!

closed 2019-01-29 by philips77

LoPoBo, 2019-02-04:
Hi again. Unfortunately I had changed project and thus were unable to test your fix. Sorry for not replying earlier!