nordicsemi/Android-DFU-Library #145: DFU bricks the device when HEX is used

philips77, 2018-11-01:
DFU Library version 1.8.0.
When HEX + BIN are used, the 1.8.0 doesn't send the whole HEX file and the device gets bricked. 

```
nRF Connect, 2018-10-31
[...]
V	14:02:00.920	[DFU] DFU service started
V	14:02:00.920	[DFU] Opening file...
I	14:02:00.948	[DFU] Firmware file opened successfully
V	14:02:00.948	[DFU] Connecting to DFU target...
D	14:02:00.948	[DFU] gatt = device.connectGatt(autoConnect = false)
I	14:02:01.083	[DFU] Connected to E5:BD:95:9A:5C:DC
V	14:02:01.083	[DFU] Discovering services...
D	14:02:01.083	[DFU] gatt.discoverServices()
I	14:02:01.083	[DFU] Services discovered
D	14:02:01.104	[DFU] wait(1000)
A	14:02:02.103	[DFU] DFU Version characteristic not found
W	14:02:02.104	[DFU] Application with buttonless update found
V	14:02:02.104	[DFU] Jumping to the DFU Bootloader...
V	14:02:02.104	[DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D	14:02:02.104	[DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D	14:02:02.104	[DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I	14:02:02.187	[DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V	14:02:02.187	[DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A	14:02:02.187	[DFU] Notifications enabled
D	14:02:02.187	[DFU] wait(1000)
V	14:02:03.196	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:02:03.197	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
D	14:02:23.249	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	14:02:23.249	[Server] Device disconnected
D	14:02:23.270	[Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
A	14:02:23.270	[DFU] Jump to bootloader sent (Op Code = 1, Upload Mode = 4)
I	14:02:23.271	[DFU] Disconnected by the remote device
D	14:02:23.271	[DFU] gatt.refresh() (hidden)
D	14:02:23.271	[DFU] gatt.close()
E	14:02:23.275	Error 8 (0x8): GATT CONN TIMEOUT
I	14:02:23.275	Disconnected
V	14:02:23.347	[DFU] Scanning for the DFU Bootloader...
D	14:02:23.348	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	14:02:23.370	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
I	14:02:23.518	[DFU] DFU Bootloader found with address E5:BD:95:9A:5C:DC
V	14:02:23.535	[DFU] DFU service started
I	14:02:23.535	[DFU] Firmware file opened successfully
D	14:02:23.535	[DFU] wait(1000)
D	14:02:24.564	[DFU] wait(1000)
V	14:02:25.529	[DFU] Connecting to DFU target...
D	14:02:25.579	[DFU] gatt = device.connectGatt(autoConnect = false)
D	14:02:25.856	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	14:02:25.857	[Server] Device with address XX:XX:XX:XX:XX:XX connected
D	14:02:25.918	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	14:02:25.947	[DFU] Connected to XX:XX:XX:XX:XX:XX
V	14:02:25.947	[DFU] Discovering services...
D	14:02:25.947	[DFU] gatt.discoverServices()
D	14:02:25.947	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	14:02:26.967	[DFU] Services discovered
D	14:02:26.968	[DFU] wait(1000)
V	14:02:27.955	[DFU] Reading DFU version number...
D	14:02:27.955	[DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I	14:02:28.007	[DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 04-00
A	14:02:28.007	[DFU] Version number read: 0.4
D	14:02:28.008	[DFU] wait(1000)
V	14:02:29.028	[DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D	14:02:29.028	[DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D	14:02:29.028	[DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I	14:02:29.077	[DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V	14:02:29.077	[DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A	14:02:29.077	[DFU] Notifications enabled
D	14:02:29.077	[DFU] wait(1000)
V	14:02:30.083	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:02:30.083	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I	14:02:30.132	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 01-04
A	14:02:30.132	[DFU] DFU Start sent (Op Code = 1, Upload Mode = 4)
V	14:02:30.132	[DFU] Writing to characteristic 00001532-1212-efde-1523-785feabcd123
D	14:02:30.132	[DFU] gatt.writeCharacteristic(00001532-1212-efde-1523-785feabcd123)
I	14:02:30.132	[DFU] Data written to 00001532-1212-efde-1523-785feabcd123, value (0x): 00-00-00-00-00-00-00-00-8C-EE-00-00
A	14:02:30.132	[DFU] Firmware image size sent (0b, 0b, 61068b)
I	14:02:32.093	[DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-01-01
A	14:02:32.093	[DFU] Response received (Op Code = 1 Status = 1)
A	14:02:32.093	[DFU] Writing Initialize DFU Parameters...
V	14:02:32.093	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:02:32.093	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I	14:02:32.160	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 02-00
V	14:02:32.160	[DFU] Writing to characteristic 00001532-1212-efde-1523-785feabcd123
D	14:02:32.160	[DFU] gatt.writeCharacteristic(00001532-1212-efde-1523-785feabcd123)
I	14:02:32.160	[DFU] Data written to 00001532-1212-efde-1523-785feabcd123, value (0x): FF-FF-FF-FF-00-00-00-00-01-00-FE-FF-E3-31
V	14:02:32.160	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:02:32.160	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I	14:02:32.210	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 02-01
A	14:02:32.210	[DFU] Initialize DFU Parameters completed
I	14:02:32.210	[DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-02-01
A	14:02:32.210	[DFU] Response received (Op Code = 2, Status = 1)
V	14:02:32.210	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:02:32.210	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I	14:02:32.278	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 08-0A-00
A	14:02:32.278	[DFU] Packet Receipt Notif Req (Op Code = 8) sent (Value = 10)
V	14:02:32.278	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:02:32.278	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I	14:02:32.346	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 03
A	14:02:32.346	[DFU] Receive Firmware Image request sent
A	14:02:32.349	[DFU] Uploading firmware...
V	14:02:32.349	[DFU] Sending firmware to characteristic 00001532-1212-efde-1523-785feabcd123...
I	14:03:18.659	[DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-03-01
A	14:03:18.659	[DFU] Response received (Op Code = 3, Status = 1)
A	14:03:18.659	[DFU] Upload completed in 46239 ms
V	14:03:18.659	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:03:18.659	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I	14:03:18.710	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 04
A	14:03:18.710	[DFU] Validate request sent
I	14:03:18.710	[DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-04-01
A	14:03:18.710	[DFU] Response received (Op Code = 4, Status = 1)
V	14:03:18.711	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:03:18.712	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I	14:03:18.776	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 05
A	14:03:18.776	[DFU] Activate and Reset request sent
D	14:03:18.934	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	14:03:18.934	[Server] Device disconnected
I	14:03:18.963	[DFU] Disconnected by the remote device
D	14:03:18.963	[DFU] gatt.refresh() (hidden)
D	14:03:18.963	[DFU] gatt.close()
D	14:03:18.963	[DFU] wait(1400)
D	14:03:18.964	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	14:03:19.024	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	14:03:20.387	gatt.close()
D	14:03:20.389	wait(200)
V	14:03:20.593	Connecting to XX:XX:XX:XX:XX:XX...
D	14:03:20.593	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D	14:03:25.714	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	14:03:25.714	[Server] Device disconnected
D	14:03:25.734	[Callback] Connection state changed with status: 133 and new state: DISCONNECTED (0)
E	14:03:25.734	Error 133 (0x85): GATT ERROR
I	14:03:25.734	Disconnected
```

philips77, 2018-11-15:
Hi, I'm still unable to reproduce this issue...

dlukashev, 2018-12-09:
We got many Atmotube devices bricked with hex app and 1.8.0. And need to replace production devices for our own expenses... There is no issue with 1.6.1 version. Haven't checked versions in between.

philips77, 2018-12-10:
Can you help me repdoducing the issue? What file format are you using, hex, bin or zip? What sdk? Do you have any logs?

dlukashev, 2018-12-10:
Hi. These are quite old devices (but we have 2000 of them sold). Softdevice version 7.3.0. 
We are using hex (application only). The problem is reproduced with Android SDK v.1.8.0. I can catch Android logs - but there are not so much useful information. From Android app the process is completed without errors.
- device moved to bootloader
- hex uploaded
- device does not start

we always used DfuService.TYPE_APPLICATION and setBinOrHex(type, path); to update these old devices

final DfuServiceInitiator starter = new DfuServiceInitiator(macAddress)
                .setDeviceName(finalName)
                .setKeepBond(false)
                .setForceDfu(false)
                .setDisableNotification(true)
                .setPacketsReceiptNotificationsEnabled(Build.VERSION.SDK_INT < Build.VERSION_CODES.M)
                .setPacketsReceiptNotificationsValue(DfuServiceInitiator.DEFAULT_PRN_VALUE)
                .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
starter.setBinOrHex(type, path);
starter.start(TheApp.getAppContext(), DfuService.class);

With 1.6.1 this code working fine. With 1.8.0 it bricks the device.
I can catch any logs you need or try to modify some DfuServiceInitiator parameters if required

philips77, 2018-12-10:
That should be enough. Thank you! I'm on it now.

philips77, 2018-12-10:
I believe I found the bug. Recently, as a fix to#131 and #148, the stream is initially being reset. There was a bug in the hex stream implementation in `reset()` and reading bytes after the stream was reset was throwing some garbage at the beginning. This should be fixed now. I'll release 1.8.1 immediately.

philips77, 2018-12-10:
Released in 1.8.1. nRF Connect with this version will be released shortly.

closed 2018-12-10 by philips77

dlukashev, 2018-12-12:
Thank you for quick fix. I can confirm 1.8.1 fixes this issue