nordicsemi/Android-DFU-Library #149: Optional use of DfuBaseService

miketrewartha, 2018-11-27:
First of all, thank you folks for a fantastic library. Adding DFU functionality to our app was a snap because of it.

I do have one issue with it, though. I totally understand how the `DfuBaseService` class allows users hook into the library very quickly. It's incredibly helpful for adding DFU functionality in a flash. On the flip side, I think it can be pretty heavy-handed to demand usage of `DfuBaseService`. It would be great to have an option: either subclass `DfuBaseService` and follow the existing paradigm *or* compose DFU functionality (provided by the library) into whatever mechanism I wanted (my own custom foreground `Service` that builds its own notifications, for example). The latter would require me to do more heavy lifting, but also allow me to tailor the experience to my app.

Any thoughts on this approach?

philips77, 2018-11-28:
I understand your point. For sure that would be useful. That's actually the approach we use in mcumgr based updates (on Zephyr): https://github.com/runtimeco/mcumgr-android, where you just create a `FirmwareUpdateManager` and call `.start(..)`, but you have to implement it in a service or viewmodel on your own.
I'll think about making similar thing in DFU. The problem is that we are lacking the resources (aka people), so it will be hard to fit it soon. But we accept PRs!

miketrewartha, 2018-12-08:
I'm swamped right now, but if I get some free time, I'll take a stab at a PR. Thanks @philips77

closed 2018-12-08 by miketrewartha