nordicsemi/Android-DFU-Library #151: DFU mode connection problem.

choa0707, 2018-12-02:
Hi, I've tried using this library in an Android project, but some device can't enter the DFU mode.
So, I tried it various way such as reconnect device. That devices with problem can enter DFU mode after more than 3 minutes. It is not normal case. It seems that a lot of similar problems have occurred. I'm wondering if any further updates to this issue have been or are in progress.

philips77, 2018-12-03:
Hi @choa0707,
Could you provide the LogCat logs? Enable them in your DfuSrevice by returning *true* from `isDebug()`, like here: https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/dfu/DfuService.java#L52

ohjunki, 2018-12-19:
HI. I have the same issue. I have tested many kinds of Android smartphone. Most of the test phone entered the DFU mode within 3 minutes. However, some smartphones had a problem, which phones are LG V10 and LG X401. My project was tested with DFU version 1.8.0.

**First LG V10**
it never enters DFU mode or needs a very long time to enter DFU mode almost 10~20minute. There is no error Toast, which I mean GATT ERROR or DISCONNECT.
The following log always repeat without end.
>12-19 19:56:35.071 27348-27348/com.dearpeople.divecomputer I/ViewRootImpl: ViewRoot's Touch Event : ACTION_DOWN
12-19 19:56:35.171 27348-27348/com.dearpeople.divecomputer I/ViewRootImpl: ViewRoot's Touch Event : ACTION_UP
12-19 19:56:35.181 27348-27348/com.dearpeople.divecomputer D/BluetoothAdapter: getBleQmState() : state = 110
12-19 19:56:35.181 27348-27348/com.dearpeople.divecomputer D/BluetoothGatt: close() : called...   getState() = 12, getBleQmState() = 110
12-19 19:56:35.181 27348-27348/com.dearpeople.divecomputer D/BluetoothGatt: close() : return_case = false
    unregisterApp() - mClientIf=7
12-19 19:56:40.301 27348-27348/com.dearpeople.divecomputer I/DfuBaseService: DFU service created. Version: 1.8.0
12-19 19:56:40.461 27348-27630/com.dearpeople.divecomputer I/DfuBaseService: Connecting to the device...
12-19 19:56:40.461 27348-27630/com.dearpeople.divecomputer D/BluetoothGatt: connect() - device: C0:A2:2F:C6:F8:CE, auto: false
    registerApp()
    registerApp() - UUID=1cd76bbe-4e0b-4a70-a70c-3be62357422f
12-19 19:56:40.461 27348-27366/com.dearpeople.divecomputer D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
12-19 19:56:40.461 27348-27367/com.dearpeople.divecomputer D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=C0:A2:2F:C6:F8:CE
12-19 19:56:40.461 27348-27367/com.dearpeople.divecomputer I/DfuBaseService: Connected to GATT server
12-19 19:56:40.461 27348-27367/com.dearpeople.divecomputer D/BluetoothGatt: discoverServices() - device: C0:A2:2F:C6:F8:CE
12-19 19:56:40.461 27348-27367/com.dearpeople.divecomputer I/DfuBaseService: Attempting to start service discovery... succeed
12-19 19:56:40.461 27348-27366/com.dearpeople.divecomputer D/BluetoothGatt: onSearchComplete() = Device=C0:A2:2F:C6:F8:CE Status=0
12-19 19:56:40.471 27348-27366/com.dearpeople.divecomputer I/DfuBaseService: Services discovered
12-19 19:56:41.491 27348-27630/com.dearpeople.divecomputer I/DfuImpl: Reading DFU version number...
12-19 19:56:41.861 27348-27630/com.dearpeople.divecomputer I/DfuImpl: Version number read: 0.1 -> Application with Legacy buttonless update from SDK 7.0 or newer
12-19 19:56:41.861 27348-27630/com.dearpeople.divecomputer W/DfuImpl: Application with legacy buttonless update found
12-19 19:56:41.861 27348-27630/com.dearpeople.divecomputer I/DfuImpl: Enabling notifications...
12-19 19:56:41.861 27348-27630/com.dearpeople.divecomputer D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
12-19 19:56:43.281 27348-27630/com.dearpeople.divecomputer I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
12-19 19:56:44.841 27348-27366/com.dearpeople.divecomputer D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=C0:A2:2F:C6:F8:CE
12-19 19:56:44.841 27348-27366/com.dearpeople.divecomputer D/BluetoothListener: BLE Disconnected!!
    reconnect try
    Start BLE
12-19 19:56:44.851 27348-27507/com.dearpeople.divecomputer D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=C0:A2:2F:C6:F8:CE
12-19 19:56:44.851 27348-27507/com.dearpeople.divecomputer I/DfuBaseService: Disconnected from GATT server
12-19 19:56:44.851 27348-27630/com.dearpeople.divecomputer D/BluetoothGatt: refresh() - device: C0:A2:2F:C6:F8:CE
12-19 19:56:44.851 27348-27348/com.dearpeople.divecomputer I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
12-19 19:56:44.861 27348-27630/com.dearpeople.divecomputer I/DfuBaseService: Refreshing result: true
    Cleaning up...
12-19 19:56:44.871 27348-27630/com.dearpeople.divecomputer D/BluetoothAdapter: getBleQmState() : state = 110
12-19 19:56:44.871 27348-27630/com.dearpeople.divecomputer D/BluetoothGatt: close() : called...   getState() = 12, getBleQmState() = 110
12-19 19:56:44.871 27348-27630/com.dearpeople.divecomputer D/BluetoothGatt: close() : return_case = false
    unregisterApp() - mClientIf=7
12-19 19:56:44.871 27348-27630/com.dearpeople.divecomputer I/DfuImpl: Starting service that will connect to the DFU bootloader
12-19 19:56:44.871 27348-27507/com.dearpeople.divecomputer D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=7
12-19 19:56:46.921 27348-27630/com.dearpeople.divecomputer I/DfuBaseService: Connecting to the device...
12-19 19:56:46.921 27348-27630/com.dearpeople.divecomputer D/BluetoothGatt: connect() - device: C0:A2:2F:C6:F8:CE, auto: false
    registerApp()
    registerApp() - UUID=faa5e4b7-9914-4029-b338-b94033d9ddd4
12-19 19:56:46.931 27348-27366/com.dearpeople.divecomputer D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
12-19 19:56:49.371 27348-27631/com.dearpeople.divecomputer D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=C0:A2:2F:C6:F8:CE
12-19 19:56:49.371 27348-27631/com.dearpeople.divecomputer I/DfuBaseService: Connected to GATT server
12-19 19:56:49.371 27348-27348/com.dearpeople.divecomputer I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
[... cut ...]
12-19 20:19:46.419 11224-12392/com.dearpeople.divecomputer I/DfuBaseService: Cleaning up...
12-19 20:19:46.420 11224-12392/com.dearpeople.divecomputer D/BluetoothGatt: close()
    unregisterApp() - mClientIf=5
12-19 20:19:47.113 11224-11526/com.dearpeople.divecomputer D/mali_winsys: EGLint new_window_surface(egl_winsys_display*, void*, EGLSurface, EGLConfig, egl_winsys_surface**, egl_color_buffer_format*, EGLBoolean) returns 0x3000
12-19 20:19:47.116 11224-11224/com.dearpeople.divecomputer I/DfuBaseService: DFU service destroyed

ohjunki, 2018-12-26:
Hi @philips77  
Could you give me some feedback? :) I have been waiting for your feedback.

philips77, 2018-12-26:
Hi,
I'm on vacations now. Coming back on January the 7th.

philips77, 2019-01-21:
Hello,
Sorry for the delay. 

In the first case, the device should reboot itself in bootloader mode after the first 0x01-04 command was sent. The service would connect to it and continue with OTA update. But instead it disconnects and stays in app mode.

Possible reasons:
1. There is no DFU bootlaoder flashed on the device, only the SoftDevice and Application.
2. The bootloader timeout is set to be very short, few seconds. In this time the device may advertise as bootloader, but then it reboots again to app mode.
3. Some other reason (jump to bootloader failed, etc).

In the second case I would blame the phone's controller. Budget phone often don' have high quality parts/software and they cause problems. I recommend turning off Bluetooth or restarting the phone and trying again. On one of our Samsung S3 a factory reset helped, but something very severe must have happened there.
Also, 133 may happen when there is a packet collision, when some packet was rejected, expected but wasn't received, or some timing issue appeared. It may happen to move to a less noisy (less Bluetooth/WiFi devices) environment to lower chances of packet collision. Do you have a BLE sniffer to check what exactly went wrong? You may also use the HCI snoop-log form the Developer Settings on Android and open it in Wireshark.

philips77, 2019-01-29:
Hello, did you manage to solve your issue?

elemprod, 2019-03-20:
I was struggling with a similar issue, my previously working DFU Activity would repeatably fail in the boot loader on a low end Huawei phone.   Rebooting the phone magically fixed the issue.  -Ben