nordicsemi/Android-DFU-Library #152: onError error: 522; errorType: 3; message: OPERATION FAILED

Neargye, 2018-12-04:
When updating device, received error `onError error: 522; errorType: 3; message: OPERATION FAILED`.
What does thismean?

philips77, 2018-12-04:
Here's how to decode 522:
522 = 0x20A.
0x200 is a mask for [ERROR_REMOTE_TYPE_SECURE](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L534)
0xA, for Secure DFU, is [OPERATION FAILED](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/error/SecureDfuError.java#L36)

Which SDK are you using?

philips77, 2018-12-04:
Also, when does the error occur? After sending the Init packet, or at the very end of the upload?

Neargye, 2018-12-04:
SDK v15.2

logs:
```
2018-12-04 17:44:49.544 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/FinchShiftFota: FinchFota.start() isDfuServiceRunning: false
2018-12-04 17:44:52.698 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/FinchShiftFota: Dfu targ C6:30:64:4B:4E:30 was found by ble scanner. Gonna start dfu.
2018-12-04 17:44:52.718 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: DFU service created. Version: 1.8.0
2018-12-04 17:44:52.738 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin W/DfuBaseService: getNotificationTarget() should not return null if the service is to be started as a foreground service
2018-12-04 17:44:52.778 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Connecting to the device...
2018-12-04 17:44:52.778 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin D/BluetoothGatt: connect() - callback: no.nordicsemi.android.dfu.DfuBaseService$5@728c564
2018-12-04 17:44:53.005 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
2018-12-04 17:44:53.006 2377-2392/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Connected to GATT server
2018-12-04 17:44:53.009 2377-2392/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Attempting to start service discovery... succeed
2018-12-04 17:44:53.438 2377-2392/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Services discovered
2018-12-04 17:44:53.441 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin W/DfuImpl: Secure DFU bootloader found
2018-12-04 17:44:53.442 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/FinchShiftFota: Dfu device connected: C6:30:64:4B:4E:30
2018-12-04 17:44:53.442 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/FinchShiftFota: Dfu process starting: C6:30:64:4B:4E:30
2018-12-04 17:44:54.442 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Requesting MTU = 517
2018-12-04 17:44:54.525 2377-2826/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: MTU changed to: 23
2018-12-04 17:44:54.525 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Enabling notifications...
2018-12-04 17:44:55.586 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
2018-12-04 17:44:55.668 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
2018-12-04 17:44:55.668 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 0)
2018-12-04 17:44:55.726 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Creating Init packet object (Op Code = 1, Type = 1, Size = 135)
2018-12-04 17:44:55.786 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending 135 bytes of init packet...
2018-12-04 17:44:55.787 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending init packet (Value = 12-84-01-0A-3E-08-01-12-3A-08-04-10-3F-1A-02-AF-01-20-00-28)
2018-12-04 17:44:55.789 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending init packet (Value = 00-30-00-38-CC-EB-07-42-24-08-03-12-20-FE-02-D7-C4-ED-58-A3)
2018-12-04 17:44:55.792 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending init packet (Value = 7E-03-C3-87-EC-A7-EF-72-5A-CD-61-48-BA-63-98-7B-28-32-56-65)
2018-12-04 17:44:55.795 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending init packet (Value = 4A-24-9C-70-E1-48-00-10-00-1A-40-DE-4A-DE-BB-77-24-9E-A7-3B)
2018-12-04 17:44:55.803 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending init packet (Value = BD-68-69-3F-E1-7E-08-3E-D7-09-29-C0-2D-E4-46-6A-4A-EC-92-EC)
2018-12-04 17:44:55.806 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending init packet (Value = 86-DD-63-0F-91-72-E5-91-78-DA-AB-64-AC-3D-44-10-1C-2F-82-53)
2018-12-04 17:44:55.809 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending init packet (Value = B5-2D-53-93-7C-86-6C-D5-4B-6F-6F-A8-F6-C2-F2)
2018-12-04 17:44:55.813 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
2018-12-04 17:44:55.877 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Checksum received (Offset = 135, CRC = 86ECFD6D)
2018-12-04 17:44:55.878 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuImpl: Executing init packet (Op Code = 4)
2018-12-04 17:44:55.938 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin E/DfuImpl: Executing object failed (error 10): OPERATION FAILED
2018-12-04 17:44:55.939 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Disconnecting from the device...
2018-12-04 17:44:55.939 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/FinchShiftFota: Dfu device disconnecting: C6:30:64:4B:4E:30
2018-12-04 17:44:55.942 2377-2392/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Disconnected from GATT server
2018-12-04 17:44:55.947 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Refreshing result: true
2018-12-04 17:44:55.947 2377-2967/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Cleaning up...
[... cut ...]
2018-12-04 17:45:38.230 2377-2992/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Refreshing result: true
2018-12-04 17:45:38.230 2377-2992/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Cleaning up...
2018-12-04 17:45:38.265 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2018-12-04 17:45:38.858 2377-2377/com.finchtechnologies.wavevr.shiftdeviceplugin I/DfuBaseService: DFU service destroyed
```

philips77, 2018-12-04:
Hello, here's the info I got from the SDK team:

There are basically three places it is returned:
1) hash calculation of init packet fails (nrf_crypto returns error)
2) signature size is wrong in init packet
3) settings cannot be written, e.g. flash error or NRF_DFU_SETTINGS_IN_APP
4) you are using crypto at the same time (NRF_ERROR_CRYPTO_BUSY)

If you are using [nrfutil](https://github.com/NordicSemiconductor/pc-nrfutil) to generate the ZIP file, the 1) and 2) options are not really possible.

In SDK 12 and 13 the OPERATION FAILED was also returned when the Init packet validation failed with some other reason, like signature, HW version mismatch, app version too small, etc., but since 14 they are returned as EXTENDED ERRORS with explanation.

philips77, 2018-12-04:
Most likely you have defined NRF_DFU_SETTINGS_IN_APP in the bootloader.
https://www.nordicsemi.com/DocLib/Content/SDK_Doc/nRF5_SDK/v15-2-0/group__nrf__dfu__config#gaab132912ffcd989a7cf2d08d65a18ab5

Neargye, 2018-12-04:
I tried several times again, with a different build of firmware and after upgrading to 40%, we get the following error: `Target device disconnected with status 19`.
What could be the problem?

philips77, 2018-12-04:
Error 19 (0x13) is: GATT CONN TERMINATE PEER USER. I don't know why could it disconnect. May be something with the flash?
If you get disconnect over and over you may try enabling PRNs with [setPacketsReceiptNotificationsEnabled(boolean)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L190) and adjusting value with [setPacketsReceiptNotificationsValue(int)](setPacketsReceiptNotificationsValue).

Neargye, 2018-12-05:
Here logs with fail ` Target device disconnected with status 19`
```
I/DfuBaseService: DFU service created. Version: 1.8.0
W/DfuBaseService: getNotificationTarget() should not return null if the service is to be started as a foreground service
I/DfuBaseService: Connecting to the device...
D/BluetoothGatt: connect() - callback: no.nordicsemi.android.dfu.DfuBaseService$5@e79dbd3
I/FinchShiftFota: RightHand disconnected and going to boot as dfu target with currentAddress C6:30:64:4B:4E:30
I/DfuBaseService: Connected to GATT server
I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
I/DfuBaseService: Attempting to start service discovery... succeed
I/DfuBaseService: Services discovered
W/DfuImpl: Secure DFU bootloader found
I/FinchShiftFota: Dfu device connected: C6:30:64:4B:4E:30
I/FinchShiftFota: Dfu process starting: C6:30:64:4B:4E:30
I/DfuImpl: Requesting MTU = 517
I/DfuImpl: MTU changed to: 247
I/DfuImpl: Enabling notifications...
I/DfuImpl: Setting object to Command (Op Code = 6, Type = 1)
I/DfuImpl: Command object info received (Max size = 256, Offset = 0, CRC = 00000000)
I/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 0)
I/DfuImpl: Creating Init packet object (Op Code = 1, Type = 1, Size = 153)
I/DfuImpl: Sending 153 bytes of init packet...
I/DfuImpl: Sending init packet (Value = 12-96-01-0A-50-08-01-12-4C-08-04-10-3F-1A-02-AF-01-20-00-28-00-30-00-38-84-E1-07-42-24-08-03-12-20-36-0E-56-A7-A6-14-F0-8C-23-E9-25-B5-75-6C-55-48-92-F1-9B-2C-C5-B5-75-D9-8B-39-FC-89-79-9A-31-8F-48-00-52-10-0F-11-6B-A8-5F-C4-6C-C7-C6-EB-9F-D1-17-46-1F-6F-10-00-1A-40-E7-74-8E-C7-FE-4C-47-08-4E-32-C9-75-84-4E-C3-45-92-01-29-08-AC-05-86-CD-60-DA-C2-5B-58-44-59-82-54-2C-40-68-D7-5E-37-4A-08-AC-6B-06-2B-0F-18-A3-39-3B-80-F3-57-80-1F-B0-68-34-0F-1E-9A-67-60-0B)
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 153, CRC = F78A7CFF)
I/DfuImpl: Executing init packet (Op Code = 4)
I/DfuImpl: Sending the number of packets before notifications (Op Code = 2, Value = 1)
I/DfuImpl: Setting object to Data (Op Code = 6, Type = 2)
I/DfuImpl: Data object info received (Max size = 4096, Offset = 0, CRC = 00000000)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/32)
I/FinchShiftFota: Dfu process started: C6:30:64:4B:4E:30
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 4096, CRC = E6FBBC97)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (2/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 8192, CRC = 3B74CA2A)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (3/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 12288, CRC = 9CAB7242)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (4/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 16384, CRC = D5EB4460)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (5/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 20480, CRC = F49893B5)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (6/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 24576, CRC = 691AF233)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (7/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 28672, CRC = A11A60FE)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (8/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 32768, CRC = 9EABF0BC)
I/DfuImpl: Executing data object (Op Code = 4)
[... 102 lines cut ...]
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 118784, CRC = 0E7EE524)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (30/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 122880, CRC = 59943905)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 4096) (31/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 126976, CRC = D2442E6D)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Creating Data object (Op Code = 1, Type = 2, Size = 132) (32/32)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
I/DfuImpl: Checksum received (Offset = 127108, CRC = B1C1DB50)
I/DfuImpl: Executing data object (Op Code = 4)
I/DfuImpl: Transfer of 127108 bytes has taken 45989 ms
W/DfuBaseService: Target device disconnected with status: 19
I/DfuBaseService: Refreshing result: true
I/DfuBaseService: Cleaning up...
I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
I/DfuBaseService: DFU service destroyed
```

Neargye, 2018-12-05:
PROGRESS_COMPLETED  not come at end.

philips77, 2018-12-05:
Did you receive any other progress?

Neargye, 2018-12-05:
setPacketsReceiptNotificationsEnabled(true) and setPacketsReceiptNotificationsValue(1) looks help.
Many thanks!

Neargye, 2018-12-05:
PROGRESS_COMPLETED did not come because I remove listener too soon

closed 2018-12-05 by Neargye