nordicsemi/Android-DFU-Library #153: 4102 DFU CHARACTERISTICS NOT FOUND

aurelienbouveret, 2018-12-04:
Hi :)
I'm running into an issue when starting a DFU update with a DFU service implementation (no activity)

The issue is not affecting all the devices we use, the S8 is the most impacted model we have (it is running Android 8.1)
DFU lib version: 1.8.0

Here is the process I use:
- Disconnection from the app
- Wait for 2 seconds
- Launch the DFU service

Here is the service initiator we use:
`new DfuServiceInitiator(getRealMac())
        .setDisableNotification(true)
        .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)
        .setZip(filePath);`

Some related log:
(The timeline goes bottom to top)

03-12 07:10:50.671  D/BluetoothGatt(27357): unregisterApp() - mClientIf=8
03-12 07:10:50.671  D/BluetoothGatt(27357): close()
03-12 07:10:50.670  D/BluetoothGatt(27357): refresh() - device: C0:4B:CB:A3:4D:51
03-12 07:10:50.668  D/BluetoothGatt(27357): onClientConnectionState() - status=0 clientIf=8 device=C0:4B:CB:A3:4D:51
03-12 07:10:50.662  D/BluetoothGatt(27357): cancelOpen() - device: C0:4B:CB:A3:4D:51
03-12 07:10:50.662  W/DfuBaseService(27357): DFU Service not found.
// Here I get the error

03-12 07:10:50.661  D/BluetoothGatt(27357): onSearchComplete() = Device=C0:4B:CB:A3:4D:51 Status=0
03-12 07:10:50.656  D/BluetoothGatt(27357): discoverServices() - device: C0:4B:CB:A3:4D:51
03-12 07:10:50.654  D/BluetoothGatt(27357): onClientConnectionState() - status=0 clientIf=8 device=C0:4B:CB:A3:4D:51
03-12 07:10:50.648  D/BluetoothGatt(27357): onClientRegistered() - status=0 clientIf=8
03-12 07:10:50.637  D/BluetoothGatt(27357): registerApp() - UUID=c4603dc0-12e7-4a5a-84ac-d1fb20a61c05
03-12 07:10:50.637  D/BluetoothGatt(27357): registerApp()
03-12 07:10:50.636  D/BluetoothAdapter(27357): isSecureModeEnabled
03-12 07:10:50.636  D/BluetoothGatt(27357): connect() - device: C0:4B:CB:A3:4D:51, auto: false
// Connection from DFU service

03-12 07:10:48.627  D/KLNordicBleManager$disconnectWithoutReconnect(27357): disconnect success
03-12 07:10:48.627  E/KolibreeBleDriver(27357): onDeviceDisconnected: C0:4B:CB:A3:4D:51 disconnected
03-12 07:10:48.626  D/BluetoothGatt(27357): unregisterApp() - mClientIf=8
03-12 07:10:48.626  D/BluetoothGatt(27357): close()
03-12 07:10:48.625  D/KLNordicBleManager(27357): gatt.close()
03-12 07:10:48.625  I/KLNordicBleManager(27357): Disconnected
03-12 07:10:48.625  D/KLNordicBleManager(27357): [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
// Disconnection from our app

Thank you very much have a good day !

philips77, 2018-12-04:
Hello!
Well, the log says, that the DFU service and required characteristics were not found on the device. Are you sure you have them there, and the UUIDs are correct?
Do you see them in nRF Connect app?

aurelienbouveret, 2018-12-04:
Thanks for the quick answer! Yes I do see them in nRF connect, and I forgot to mention that the issue is not reproducible 100% times, from the QA team reports it is happening with a frequency of once every 4 or 5 times

philips77, 2018-12-04:
Are you clearing the device cache in your app? Or connecting to the same device during DFU? Can you reproduce this issue in nRF Connect, by trying 5+ times?

aurelienbouveret, 2018-12-05:
Hi philips77, sorry for the delay I don't have a phone that reproduce the issue. My friend tested with the S7 (which is reproducing the issue in our app) and has been able to run 5 updates successfully so the issue is coming from us!

Thank you for your time, have a good day

closed 2018-12-05 by aurelienbouveret