nordicsemi/Android-DFU-Library #157: Rely on dfu to report an error

mshandan, 2018-12-19:
Error:(61, 0) Could not find method implementation() for arguments [com.android.support:support-core-utils:28.0.0] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

Please install the Android Support Repository from the Android SDK Manager.
<a href="openAndroidSdkManager">Open Android SDK Manager</a>

so what should I do？ Thanks.

philips77, 2018-12-19:
Please check out this link: https://stackoverflow.com/questions/45615474/gradle-error-could-not-find-method-implementation-for-arguments-com-android
You need to set up your Android Studio properly. We are using Gradle wrapper 4.10.2 and Android Studio 3.2.1.

closed 2018-12-19 by philips77

mshandan, 2019-01-04:
Thank you so much for your assistance.I wonder how it can support in Android Studio 3.0.1 because of   It's difficult to change the version of my APP.Or is there any other way to solve this problem.