nordicsemi/Android-DFU-Library #160: How do lower versions of apps rely on dfu

mshandan, 2019-01-07:
Thank you so much for your assistance.I wonder how it can support in Android Studio 3.0.1 because of   It's difficult to change the version of my APP.Or is there any other way to solve this problem.

philips77, 2019-01-07:
Hello!
DFU will work only with Android 4.3, as this version added support for Bluetooth LE. You may add the library to your app, but disable this feature for older Android versions.
You'd have to include the DFU library as a module (clone repo, and add as module as described in Readme), change minSdkVersion to what you need and don't use it before API 18 as it will crash. Perhaps you'll need to suppress some warnings.

philips77, 2019-01-07:
https://github.com/NordicSemiconductor/Android-DFU-Library/tree/release/documentation#integration

mshandan, 2019-01-07:
Yes, I do need to use Android Studio 3.0.1 , but what should I do? Can I be more specific? Thank you！

mshandan, 2019-01-08:
Hello,when I change compileSdkVersion 28 to compileSdkVersion 25,the following error was generated:
 Could not resolve all dependencies for configuration ':app:_debugApk'.
   > A problem occurred configuring project ':dfu'.
      > Could not resolve all dependencies for configuration ':dfu:_releasePublishCopy'.
         > Could not resolve com.android.support:support-core-utils:25.4.0.
           Required by:
               project :dfu
            > Could not resolve com.android.support:support-core-utils:25.4.0.

mshandan, 2019-01-08:
[image]

philips77, 2019-01-08:
Well.. in API 25 there was no API for Bluetooth 5 features, so you need to remove/comment out those parts. All related to PHY, perhaps something more.
I assume you've solved the problem with https://mvnrepository.com/artifact/com.android.support/support-core-utils/25.4.0?

mshandan, 2019-01-09:
Okay, so I've removed those errors and the program works, and I've created DfuService and NotificationActivity as the documentation says.But now what can I do to achieve a buttonless dfu upgrade?

philips77, 2019-01-09:
The buttonless upgrade should work out of the box if you didn't modify the bootloader or the buttonless service in your firmware. Just start the DFU process using `DfuServiceInitiator` as described in Readme.

mshandan, 2019-01-14:
If I use a local zip package to implement the dfu upgrade, how do I set mFileStreamUri and mFilePath?

if (mFileType == DfuService.TYPE_AUTO)
    starter.setZip(mFileStreamUri, mFilePath);

roshanrajaratnam, 2019-01-14:
Hi @mshandan, In order to do this you will have to get the uri of the file you wish to flash using a file manager that's available in the device. You need to open the file chooser app and then upon selecting the file you can get the uri and filepath in the `onActivityResult(final int requestCode, final int resultCode, final Intent data)` callback. Please take a look our toolbox example on how to [open](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/dfu/DfuActivity.java#L672) the file chooser and how to get the [uri](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/dfu/DfuActivity.java#L425).

Hope this helps

mshandan, 2019-01-14:
This is what I wrote after connecting bluetooth. Below are the code and the contents in debug. I want to know why it doesn't work and what I should do.Thank you very much for your help.

            final DfuServiceInitiator starter = new DfuServiceInitiator(gatt.getDevice().getAddress())
                    .setDeviceName(gatt.getDevice().getName())
                    .setKeepBond(true);
          
            starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
            System.out.println(" 11****dfu: "+ starter);
            starter.setZip(R.raw.app2);
            System.out.println(" ****dfu: "+ starter);
            final DfuServiceController controller =starter.start(context,DfuService.class);

System.out:  11****dfu: no.nordicsemi.android.dfu.DfuServiceInitiator@7d1e772
System.out:  ****dfu: no.nordicsemi.android.dfu.DfuServiceInitiator@7d1e772

mshandan, 2019-01-15:
Hi, do I need to create a new class to implement DfuServiceInitiator starter, or just connect to bluetooth and get the MAC address and name to implement it.

philips77, 2019-01-15:
Your code looks ok. Did you add your DfuService to the *AndroidManifest.xml*?

Is your device based on SDK 12? If not, you may remove `starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);`
Is your device bonded and remains bonded while DFU process? This was introduces in SDK 14, afair, or SDK 8-11. If not, you may remove `.setKeepBond(true)`.

mshandan, 2019-01-15:
I've been adding Dfuservice to androidmanifest.xml since the beginning.My device is based on sdk12.3, so I deleted the starter setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled (true) and.SetKeepBond (true).

But there is no effect. I am thinking whether there is something wrong with my service by connecting bluetooth. 
If I do other operations at the same time, I cannot upgrade dfu.

philips77, 2019-01-15:
Hmm,.. yes, that may be related. I've heard that DFU service isn't working with Firebase connection, etc. I can't find this issue now.
In SDK 12.3 you should leave `starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);`, as this service was used for buttonless in SDK 12.x, as written in docs.

mshandan, 2019-01-16:
I import no. Nordicsemi. Android. Nrftoolbox source code, and test the hardware upgrade process, found that there is no this step in my APP, so I guess I lack what code, what should I do?

[image]

mshandan, 2019-01-17:
If you have time, I hope you can reply to me as soon as possible, I need to solve this problem immediately, thank you for your help！

philips77, 2019-01-17:
Hello @mshandan,
Starting the DFU service should be straightforward. The only requirements are adding DfuService as `<service...>` tag to the AndroidManifest and then calling `start(context, DfuService.class)` as you do. To get the detailed logs on LogCat you have to extend `isDebug()` method in your DfuService and return true. Check out here:
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/9a29921769d775f2ff2dc1276a7062fe0c25c3b7/app/src/main/java/no/nordicsemi/android/nrftoolbox/dfu/DfuService.java#L49

mshandan, 2019-01-17:
I suddenly thought of a question, my hardware product can run on the latest no. Nordicsemi. Android. Nrftoolbox, is it possible that I failed to upgrade due to the different version of dfu I imported?

philips77, 2019-01-17:
The DFU library is compatible with all versions of DFU bootloader released by Nordic and even some modifications. Each next DFU library is supposed to have less bugs, but still all are backwards compatible, unless there is a bug. I don't remember which version of DFU Lib is used by the current version of nRF Toolbox, but it's written on the screen.
DFU may fail only if some breaking changes were made in the bootloader/buttonless service, but that's not your case.

mshandan, 2019-01-18:
I think my problem is that my hardware isn't in dfu mode, so it's problematic to boot directly through DfuService.What do you think?If so, what can I do to get the hardware into dfu mode?

philips77, 2019-01-18:
If you are using any of the buttonless services that we had in Nordic (first buttonless service was added in SDK 6.1, then it has evolved until get the recent shape) then everything should work automatically. When you start DFU by giving any address, the DFU service will connect to it (the device may already be connected, it doesn't matter, but the service will create its own GATT client to it), will perform service discovery and check what's on the device and, depending on what has been found, will perform DFU or jump to DFU mode using the found service.

mshandan, 2019-02-21:
Hello, my device can be upgraded with dfu on the NRF toolbox, but that is achieved by clicking the button. Does this mean that it is not feasible for me to use the buttonless service? The hardware is also written according to relevant documents, I am not sure whether there is relevant code of the buttonless service in it.

philips77, 2019-02-21:
Try connecting to your device with nRF Connect for Android. You will see the device's services there. If Buttonless update is supported you should see there Experimental buttonless service (assuming you are using SDK 12.3).

mshandan, 2019-02-21:
[image]
The last one is support?

philips77, 2019-02-21:
Yes, that one is supported. You have to call `.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)`, like here:
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/9a29921769d775f2ff2dc1276a7062fe0c25c3b7/app/src/main/java/no/nordicsemi/android/nrftoolbox/dfu/DfuActivity.java#L742
and it should work. You may test it by:
1. Enable notifications on the characteristic inside,
2. Write 0x01 to the characteristic.
You should get 0x20-01-01 as return and the device should restart in DFU mode. The notification may not come, but the device should reboot.

mshandan, 2019-02-21:
Is this test on the Android side?Can I be more specific? Thank you!

philips77, 2019-02-21:
Yes, just use nRF Connect. After connecting and clicking the Experiments service you should see the Buttonless characteristic with 2 icons on the right. Click the arrows pointing down to enable notifications and then the other one pointing up to send data. There should be a dialog "Do you want to reboot in Bootloader mode?", just select YES an SEND.

mshandan, 2019-02-21:
According to what you said, I found that the device has been restarted and the device name has changed, but the device in my APP still cannot be upgraded. When the dfu part is executed, the dfu service does not seem to start.

philips77, 2019-02-21:
Could you post logs fro LogCat here? To get them, return true from `isDebug()` method in your DfuService.

mshandan, 2019-02-22:
[image]
When you say logcat log, do you mean this one?Or a file?Now I don't know if the dfu service is started.

philips77, 2019-02-22:
Yes, I need the relevant logs from LogCat. You may filter by "dfu".
Did you add your DfuService class to AndroidManifest?

mshandan, 2019-02-22:
Ok, I initially added the DfuService class to the AndroidManifest, but there is no dfu log associated with it.I used
Intent intent1 = new Intent(controlCenterActivity.this, DfuService.class);
            startService(intent1); 
to force dfuservice on。
Logcat as follows：
02-22 18:39:37.305 15100-15591/com.feijia.alfapreg D/dfu: getNotificationTarget: 
02-22 18:39:37.309 15100-15591/com.feijia.alfapreg E/DfuBaseService: An exception occurred while opening files. Did you set the firmware file?
                                                                     java.lang.NullPointerException: Attempt to invoke virtual method 'int java.io.InputStream.available()' on a null object reference
                                                                         at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1062)
                                                                         at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:76)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                         at android.os.Looper.loop(Looper.java:173)
                                                                         at android.os.HandlerThread.run(HandlerThread.java:65)
02-22 18:39:37.309 15100-15591/com.feijia.alfapreg D/dfu: getNotificationTarget: 
02-22 18:39:37.318 15100-15100/com.feijia.alfapreg I/DfuBaseService: DFU service destroyed

philips77, 2019-02-22:
You have to use the `DfuServiceInitiator` to start DFU, not like you are doing.
You have it described here: https://github.com/NordicSemiconductor/Android-DFU-Library/tree/release/documentation

mshandan, 2019-02-23:
Sorry, you may not understand what I mean, I mean I wrote everything you said, there is no dfuservice logcat, but if I add that there will be a logcat. 
final DfuServiceInitiator starter = new DfuServiceInitiator(gatt.getDevice().getAddress())
                .setDeviceName(gatt.getDevice().getName())
                .setKeepBond(true);
        starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
        starter.setZip(R.raw.app2);
        final DfuServiceController controller =starter.start(context,DfuService.class);

        Intent intent1 = new Intent(controlCenterActivity.this, DfuService.class);
        startService(intent1);

philips77, 2019-02-23:
You should not start DfuService on your own. Initiator will start it. Remove the last 2 lines.

mshandan, 2019-02-24:
Yes, I know the last two lines are not needed, but Initiator did not start DfuService because I did not see any logs about DfuService.

mshandan, 2019-02-25:
I have found the reason why the dfu service is not started. I am very sorry for wasting your time.Below is the logcat：

philips77, 2019-02-25:
Congratulations, looks like the DFU worked! :)

closed 2019-02-25 by philips77

mshandan, 2019-02-25:
But I don't think it worked, because the name of the device hasn't changed

mshandan, 2019-02-25:
There should not be a successful logcat upgrade？

mshandan, 2019-02-25:
I have posted all my logcat, but I remember that if the dfu upgrade is successful, there will be a successful logcat, and the name of the device will change, but now the device name has not changed, so I feel there is something missing, I will be successful, I hope you can help me, thank you.

philips77, 2019-02-25:
Well.. the device name change depends on what you have send as an update. If it's the same fw as was already running, nothing will change.
In LogCat success is when you don't get any error. Also, you got: `Transfer of 32976 bytes has taken 8878 ms.`.

mshandan, 2019-02-25:
The hardware told me that the name of the device will change after the successful upgrade. The upgrade time has been over 8 seconds. I think there must be some mistake.