nordicsemi/Android-DFU-Library #161: DFU process interruption at 30%

hyyu, 2019-01-08:
Hi,

I'm trying to make the DFU autonomous on my app. The goal is that the user doesn't pick the zip by himself and the update processes just by tapping an "update" button.

When I'm implementing the method proposed by the repo's documentation, the problem is that the update always interrupts around 30%

To continue the update, I have to go manually on an app like nrfConnect and update it manually as it went under the name "DfuTarg". Is there a way to make it automatic, without having to manually finish the update?

Here's the log from the start of the dfu process to its interruption (around 30%):

`2019-01-08 10:54:35.481 3394-3394/com.thingsaremoving.enovap E/DfuListener: onDeviceConnected
2019-01-08 10:54:35.481 3394-3394/com.thingsaremoving.enovap E/DfuListener: onDfuProcessStarting
2019-01-08 10:54:36.492 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90003-f315-4f60-9fb8-838830daea50 enable: true
2019-01-08 10:54:41.741 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=11 device=FE:E6:FD:B8:13:DE
2019-01-08 10:54:41.741 3394-3592/com.thingsaremoving.enovap D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=12 device=FE:E6:FD:B8:13:DE
2019-01-08 10:54:41.743 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: close()
2019-01-08 10:54:41.743 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: unregisterApp() - mClientIf=12
2019-01-08 10:54:41.755 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: setCharacteristicNotification() - uuid: 23058a49-636d-82ab-d641-badd3648cc9a enable: false
2019-01-08 10:54:41.755 3394-3598/com.thingsaremoving.enovap D/BluetoothManager: getConnectionState()
2019-01-08 10:54:41.755 3394-3598/com.thingsaremoving.enovap D/BluetoothManager: getConnectedDevices
2019-01-08 10:54:41.756 3394-3698/com.thingsaremoving.enovap D/BluetoothAdapter: isLeEnabled(): ON
2019-01-08 10:54:41.757 3394-3406/com.thingsaremoving.enovap E/Companion: Charac change error Disconnected from FE:E6:FD:B8:13:DE with status 8 (GATT_INSUF_AUTHORIZATION)
2019-01-08 10:54:41.757 3394-3406/com.thingsaremoving.enovap E/BleManager: release()
2019-01-08 10:54:41.758 3394-3406/com.thingsaremoving.enovap I/Companion: onConnectionStateChanged : RxBleConnectionState{DISCONNECTED}
2019-01-08 10:54:41.758 3394-3406/com.thingsaremoving.enovap E/BleManager: release()
2019-01-08 10:54:41.759 3394-3406/com.thingsaremoving.enovap E/bleManager: error characteristic change : Disconnected from FE:E6:FD:B8:13:DE with status 8 (GATT_INSUF_AUTHORIZATION)
2019-01-08 10:54:41.761 3394-3598/com.thingsaremoving.enovap D/BluetoothGatt: close()
2019-01-08 10:54:41.761 3394-3598/com.thingsaremoving.enovap D/BluetoothGatt: unregisterApp() - mClientIf=11
2019-01-08 10:54:41.764 3394-3406/com.thingsaremoving.enovap D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2019-01-08 10:54:41.787 3394-3394/com.thingsaremoving.enovap E/ConnectionState: disconnected
2019-01-08 10:54:41.872 3394-3698/com.thingsaremoving.enovap D/BluetoothAdapter: isLeEnabled(): ON
2019-01-08 10:54:43.905 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: connect() - device: FE:E6:FD:B8:13:DF, auto: false
2019-01-08 10:54:43.906 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: registerApp()
2019-01-08 10:54:43.906 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: registerApp() - UUID=766e5e19-e1f9-48d3-a0a0-9c5eeba2b66a
2019-01-08 10:54:43.910 3394-3592/com.thingsaremoving.enovap D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2019-01-08 10:54:44.059 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=FE:E6:FD:B8:13:DF
2019-01-08 10:54:44.060 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: discoverServices() - device: FE:E6:FD:B8:13:DF
2019-01-08 10:54:44.746 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onConnectionUpdated() - Device=FE:E6:FD:B8:13:DF interval=24 latency=0 timeout=400 status=0
2019-01-08 10:54:45.045 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onConnectionUpdated() - Device=FE:E6:FD:B8:13:DF interval=6 latency=0 timeout=500 status=0
2019-01-08 10:54:45.135 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onSearchComplete() = Device=FE:E6:FD:B8:13:DF Status=0
2019-01-08 10:54:45.224 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onConnectionUpdated() - Device=FE:E6:FD:B8:13:DF interval=24 latency=0 timeout=400 status=0
2019-01-08 10:54:45.825 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onConnectionUpdated() - Device=FE:E6:FD:B8:13:DF interval=24 latency=0 timeout=400 status=0
2019-01-08 10:54:46.149 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: configureMTU() - device: FE:E6:FD:B8:13:DF mtu: 517
2019-01-08 10:54:46.244 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onConfigureMTU() - Device=FE:E6:FD:B8:13:DF mtu=23 status=0
2019-01-08 10:54:46.246 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90001-f315-4f60-9fb8-838830daea50 enable: true
2019-01-08 10:54:49.738 3394-3394/com.thingsaremoving.enovap E/timezone: time : 1.0
2019-01-08 10:54:49.738 3394-3394/com.thingsaremoving.enovap I/LocationUpdate: 2.2889202 : 48.8515689
2019-01-08 10:55:09.687 3394-3394/com.thingsaremoving.enovap E/timezone: time : 1.0
2019-01-08 10:55:09.687 3394-3394/com.thingsaremoving.enovap I/LocationUpdate: 2.2889202 : 48.8515689
2019-01-08 10:55:16.871 3394-3406/com.thingsaremoving.enovap E/DfuImpl: Characteristic write error: 133
2019-01-08 10:55:16.871 3394-3406/com.thingsaremoving.enovap D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=11 device=FE:E6:FD:B8:13:DF
2019-01-08 10:55:16.873 3394-3698/com.thingsaremoving.enovap E/DfuBaseService: Unable to write Op Code 3 (error 133)
2019-01-08 10:55:16.874 3394-3698/com.thingsaremoving.enovap W/removing.enova: Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (light greylist, reflection)
2019-01-08 10:55:16.875 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: refresh() - device: FE:E6:FD:B8:13:DF
2019-01-08 10:55:16.876 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: close()
2019-01-08 10:55:16.876 3394-3698/com.thingsaremoving.enovap D/BluetoothGatt: unregisterApp() - mClientIf=11`

philips77, 2019-01-08:
Does it work for you in nRF Connect?
The error 133 that you get happens from time to time. It may happen during the upload, not necessary around 30%. Try setting PRN in DfuServiceInitiator (2 methods are required: setPacketReceiptNotificationsEnabled() and value).
The solution for 133 is to start DFU again. This happens when something weird has happened, like packet collision, etc. 
I agree that the lib should reconnect and continue on its own, otherwise you need to scan for the DfuTarg on your own.

hyyu, 2019-01-09:
Hi,

I've come to realize that the bootloader wasn't adequate to the device I was using. I used a newer device (with a newer version of the bootloader) and there is no interruption with it.

Thanks for your answer, it can be useful if that occurs again.

closed 2019-01-09 by philips77

philips77, 2019-01-30:
When the DFU fails the device may still be in DFU mode and may advertise with address+1. That means you can't use the same `BluetoothDevice` object that you used before (actually its address). You should instead scan for the new address and restart DFU with it.

I agree that this should be handled automatically by the app. Disconnections are far too common to let it implement to users. It's a common issue.

philips77, 2019-01-30:
#164

hyyu, 2019-01-30:
I'm already scanning and launching a new connection after the device is disconnecting. Thanks for the reference and the task ticket!