nordicsemi/Android-DFU-Library #162: CRC Error

PJCSpencer, 2019-01-08:
Hi there,

I'm continuing R+D into failed DFU transfer and would like to understand some trace information which the library outputs. I would like to understand if the following are related to library generated CRC values or maybe command codes? Please accept my apologies if this belongs on the DevZone.

Command object sent (CRC = 1E191FF4)
Checksum (Offset = 0, CRC = 00000000) received
Command object info (Max size = 256, Offset = 0, CRC = 00000000) received

Also is it possible to disable the CRC check?

philips77, 2019-01-08:
Hello!
The library calculates the CRC after each object is sent and compares it with one received from the target. Your device responds with offset 0 and CRC 0 to the Calculate CRC command, which is weird, and I would suggest asking on DevZone. There is no option to disable CRC check. Besides, it would fail anyway. You must find out why the bootlodaer does not save the data properly.

philips77, 2019-01-08:
target = device = your device that you are updating
iOS DFU Library works very similarly to Android one. The delegate (DfuPeripheralSelector) is only used to find the 'target' after it has switched to bootloader mode (and has different address).
The method for calculating CRC is built into Secure DFU spec, so both libs (Android and iOS) know how to do this.

PJCSpencer, 2019-01-09:
I have another question.

During our challenges we've come across a number of threads on various channels which suggest some form of latency issue maybe related to BLE write/read speed. Here are a couple of examples:

https://github.com/aritchie/bluetoothle/issues/257
https://github.com/NordicSemiconductor/Android-DFU-Library/issues/128

I was wondering if this is something you have experience and if you have anything which could contribute to resolving the situation we have. Currently our iOS application we implements write with response. We have experimented with variable write 'type' without consistent success.

In attempting to cover all bases we are also investigating how our device functions.

closed 2019-01-29 by philips77