nordicsemi/Android-DFU-Library #166: Exit DFU mode on abort

Del-S, 2019-02-05:
Hi,

I noticed that device restarts into application mode after update is done. But it does not exit DFU mode when update is aborted. Our device then sits and does nothing for 2 minutes before it restarts automatically.

I was wondering if it could be added or if there is any reason it is not implemented.
Thanks for the info.

philips77, 2019-02-05:
Hi, I assume you using secure dfu.
There is no reset command like there was in legacy dfu. As far as I remember there were talk about creating a signed command object for it, but eventually that wasn't implemented.
The reason is I guess that is not required. Am update should be sent to make the remote device better/more secure and actually use facing apps should not allow aborting and should resume asap if upload failed.

Del-S, 2019-02-06:
Oh I thought you can leave dfu mode by sending (opcode = 6). I guess I did not notice it is legacy only. So there is currently no way to leave dfu mode early using secure dfu, if I understand correctly.

Well there are actually two issues I'm dealing with. It's not really breaking, just inconvenient.
1. When device enters dfu mode then there is a chance it will not connect again because of gatt 133. So update never runs and device needs to wait before restarting. I tried to retry the connection but it does not work.
2) When I abort the update it again sits in dfu before it restarts. I'm aborting it when user leaves the update screen (I could remove this, but it will be aborted when user kills the app anyway). Again I tried to reconnect and run the update again but device is unconnectable and update cannot be restarted.

I'm not sure if the "unconnectable" issue is connected to dfu mode or if the problem is in our implementation but i've been told that: "We are using Nordic DFU service without any changes.".

philips77, 2019-02-06:
In the DFU mode the device is using MAC address incremented by 1. Many customers are actually scanning for this +1 address when update fails and retrying using the new address (note that you have to scan the device before starting the server with this new address, otherwise `connectGatt` may not work).
The lib in fact should reconnect and retry on its own in case of 133 or any other-than-abort-or-kill event. There is a #164 for this.

philips77, 2019-02-06:
Also, #151 may be related.

Del-S, 2019-02-06:
> In the DFU mode the device is using MAC address incremented by 1.

Oh nice I'm gonna try that thanks.

Well for me it does not try to restart. I don't have the exact logs for not but I get this message.

> Device not reachable. Check if the device with address " + deviceAddress + " is in range, is advertising and is connectable

And I checked the code of **DfuBaseService** and it should not retry if `mConnectionState == STATE_DISCONNECTED` (line 1163). Which happens in my case. So the device connects, then restarts into the Dfu and cannot be connected again due to it being disconnected. Or well it seems that is the case.

philips77, 2019-02-06:
Right now the retrying is not implemented yet. 
The main problem with it not being implemented is when you update SD+BL+App using a single ZIP file, it will first upgrade SD+BL, then reboot, reconnect to the new bootloader and flash App. but if the second update fails, the lib will now report an error and stop DFU. If you then try to retry, you will fail, as the SD_REQ set in Init Packet for the SD+BL update will not match the currently flashed SD. In order to proceed you have to set the Scope to APPLICATION_ONLY, so that it will skip the SD+BL upload this time. But, they you have to keep track on which part is being flashed and restart DFU accordingly. This should be handled by the lib itself.
Additionally it would help with 133 problems.

closed 2020-06-24 by philips77