nordicsemi/Android-DFU-Library #167: DFU PAIRED Device

sivapraka, 2019-02-07:
Device will be automatically disconnected when i tried to update the dfu to paired device.

philips77, 2019-02-07:
What exactly is the issue? Which SDK are you using the DFU from? Did you use [setKeepBond(true)](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L159) in the initiator? Does your app and bootloader support passing pairing information?
The device disconnencts as it goes to DFU mode and then once again after the upload is complete.

philips77, 2019-02-21:
Also, are you using Buttonless service with Bond sharing (added in SDK 14 or 15, don't remember).

beegee-tokyo, 2019-03-26:
Maybe similar issue and my solution: #173