nordicsemi/Android-DFU-Library #171: How do dfu upgrades use zip on the server

mshandan, 2019-02-27:
How do dfu upgrades use zip on the server or use zip downloaded from the server

philips77, 2019-02-27:
Hello,
You may download the ZIP on your own, for example using Retrofit, save it in app's storage and pass Path or URI to this file to the `DfuServiceInitiator`. The DFU service expects the file as resource id, or a stream given as path or URI. There isn't an option to just pass the bytes as an array, at least in the current version.

E3V3A, 2019-03-26:
Where and what is *Retrofit*?

philips77, 2019-03-27:
https://square.github.io/retrofit/

closed 2019-03-27 by philips77

E3V3A, 2019-03-27:
I see. 
https://github.com/square/retrofit

But that is just the protocol and not where to find the actual files.

mshandan, 2019-03-27:
Do you want to download the file from the server and use it？

E3V3A, 2019-03-27:
Yes. In fact that's what I was hoping (and thought) the app would do for you, automatically.
(How else would I get the correct zip file onto the phone?)

philips77, 2019-03-27:
nRF Toolbox allows you to select the file using any File Browser app, not necessarily from the phone's External Storage. You may use Google Drive, Microsoft One, etc. They are available in the navigation menu of the default file chooser (since KitKat).