nordicsemi/Android-DFU-Library #172: DFU service destroyed

delhmela, 2019-03-04:
Hi
I implemented DFU update library for my android application,I started DFU service and ble device updated but DFU service destroyed did not called after the Cleaning up log. Whats the problem?
I share my log and thanks your advice.

`/DfuBaseService: Connecting to the device...
I/DfuBaseService: Connected to GATT server
 I/DfuBaseService: Waiting 1600 ms for a possible Service Changed indication...
I/DfuBaseService: Attempting to start service discovery... succeed
 I/DfuBaseService: Services discovered
I/DfuImpl: Reading DFU version number...
 I/DfuImpl: Version number read: 0.8 -> Bootloader from SDK 9.0 or newer. Signature supported
 I/DfuImpl: Reading Service Changed CCCD value...
 W/DfuImpl: Legacy DFU bootloader found
 I/DfuImpl: Enabling notifications...
I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
 I/DfuImpl: Sending image size array to DFU Packet (0b, 0b, 94856b)
 I/DfuImpl: Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
 I/DfuImpl: Sending 14 bytes of init packet
    Sending init packet (Value = FF-FF-FF-FF-FF-FF-FF-FF-01-00-67-00-A6-71)
 I/DfuImpl: Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
I/DfuImpl: Sending Receive Firmware Image request (Op Code = 3)
I/DfuImpl: Uploading firmware...
I/DfuImpl: Response received (Op Code = 16, Req Op Code = 3, Status = 1)
 I/DfuImpl: Transfer of 94856 bytes has taken 49047 ms
    Sending Validate request (Op Code = 4)
 I/DfuImpl: Response received (Op Code = 16, Req Op Code = 4, Status = 1)
 I/DfuImpl: Sending Activate and Reset request (Op Code = 5)
 W/DfuBaseService: Target device disconnected with status: 8
/DfuBaseService: Cleaning up...
 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
 I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
 I/DfuBaseService: Connecting to the device...
 I/DfuBaseService: Connected to GATT server
 I/DfuBaseService: Waiting 1600 ms for a possible Service Changed indication...
 I/DfuBaseService: Attempting to start service discovery... succeed
 D/BluetoothGatt: onSearchComplete() = Device=DD:26:80:7F:D3:8B Status=0
 I/DfuBaseService: Services discovered
I/DfuImpl: Reading DFU version number...
I/DfuImpl: Version number read: 20.0 -> Unknown version
 I/DfuImpl: Reading Service Changed CCCD value...
I/DfuImpl: Enabling indications...
 W/DfuImpl: Legacy DFU bootloader found
I/DfuImpl: Enabling notifications...
 I/DfuImpl: Sending Start DFU command (Op Code = 1, Upload Mode = 4)
 E/DfuImpl: Characteristic write error: 3
 E/DfuBaseService: Unable to write Op Code 1 (error 3)
 I/DfuBaseService: Disconnecting from the device...
 D/BluetoothGatt: cancelOpen() - device: DD:26:80:7F:D3:8B
 I/DfuBaseService: Disconnected from GATT server
 I/DfuBaseService: Cleaning up...
 I/DfuBaseService: DFU service destroyed
`
DFU service staeting twice but I dont know why?

philips77, 2019-03-04:
Are you using a ZIP file? Is there anything else in the ZIP except one bin and one dat file?
For me, it looks as if you'd start DFU twice. The second call is queued until the first one is started, as it's an `IntentService`.

delhmela, 2019-03-05:
Yes I am using ZIP file and there is not anything else ,only one dat and bin files.But I used buttonless DFU version and I started only one time.How can I kill DFU Base service manually or something like that?

philips77, 2019-03-06:
I never had issues with it running twice, unless there was a connection error and the service restarts to try again, which is not your case.
Are you sure you are not calling `initializer.startDfu(...)` twice, or not calling `startService(DfuService.class)` afterwards? You have to do some debugging to check where the second run is started.

delhmela, 2019-03-07:
Okay I am sure but I will check to debugging

delhmela, 2019-03-12:
I found the problem thanks a lot .I am closing the issue

closed 2019-03-12 by delhmela

AbdulMateenChughtai, 2021-02-19:
> I found the problem thanks a lot .I am closing the issue

Hey can you please explain what was causing the issue in your case?

ThaddeusLim99, 2022-03-11:
> > I found the problem thanks a lot .I am closing the issue
> 
> Hey can you please explain what was causing the issue in your case?

Same here. Could you explain what was causing the issue? Thank you!