nordicsemi/Android-DFU-Library #174: DFU update Fail

qx-mark-wu, 2019-03-25:
使用1.8.1的库在不分手机上进行升级经常出现以下错误
1、DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED
2、DFU DEVICE DISCONNECT
请问是什么原因呢？
加急！！！
谢谢！！！

qx-mark-wu, 2019-03-25:
还有这个错误也经常有

03-25 16:32:53.233 26743-26762/com.qx.qgbox E/DfuBaseService: Connection state change error: 133 newState: 0
03-25 16:32:53.234 26743-27004/com.qx.qgbox E/DfuBaseService: Device not reachable. Check if the device with address E2:17:5B:71:F8:63 is in range, is advertising and is connectable

qx-mark-wu, 2019-03-25:
很多Android手机升级出现失败的情况基本都是这两种情况导致的；
1、1、DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED
2、DfuBaseService: Connection state change error: 133 newState: 0；
DfuBaseService: Device not reachable. Check if the device with address E2:17:5B:71:F8:63 is in range, is advertising and is connectable

beegee-tokyo, 2019-03-26:
(Using Google translation to Chinese, hope it makes sense)
（使用谷歌翻译成中文，希望有意义）
Maybe similar error to my problem.
也许与我的问题类似的错误。

check issue #173
检查问题 #173 

我的发现/解决方法
如果设备已配对/绑定，则Android 8.0和8.1上的DFU仅会失败。
如果我取消配对/删除设备的绑定，则DFU更新适用于Android 8.0和8.1。

我认为可能有必要在Android> = 8上添加第三步到DFU序列（SDK> = 26）
1）选择设备后，检查是否已粘合。如果粘合，则去除粘合
2）重新连接到未配对的设备并发送Start DFU命令
3）重新连接到设备并启动DFU

目前我的解决方法是建议用户在配对/绑定时取消配对设备。我的应用程序不需要配对工作，这对我来说是一个临时解决方案。

另外，我必须将PRN设置为4才能使其正常工作。
默认PRN设置为12，但在LegacyDfuImpl.java第455行中它表示
```//已经测试过，PRN = 10可能是支持的最高值.``
这与默认值不一致。
也许在Android 8.0及更高版本上，PRN默认值应设置为4，在较旧的SDK上，默认值应设置为10而不是12。

johncheung233, 2020-01-03:
> 使用1.8.1的库在不分手机上进行升级经常出现以下错误
> 1、DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED
> 2、DFU DEVICE DISCONNECT
> 请问是什么原因呢？
> 加急！！！
> 谢谢！！！

我用这个SDK以前可以用，最近又不行了，跟你同样的问题，请问解决了吗

johncheung233, 2020-01-13:
> > 使用1.8.1的库在不分手机上进行升级经常出现以下错误
> > 1、DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED
> > 2、DFU DEVICE DISCONNECT
> > 请问是什么原因呢？
> > 加急！！！
> > 谢谢！！！
> 
> 我用这个SDK以前可以用，最近又不行了，跟你同样的问题，请问解决了吗

同感，之前我也是行的，现在又不行，而且是偶发性

qx-mark-wu, 2020-01-13:
这个我现在也没有很大改善，感觉应该是他这个SDK的兼容问题。
这个升级的问题一直被客户投诉，解决办法一直都是让客户重复尝试或换手机尝试。
我也并没有找到合适的解决办法

| |
吴福亮
|
|
m15919474028_1@163.com
|
签名由网易邮箱大师定制

philips77, 2020-06-24:
I have no idea what's above... closing due to inactivity.

closed 2020-06-24 by philips77