nordicsemi/Android-DFU-Library #175: Executing object failed

zhouzhijin0908, 2019-03-28:
private void writeExecute() throws DfuException, DeviceDisconnectedException, UploadAbortedException, UnknownResponseException, RemoteDfuException {
		if (!mConnected)
			throw new DeviceDisconnectedException("Unable to read Checksum: device disconnected");

		writeOpCode(mControlPointCharacteristic, OP_CODE_EXECUTE);

		final byte[] response = readNotificationResponse();
		final int status = getStatusCode(response, OP_CODE_EXECUTE_KEY);
		if (status == SecureDfuError.EXTENDED_ERROR)
			throw new RemoteDfuExtendedErrorException("Executing object failed", response[3]);
		if (status != DFU_STATUS_SUCCESS)
			throw new RemoteDfuException("Executing object failed", status);
	}

I want to disable the function"breakpoint dfu", how can i do ?

qx-mark-wu, 2019-04-01:
兄弟，你调试的时候有没有遇到不分Android手机升级出现这两种失败的情况吗？
是否有可行的解决方法指导一下呢？
多谢了！
1、1、DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED
2、DfuBaseService: Connection state change error: 133 newState: 0；
DfuBaseService: Device not reachable. Check if the device with address E2:17:5B:71:F8:63 is in range, is advertising and is connectable

johncheung233, 2020-01-13:
> 兄弟，你调试的时候有没有遇到不分Android手机升级出现这两种失败的情况吗？
> 是否有可行的解决方法指导一下呢？
> 多谢了！
> 1、1、DfuImpl: Device returned error after sending file (error 6): OPERATION FAILED
> 2、DfuBaseService: Connection state change error: 133 [...]

我也遇到了第一个问题，请问解决了吗

closed 2020-06-24 by philips77