nordicsemi/Android-DFU-Library #176: Successive dfu updates not working

hyyu, 2019-04-01:
Hi,

I'm using the Nordic DFU lib in my app to make DFU updates on a connected device.

I have 2 updates to do: 1 for the Bootloader, 1 for the firmware and I do the updates 1 after another.

For the Bootloader, I know that when the version of the Bootloader on the device is already up to date, when the update is launched, a FW_VERSION_FAILURE is sent.

Once the Bootloader update is done or gets the error that I mentioned above, I launch the second update (for the Firmware).

My problem is that when the device is already in DfuTarg mode, the Bootloader starts and gets the error mentioned above as predicted, but when the firmware update starts, it gets the same error.

I tried to launch only the firmware update on a device already in DfuTarg mode, and the update worked.

My question is: Is starting a new DFU update from a `DfuProgressListener` something that isn't designed for in the DFU lib? I'm putting my code below to show how I implemented it.

Code to start the Bootloader update:

```
private fun startBootloaderDFUProcess() {
		Log.i(TAG, "Starting Bootloader DFU update")
		DfuServiceListenerHelper.registerProgressListener(context!!, dfuBootloaderProgressListener)

		val starter = DfuServiceInitiator(deviceMacAddress!!)
			.setDeviceName(deviceName!!)
			.setKeepBond(true)

		starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)
		starter.setZip(bootloaderPathName!!)
		starter.start(context!!, DfuService::class.java)

		if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
			DfuServiceInitiator.createDfuNotificationChannel(context!!)
		}

	}
```

Code for the `DfuProgressListener` for the Bootloader:

```
	private val dfuBootloaderProgressListener = object : DfuProgressListenerAdapter() {
		override fun onProgressChanged(deviceAddress: String, percent: Int, speed: Float, avgSpeed: Float, currentPart: Int, partsTotal: Int) {
			super.onProgressChanged(deviceAddress, percent, speed, avgSpeed, currentPart, partsTotal)
			mPercent = percent / 2
			dfu_loading_bar.progress = mPercent
			tv_update_percent.text = getString(R.string.dfu_percent, mPercent)
		}

		override fun onError(deviceAddress: String, error: Int, errorType: Int, message: String?) {
			super.onError(deviceAddress, error, errorType, message)
			Log.v(TAG, "onError : $error type : $errorType message : $message")
			when (error) {
				BOOTLOADER_FW_VERSION_FAILURE -> {
					if (PrefEnovap.getSavedDeviceName()?.toLowerCase() == DFU_TARG_NAME) {
						startFirmwareDFUProcess()
					} else {
						viewModel.launchScanForDfuTarg()
					}
				}
				else -> {
					activity?.finish()
				}
			}
		}

		override fun onDfuAborted(deviceAddress: String) {
			super.onDfuAborted(deviceAddress)
			Log.v(TAG, "dfu aborted")
		}

		override fun onFirmwareValidating(deviceAddress: String) {
			super.onFirmwareValidating(deviceAddress)
			Log.v(TAG, "onFirmwareValidating")
		}

		override fun onDeviceDisconnected(deviceAddress: String) {
			super.onDeviceDisconnected(deviceAddress)
			Log.v(TAG, "onDeviceDisconnected")
		}

		override fun onDeviceConnected(deviceAddress: String) {
			super.onDeviceConnected(deviceAddress)
			Log.v(TAG, "onDeviceConnected")
		}

		override fun onDfuProcessStarting(deviceAddress: String) {
			super.onDfuProcessStarting(deviceAddress)
			Log.v(TAG, "onDfuProcessStarting")
			mPercent = 0
			dfu_loading_bar.progress = mPercent
			tv_update_percent.text = getString(R.string.dfu_percent, mPercent)
		}

		override fun onDfuProcessStarted(deviceAddress: String) {
			super.onDfuProcessStarted(deviceAddress)
			Log.v(TAG, "onDfuProcessStarted")
		}

		override fun onDeviceConnecting(deviceAddress: String) {
			super.onDeviceConnecting(deviceAddress)
			Log.v(TAG, "onDeviceConnecting")
		}

		override fun onDfuCompleted(deviceAddress: String) {
			super.onDfuCompleted(deviceAddress)
			DfuServiceListenerHelper.unregisterProgressListener(context!!, this)
			startFirmwareDFUProcess()
		}
	}
```

Code to start the Firmware update:

```
	private fun startFirmwareDFUProcess() {
		Log.i(TAG, "Starting Firmware DFU update")
		DfuServiceListenerHelper.registerProgressListener(context!!, dfuFirmwareProgressListener)

		val starter = DfuServiceInitiator(deviceMacAddress!!)
			.setDeviceName(deviceName!!)
			.setKeepBond(true)

		starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)
		starter.setZip(firmwarePathName!!)
		starter.start(context!!, DfuService::class.java)

		if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
			DfuServiceInitiator.createDfuNotificationChannel(context!!)
		}
	}
```

Code for the `DfuProgressListener` for the Bootloader:
```
	private val dfuFirmwareProgressListener = object : DfuProgressListenerAdapter() {
		override fun onProgressChanged(deviceAddress: String, percent: Int, speed: Float, avgSpeed: Float, currentPart: Int, partsTotal: Int) {
			super.onProgressChanged(deviceAddress, percent, speed, avgSpeed, currentPart, partsTotal)
			mPercent = FIRMWARE_START_PERCENT + percent / 2
			dfu_loading_bar.progress = mPercent
			tv_update_percent.text = getString(R.string.dfu_percent, mPercent)
		}

		override fun onError(deviceAddress: String, error: Int, errorType: Int, message: String?) {
			super.onError(deviceAddress, error, errorType, message)
			Log.v(TAG, "onError : $error type : $errorType message : $message")
//			 activity?.finish()
		}

		override fun onDfuAborted(deviceAddress: String) {
			super.onDfuAborted(deviceAddress)
			Log.v(TAG, "dfu aborted")
		}

		override fun onFirmwareValidating(deviceAddress: String) {
			super.onFirmwareValidating(deviceAddress)
			Log.v(TAG, "onFirmwareValidating")
		}

		override fun onDeviceDisconnected(deviceAddress: String) {
			super.onDeviceDisconnected(deviceAddress)
			Log.v(TAG, "onDeviceDisconnected")
		}

		override fun onDeviceConnected(deviceAddress: String) {
			super.onDeviceConnected(deviceAddress)
			Log.v(TAG, "onDeviceConnected")
		}

		override fun onDfuProcessStarting(deviceAddress: String) {
			super.onDfuProcessStarting(deviceAddress)
			Log.v(TAG, "onDfuProcessStarting")
			dfu_loading_bar.progress = FIRMWARE_START_PERCENT
			tv_update_percent.text = getString(R.string.dfu_percent,
				FIRMWARE_START_PERCENT
			)
		}

		override fun onDfuProcessStarted(deviceAddress: String) {
			super.onDfuProcessStarted(deviceAddress)
			Log.v(TAG, "onDfuProcessStarted")
		}

		override fun onDeviceConnecting(deviceAddress: String) {
			super.onDeviceConnecting(deviceAddress)
			Log.v(TAG, "onDeviceConnecting")
		}

		override fun onDfuCompleted(deviceAddress: String) {
			super.onDfuCompleted(deviceAddress)
			val f = DFUUpToDateFragment.newInstance()

			DfuServiceListenerHelper.unregisterProgressListener(context!!, this)
			activity?.runOnUiThread {
				enovap_button_dfu.revertAnimation()
				frameListener.onStepFinished(f, DFUUpToDateFragment.TAG)
			}
		}
	}
```

philips77, 2019-04-08:
Starting new DFU from the listener should work, as it will just start the service again after the currently running one finishes.
But, I thing starting from 1.9.0, you may also put your bootloader and app upadate in one ZIP file and try updating both. When the bootloader fail to update, the lib will automatically try updating an app, without the need for scanning for DFU targ, etc.
Also, could you paste logs form LogCat here?

hyyu, 2019-04-08:
Hi,

I've found another way to perform both updates and it works, so I don't have the logcat for the problem anymore. The problem was solved by not holding the BLE connection while updating with BLE instead of waiting from the DFUProcess to disconnect the device automatically and it works much better. However I don't understand by "you may also put your bootloader and app upadate in one ZIP file and try updating both". Do you mean just launch 1 time the DFUProcess with the ZIP that has both bootloader and firmware updates?

philips77, 2019-04-08:
Yes. You may create a ZIP file with 2 bin files, 2 dat files and a json file using nrf util on GitHub. The lib will first try to send system component (softdevice and/or bootloader) and then app.

hyyu, 2019-04-08:
Ok, thanks, I'll try it then!

closed 2020-06-24 by philips77

shubhamsinghshubham777, 2023-07-31:
> The problem was solved by not holding the BLE connection while updating with BLE instead of waiting from the DFUProcess to disconnect the device automatically and it works much better.

@hyyu Could you please share the code for this?