nordicsemi/Android-DFU-Library #177: Error when tries to embbed DFU Library inside another library,

KelvinEC, 2019-04-02:
When i try to use DFU library and package it as a .aar, adding the recommended line to proguard result in:

`error: cannot access DfuBaseService
class file for no.nordicsemi.android.dfu.DfuBaseService not found`

I can use it normally before build the .aar, but when do it and use on project, results on that.

i'm using library 1.9.0 version with Gradle 3.3.2 and compileSDKVersion 27.

philips77, 2019-04-08:
Did you add the dependency as `api`, not `compile`? Perhaps the inner lib is hidden from the app and available only to the another library.
I'm not sure if it also applies when you use aar files, not just jcenter dependencies.

KelvinEC, 2019-05-21:
The solution is to add the MainLibrary as Api and DFU library as implementation on App.

closed 2019-05-21 by KelvinEC