nordicsemi/Android-DFU-Library #182: KitKat Descriptor write error (133)

NevilleKing, 2019-04-25:
I am testing an app which performs buttonless bonded DFU with a KitKat phone (Nexus 5). I get the following error when attempting an update:

````
I/DfuBaseService: Services discovered
I/DfuImpl: Reading Service Changed CCCD value...
D/BluetoothGatt: readDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
D/BluetoothGatt: onDescriptorRead() - Device=C1:B7:36:18:E3:44 UUID=00002a05-0000-1000-8000-00805f9b34fb
I/DfuImpl: Buttonless service with bond sharing found -> SDK 14 or newer
I/DfuImpl: Enabling indications...
D/BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90004-f315-4f60-9fb8-838830daea50 enable: true
D/BluetoothGatt: writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
D/BluetoothGatt: onDescriptorWrite() - Device=C1:B7:36:18:E3:44 UUID=8ec90004-f315-4f60-9fb8-838830daea50
E/DfuImpl: Descriptor write error: 133
````

The device correctly enters the bootloader and the issue seems to occur after connecting for the second time and attempting to write a descriptor. I'm using v1.9.0 of the DFU library and SDK 14.2 on the device.

The issue only occurs on KitKat. I've tested with multiple Android phones running different versions and they seem to work fine with the same update package. I've tested the same package with nRF connect and the update is applied successfully. I've also tested with nRF toolbox and the same issue is present.

Any ideas on what the issue could be? Any help would be appreciated.

philips77, 2019-04-26:
Hi @NevilleKing,
It seems like Nexus 5 or KitKat fails to refresh services after receiving Service Changed indication. The process should go like this:
1. You are connected to a bonded device.
2. The DFU library sends Jump to bootloader command to Buttonless service. Before that it ensures that Service Changed indications are enabled.
3. The device reboots and starts to advertise with the same MAC address, directly to your phone (?).
4. The phone reconnects, re-initializes encryption and receives Service Changed indication.
5. Android system should invalidate the cached services and start "internal" service discovery.
6. The library, after reconnecting, waits for 1600 ms and calls `gatt.discoverServices()`, hoping that the caches were invalidated and the internal service discovery at least has started.
7. It should receive a new set of services after a while.

However, in your case, based on the fact that you wrote

> The device correctly enters the bootloader and the issue seems to occur after connecting for the second time

and 
```
I/DfuImpl: Buttonless service with bond sharing found -> SDK 14 or newer
```
seems like the cache was not invalidated and the library for the second time got the same services. If found Buttonless service among them and tried to enable indications again, but the handle number is wrong now, so writing descriptor has failed.

To be honest, I thought that invalidating cache was working on all Android versions, but seems like it doesn't, at least not on Nexus 5. I will try to replicate your issue, as I have that phone on my desk. 

As a quick fix for you I suggest adding the DFU library as a module (instruction how to is in the Readme.md file) and change `force` to `true` (or adding `|| Build.VERSION_SDK == Build.VERSION_CODES.KITKAT`) in this line: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/ec14c8c522bebe801a9a4c3dfbbeb1f53262c03f/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1614 
The current implementation in 1.9.0 and before tries to not force cache refresh on bonded devices for 2 reasons:
1. I thought SC indications were working on all Android versions
2. Calling `refresh()` is a hack, as this method is not in public API.

NevilleKing, 2019-04-26:
Hi @philips77,

Thanks for your quick and detailed reply. I've imported the DFU library as a module as you suggested, and after implementing the fix the DFU update is applied successfully on the Nexus.

Just another quick question. In my current implementation, there could by multiple DFU updates that need to be applied sequentially. When first working on this feature, I had to implement the hidden `refresh` method in between updates and at the end otherwise the phone would only show DFU related services after the update was applied (even in nRF connect). I would rather not use this method and I assume the service changed characteristic should handle this instead. Is there anything extra I should be doing or anything I could point the firmware team to? They inform me that the service changed characteristic is enabled on their end.

Thanks for your help.

philips77, 2019-04-26:
If the Service Changed feature doesn't work on Nexus, you need to call `refresh()` also at the end, as there's no third way to refresh devices. You may also remove the bond and bind again, which will also do the trick, buy you're losing security and the `removeBond` method is also hidden. Or, you could start using a different Mac address after the update, but this also requires new bonding and removing old. From all those, calling refresh seems to be the best choice.

NevilleKing, 2019-04-29:
Yeah I agree, that does seem like the best option. 

Now I reread my previous comment, I realise that I forgot to mention a crucial detail. I'm using the `refresh` method on every version of Android, as the services don't seem to switch back correctly at the end of a DFU update. Checking with nRF connect the services seem to be stuck with the bootloader services, and not switching back to the firmware application. I assume there is something else I should be doing in the Android app or something I should pass onto the firmware team, as I don't see many of these issues mentioned around on the internet. The DFU update is applied fine, but without the force refresh a second update will fail due to some kind of characteristic read/write issue.

[image]

Above are the services shown after DFU on nRF connect. Is there something else we should be doing? I assume the `refresh` method should not be required across all Android versions. As you've mentioned it's ideal to not be using the `refresh` method at all. I'm at a bit of a loss now, so any help is appreciated.

philips77, 2019-04-29:
Make sure the option "Keep bond information* is enabled in Settings/Dfu. In the lib you also need to enable it in the initiator, as far as I remember.
To refresh devices in nRF Connect other menu in top right corner and select Refresh services (or something like that).

NevilleKing, 2019-05-03:
That setting in nRF connect is turned on, and I am using the `setKeepBond(true)` within the initiator. What I'm trying to ask is should the Android OS automatically refresh the services after a DFU update has occurred when using this library? If I don't manually refresh the services using the hidden `refresh` method after an update has been applied, Android seems to hold onto the bootloader services in it's internal cache. This obviously caused issues because I'm expecting the services to switch back to the firmware app ones.

This is obviously not a high priority issue right now as DFU is working on all devices and Android OS versions that I have tested on. I just assume that using the `refresh` method is not a great solution and was hoping to move away from that in the future. Let me know if I'm missing anything.

philips77, 2019-05-04:
If you're using Buttonless service with bond, the dfu library relies on Service Changed indication when switching to bootloader mode and after dfu is complete. That means, that it doesn't call `refresh`.
However, as I wrote above, SC indication is sent a moment after you receive `onConnectionStateChange`. That means, that after the dfu is complete, in your app, when you connect again to the device, you should wait a bit (dfu lib waits for 1600ms), before calling `gatt.discoverServices()`, otherwise you'll get cached services that moment later will get invalidated, but your app won't be aware of it. At least that's what happens when SC is received and handled correctly by the system. The API is very bad in regard of cache refreshing. If the delay doesn't help, just refresh on your own.

NevilleKing, 2019-05-09:
The SC wait seems like the best way to go.

It’s very frustrating that the Android BLE stack is so finicky, requiring the use of hidden methods to get basic functionality working. It makes it a bit of a nightmare to work with it, especially on the lower Android versions. Thanks for all your help though. You’ve made it all a lot easier 😃

closed 2020-07-03 by philips77