nordicsemi/Android-DFU-Library #183: dfubaseservice: connected. dfu service not found

M-k-03, 2019-04-29:
Hi Team,

I am trying to connect and update my Bluetooth Device using this library. 

when i checked through logs below mentioned error happens and device getting disconnected.
[DfuBaseService.java]
dfubaseservice: connected. dfu service not found.

I would like to know why this error is happening.  Kindly Guide me

philips77, 2019-04-29:
Hello,
Do you have the DFU service on your device? Try connecting to it using nRF Connect for Android and see if the Legacy or Secure DFU service are there.

M-k-03, 2019-05-08:
[image]

Hi sorry for delayed response. I connected with nRf connect for Android.. Legacy..

M-k-03, 2019-05-11:
[image]
[image]

Hi After Connecting using nRF Connect with my band mentioned above service listed. :( still i cant find Legacy or Secure DFU

philips77, 2019-05-12:
Well, that's because your device doesn't support Dfu. Please, go to the documentation on our website to find out how to make it up and running.

closed 2019-05-12 by philips77

M-k-03, 2019-05-13:
Hi My device does not support dfu now as you mentioned. 
I have few questions regarding this.

1: Is it possible to make my device dfu compatible. If yes please provide the link so that I can proceed further.