nordicsemi/Android-DFU-Library #187: Kotlin Support

MMooreGitHub, 2019-05-07:
Good Evening,

Google just announced in Google I/O 2019 that Android will now be a[ Kotlin-first platform](https://android-developers.googleblog.com/2019/05/google-io-2019-empowering-developers-to-build-experiences-on-Android-Play.html?utm_source=feedburner&utm_medium=feed&utm_campaign=Feed:+blogspot/hsDu+(Android+Developers+Blog)&m=1).

Are there any plans to port the Android-DFU-Library to Kotlin?

philips77, 2019-05-08:
Hi, of course there are plans. But right now we have no resources and this seems to be low priority for now as the lib is compatible with kotlin as is.
But if like to move everything to kotlin, like we switched to Swift on iOS. Just takes time...

MMooreGitHub, 2019-05-10:
Ok, thanks, Philip.

closed 2020-06-11 by philips77