nordicsemi/Android-DFU-Library #188: DFU DEVICE DISCONNNECTED

HarlanSong, 2019-05-15:
After the bracelet enters upgrade mode, execute the following code：
`final DfuServiceInitiator starter = new DfuServiceInitiator(newMacAddress)
                .setDisableNotification(true)
                .setDeviceName("DfuTarg");
        starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
        Uri mFileStreamUri = Uri.fromFile(new File(filePath));
        starter.setZip(mFileStreamUri, filePath);
        starter.start(this, NordicDfuService.class);`

Log:
`I/DfuBaseService: DFU service created. Version: 1.8.1
I/DfuBaseService: Connecting to the device...
E/DfuBaseService: Connection state change error: 133 newState: 0
E/DfuBaseService: Device not reachable. Check if the device with address DE:D7:2A:7C:BE:A9 is in range, is advertising and is connectable
I/DfuBaseService: Refreshing result: true
I/DfuBaseService: Cleaning up...
I/DfuBaseService: DFU service destroyed`

PS:Galaxy C9 Pro

philips77, 2019-05-15:
Do you have a custom method for switching the device to bootloader mode? If not, this will be handled by the library automatically.
If you do, ensure that the address is correct. The device may advertise with address incremented by 1 in DFU mode to avoid caching issues. It will not be the same as your normal device address, unless you use bonding and the Buttonless Service that supports bond (this was not implemented in SDK 12.x, and based on how you call `setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled` I assume you use SDK 12.x).

HarlanSong, 2019-05-16:
We're done. Thank you

closed 2019-05-16 by philips77