nordicsemi/Android-DFU-Library #189: Possibility to update several devices at same time

mcolunas, 2019-07-08:
I have tested to use diferent DfuBaseService extending classes and seems to work, however it would be possible to do that only registering only one service on Android Manifest?
This approach will accelerate the process instead of updating only one device at the same time?

philips77, 2019-07-08:
We don't have plans to implement such feature for dfu right now. Fir Zephyr based devices, the mcu mgr library handles dfu on a lower level, without the service, so what you described is possible there. However, for dfu we were initially focused on ease of use, therefore the update is handled by a predefined service with just few lines required to start the process. Updating multiple devices is not a common use case.
Thanks for testing dfu with multiple services in Android manifest. I was hoping it would work.

mcolunas, 2019-08-12:
Thanks for your help

closed 2019-08-12 by mcolunas