nordicsemi/Android-DFU-Library #192: RemoteServiceException: Bad notification for startForeground crash on >= Android 8.1

krtkv, 2019-07-19:
There is an issue related to the updates in Android system,
Since Android 8.1 you need to create own notification channel for foreground service

So in DfuBaseService in startForeground method you need to add something like this:

`if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
			NotificationChannel chan = new NotificationChannel(NOTIFICATION_CHANNEL_DFU, "Update service",NotificationManager.IMPORTANCE_NONE);
			chan.setLightColor(Color.BLUE);
			chan.setLockscreenVisibility(Notification.VISIBILITY_PRIVATE);
			NotificationManager service = (NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
			service.createNotificationChannel(chan);
		}`

philips77, 2019-07-20:
Yes, you'll find this code in DfuServiceIntiator at the very bottom.

daminhtung, 2019-08-06:
I got the same on android 9.0
 ```
android.app.RemoteServiceException: Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=dfu pri=-1 contentView=null vibrate=null sound=null defaults=0x0 flags=0x42 color=0xff888888 vis=PRIVATE)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1745)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6718)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
```

chris-guidry, 2019-08-13:
@philips77 I am experiencing the same issue, which crashes our app whenever we try to run a DFU.

Here is the error:
```
FATAL EXCEPTION: main
android.app.RemoteServiceException: Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=dfu pri=-1 contentView=null vibrate=null sound=null defaults=0x0 flags=0x42 color=0xff888888 number=0 vis=PRIVATE semFlags=0x0 semPriority=0 semMissedCount=0)
 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1906)
 	at android.os.Handler.dispatchMessage(Handler.java:106)
 	at android.os.Looper.loop(Looper.java:164)
 	at android.app.ActivityThread.main(ActivityThread.java:7000)
 	at java.lang.reflect.Method.invoke(Native Method)
 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)
```

Based on [this Stack Overflow post](https://stackoverflow.com/a/51281297/648560), and the research by @krtkv, it seems the function startForeground in DfuBaseService.java should be updated to include the NotificationChannel. I could work on changing it, but I don't have a great way to test the changes before creating a pull request. Is that something you are able to change/test/push/version? If so, it would be much appreciated!

chris-guidry, 2019-08-13:
@philips77 I was just rereading [your previous comment](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/192#issuecomment-513493539). Are you saying this issue should be resolved in the latest version? I am using version 1.8.0, but could try 1.9.0 if you think it's resolved in that version. I am assuming the notifications are necessary and it seems like the NotificationChannel would be required in order for the notifications to work.

closed 2020-02-12 by philips77

AndroidDeveloperLB, 2022-02-23:
It was finally fixed on Android 12:
https://issuetracker.google.com/issues/192032398#comment6

SB2318, 2022-05-28:
I am sorry Sir. I mentioned this issue without your permission.
But my intentions are not bad. I'm a learner, I wanted to preserve this solution in my repository. If you have an objection, then I will remove the comment.