nordicsemi/Android-DFU-Library #193: Fatal Exception: android.app.RemoteServiceException: Context.startForegroundService() did not then call Service.startForeground()

dlukashev, 2019-07-22:
In some cases, **startForegroundService()** is not called within 5 seconds. I suppose disk read (shared preferences) can cause such delay or something else. This is device specific according to the fabric logs.

Library version: 1.9.0
Device: Xiaomi A1, Android 9.0

`Fatal Exception: android.app.RemoteServiceException: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{fb002d0 u0 com.atmotube.app/.ble.dfu.DfuService}
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1747)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:193)
       at android.app.ActivityThread.main(ActivityThread.java:6762)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)`

**DfuService** is a simple class.

`public class DfuService extends DfuBaseService {

    @Override
    protected Class<? extends Activity> getNotificationTarget() {
        return UpdateDeviceActivity.class;
    }

    @Override
    protected void refreshDeviceCache(BluetoothGatt gatt, boolean force) {
        super.refreshDeviceCache(gatt, true);
    }

    @Override
    protected boolean isDebug() {
        return BuildConfig.DEBUG;
    }

    @Override
    protected void attachBaseContext(Context base) {
        super.attachBaseContext(LocaleManager.setLocale(base));
    }
}`

daminhtung, 2019-08-06:
I got this error on Android 9.0 with the same `**DfuService**`

    ```
android.app.RemoteServiceException: Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=dfu pri=-1 contentView=null vibrate=null sound=null defaults=0x0 flags=0x42 color=0xff888888 vis=PRIVATE)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1745)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6718)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
```

horsejockey, 2019-12-11:
I am seeing this same behavior on a Google Pixel 3 running Android 10. It happens about 60% of the time.

philips77, 2019-12-12:
The URI you give to the library, is it local one or to remote?

philips77, 2020-02-12:
I will move `startForeground(...)` call a bit higher in 1.9.2.

philips77, 2020-02-12:
Please, check with the 1.9.2 or 1.10 version.

closed 2020-06-11 by philips77