nordicsemi/Android-DFU-Library #195: Foreground service getting killed and causing DFU abort

smithc42, 2019-07-25:
**Device:** Samsung Galaxy A5 (8.0 Oreo)
**DFU library version** 1.9.0
**minSdkVersion** 23
**targetSdkVersion** 28
**compileSdkVersion** 28

**Issue**
Foreground service getting killed and causing DFU abort.

**Steps:**

1. Set notification channel + start update (code below)
2. Wait for update to begin and kill app via swipe when progress reaches 50%

**Expected results:**
Update continues until completion

**Actual results:**
Service is killed and update is aborted:
`DfuBaseService Upload aborted`

**Notes**

- Notification appears and updates with percentage, until the app is killed
- When I attach the debugger, `startForegroundService` is called when the update begins

**Code**
```DfuServiceInitiator.createDfuNotificationChannel(context)
        DfuServiceInitiator(macAddress())
            .setDeviceName(bleDevice.name)
            .setKeepBond(true)
            .setForceDfu(false)
            .setForeground(true)
            .setZip(updateFilePath)
            .setNumberOfRetries(3)
            .start(context, DFUService::class.java)
```

**Manifest**
```
<uses-permission android:name="android.permission.FOREGROUND_SERVICE" />
...
<service
            android:name="com.example.bluetooth.update.DFUService"
            android:exported="true"
            android:label="@string/dfu_service_title">
            <intent-filter>
                <action android:name="no.nordicsemi.android.action.DFU_UPLOAD" />

                <category android:name="android.intent.category.DEFAULT" />
            </intent-filter>
        </service>

```

philips77, 2019-07-25:
The behavior for swiping an app from Recents want defined until, I think, Android 9. Still now it's just Google recommendation how OEMs should preform. There are phones that force kill the app, other that just kill it more gentle. Any way, a service running, whether it's background or promoted to foreground, will receive onTaskRemoved() callback. Dfu service is aborting process there and stopping dfu, as far as I remember. But even without it, it would be stopped and restarted again with the latest Intent (if STICKY mode was used). One could experiment with creating a service in another process (check out process attribute in Service tag in Android Manifest), but then it's more difficult to pass data to it, you have to use Messages.
Either way, when user is swiping the app away, their intention is to get rid of it.

smithc42, 2019-07-26:
Thanks for the reply. 

I managed to recover by connecting to the DFU mac address (provided by onDeviceConnecting call back in DfuProgressListenerAdapter) and restarting the update process when the user re-opens the app.

closed 2019-07-28 by philips77