nordicsemi/Android-DFU-Library #199: Is "android.permission.READ_EXTERNAL_STORAGE" needed for uploading a DFU file?

Jiqing1107, 2019-08-13:
Initially, when we developed our project based on your DFU library, we didn't add the READ_EXTERNAL_STORAGE permission and didn't find any errors.  However, when our customer tested our project, they reported an issue:

When start DFU, application did not read zip file.
- "Upload failed: DFUFILENOTFOUND" are logging.

The engineer of our customer's side solved the issue by adding "<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" in "app/src/main/AndroidManifest.xml". However, the testers from our side couldn't reproduce their issue even without this permission. Could u please help explain? Why the permission is required occasionally? Thanks!

philips77, 2019-08-13:
It depends how do you obtain the file. The library may work just with URIs, where one time permission is granted by the file browser app, cb or even with raw resources. On which version do you get this error?

Jiqing1107, 2019-08-13:
Got it...so the file browser app of our phones should probably have the one time permission while our customer's should not. Our customer is from Japan, they tested on GalaxyFeel (SC-04J) samsung android  7.0.0. I think maybe their phones made some local changes. Anyway, they solved the issue by adding the READ_EXTERNAL_STORAGE permission. Thanks!

closed 2020-06-11 by philips77