nordicsemi/Android-DFU-Library #202: huwei mate20 crash by  invalid channel for service notification: Notification(channel=dfu pri=-1 contentView=null vibrate=null sound=null defaults=0x0 flags=0x42 color=0xff888888 vis=PRIVATE)

GarenGuogf, 2019-09-19:
error is ：
android.app.RemoteServiceException

Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=dfu pri=-1 contentView=null vibrate=null sound=null defaults=0x0 flags=0x42 color=0xff888888 vis=PRIVATE)

MarkWuFuLiang, 2019-09-19:
I also encountered the same problem, has the landlord solved it? The solution is?
Phone Number:Google Pixel 2 XL

philips77, 2019-09-19:
What phone do you experience it on? Did you try to setting up the channel on your own with different settings?

MarkWuFuLiang, 2019-09-20:
> What phone do you experience it on? Did you try to setting up the channel on your own with different settings?

Google Pixel 2 XL
new DfuServiceInitiator(mDeviceAddress).
                        setDisableNotification(true).setZip(
                        R.raw.dfu).
                        start((getBaseContext()), DfuService.class);
crash by invalid channel for service notification: Notification(channel=dfu pri=-1 contentView=null vibrate=null sound=null defaults=0x0 flags=0x42 color=0xff888888 vis=PRIVATE) #202

The DfuService.class and NotificationActivity.class is by yours demo.

I don't know how to setting up the dofferent channel!
Can you show me?

philips77, 2019-09-20:
Starting from Android Oreo onward you have to create a Notification Channel, also for foreground service notifications.
Have a look here: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/d92a57d856921e23201aa5c9ac688b540fba66ed/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L828-L842
You have to call this method of the initiator in your Application or main Activity.

Read more about notifications here: https://developer.android.com/guide/topics/ui/notifiers/notifications and https://developer.android.com/training/notify-user/channels and https://developer.android.com/guide/components/services.html#Foreground

qx-mark-wu, 2019-09-20:
> Starting from Android Oreo onward you have to create a Notification Channel, also for foreground service notifications.
> Have a look here: [...]

Thank you very much!
I have solved this trouble!
But I am now experiencing new problems.
Is that after I successfully executed the Start method.
Prompt upgrade failure message DFU CHARACTERISTICS NOT FOUND
DfuProgressListener has three steps back:
1.onDeviceDisconnecting
2.onDeviceDisconnected
3.onError

philips77, 2019-09-20:
You may check, using nRF Connect for Android, whether your device has the DFU Service (Legacy or Secure DFU) and required characteristics inside. Try refreshing the cache (after connecting, go to menu -> Refresh device cache).

qx-mark-wu, 2019-09-21:
Thank you very much
Can the 1.8.1 SDK be DFU upgraded in the paired state?
I found that I did not get the DFU CHARACTERISTICS NOT FOUND error when I upgraded in the pairing state, but the reliability of the upgrade is still verified.

philips77, 2019-09-21:
Are you using the Nordic SDK 14 or newer? In that case it contains dfu bootloader and buttonless service for bonded devices, which is supported by this (and iOS) library.
To avoid caching problems make sure you have Service Changed characteristic enabled on both app and bootloader modes. This is also a requirement when you use bonding.

qx-mark-wu, 2019-09-21:
I use the SDK is 10.0

| |
吴福亮
邮箱：15919474028@163.com
|

签名由 网易邮箱大师 定制

On 09/21/2019 13:53, Aleksander Nowakowski wrote:

Are you using the Nordic SDK 14 or newer? In that case it contains dfu bootloader and buttonless service for bonded devices, which is supported by this (and iOS) library.
To avoid caching problems make sure you have Service Changed characteristic enabled on both app and bootloader modes. This is also a requirement when you use bonding.

closed 2019-11-13 by philips77

GarenGuogf, 2019-11-13:
tks

| |
GAREN
邮箱：garenguo@163.com
|

Signature is customized by Netease Mail Master

On 11/13/2019 22:07, Aleksander Nowakowski wrote:

Closed #202.