nordicsemi/Android-DFU-Library #204: Custom title, message and icon for foreground notification?

cheeyi, 2019-10-15:
Hello, is there a way to specify a custom title, message and icon for the foreground notification that is shown when a firmware update is in process? Thank you.

philips77, 2019-10-17:
There's a method in BaseDfuService that you can override in you service to modify the notification.

cheeyi, 2019-10-17:
> There's a method in BaseDfuService that you can override in you service to modify the notification.

@philips77 Is `updateProgressNotification()` the method you were referring to? Or is it `updateForegroundNotification()`? Thanks in advance.

philips77, 2019-10-17:
Well... The names are self explanatory. Remember that you may disable the default progress notifications, or start dfu in background, not foreground. Appropriate methods are in the initiator.

cheeyi, 2019-10-18:
Thanks @philips77 for your kind help and explanation! Closing this issue.

closed 2019-10-18 by cheeyi