nordicsemi/Android-DFU-Library #205: OPERATION FAILED with ErrorCode 522

HansKrywaa, 2019-10-28:
Hey,

i'm creating a Cordova App and use this Plugin, that is a Wrapper for your Library: https://www.npmjs.com/package/cordova-plugin-nordic-update

When i try to update the Firmware this Steps occur:

- deviceConnected
- dfuProcessStarting
- deviceDisconnecting
- deviceDisconnected
- error: 522, errorType: 3, message: "OPERATION FAILED"

What is ErrorCode 522, how can i solve it?

philips77, 2019-10-28:
Hello,
Try updating you device with the same ZIP file using nRF Connect for Android. You'll get logs there. If you still get this error, please post the logs here. Error 522 is an error 10 (Operation Failed) returned from the target device.

HansKrywaa, 2019-10-28:
Hey @philips77 ,

i did, here is the log:
```
nRF Connect, 2019-10-28
DfuTarg (E6:73:8D:C5:CC:49)
V	14:49:33.737	Connecting to E6:73:8D:C5:CC:49...
D	14:49:33.737	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	14:49:34.268	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	14:49:34.268	Connected to E6:73:8D:C5:CC:49
D	14:49:34.300	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	14:49:34.319	Discovering services...
D	14:49:34.319	gatt.discoverServices()
I	14:49:35.016	Connection parameters updated (interval: 16.25ms, latency: 0, timeout: 4000ms)
I	14:49:35.131	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	14:49:35.151	[Callback] Services discovered with status: 0
I	14:49:35.151	Services discovered
V	14:49:35.162	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
Secure DFU Service (0xFE59)
- DFU Packet [WNR] (8ec90002-f315-4f60-9fb8-838830daea50)
- DFU Control Point [N W] (8ec90001-f315-4f60-9fb8-838830daea50)
Client Characteristic Configuration (0x2902)
I	14:49:35.205	Connection parameters updated (interval: 16.25ms, latency: 0, timeout: 4000ms)
I	14:49:35.745	Connection parameters updated (interval: 16.25ms, latency: 0, timeout: 4000ms)
V	14:49:41.628	[DFU] DFU service started
V	14:49:41.628	[DFU] Opening file...
I	14:49:41.655	[DFU] Firmware file opened successfully
V	14:49:41.655	[DFU] Connecting to DFU target...
D	14:49:41.656	[DFU] gatt = device.connectGatt(autoConnect = false)
I	14:49:41.666	[DFU] Connected to E6:73:8D:C5:CC:49
V	14:49:41.666	[DFU] Discovering services...
D	14:49:41.666	[DFU] gatt.discoverServices()
I	14:49:41.669	[DFU] Services discovered
D	14:49:41.686	[DFU] wait(1000)
V	14:49:42.725	[DFU] Requesting new MTU...
D	14:49:42.725	[DFU] gatt.requestMtu(517)
I	14:49:42.725	[DFU] MTU changed to: 23
V	14:49:42.725	[DFU] Enabling notifications for 8ec90001-f315-4f60-9fb8-838830daea50
D	14:49:42.725	[DFU] gatt.setCharacteristicNotification(8ec90001-f315-4f60-9fb8-838830daea50, true)
D	14:49:42.725	[DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I	14:49:42.732	[DFU] Data written to descr.8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-00
V	14:49:42.732	[DFU] Notifications enabled for 8ec90001-f315-4f60-9fb8-838830daea50
A	14:49:42.732	[DFU] Notifications enabled
D	14:49:42.732	[DFU] wait(1000)
V	14:49:43.751	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	14:49:43.751	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	14:49:43.771	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 06-01
I	14:49:43.847	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-06-01-00-01-00-00-8D-00-00-00-97-CB-47-90
A	14:49:43.847	[DFU] Command object info received (Max size = 256, Offset = 141, CRC = 9047CB97)
A	14:49:43.847	[DFU] Received CRC match Init packet
V	14:49:43.847	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	14:49:43.847	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	14:49:43.847	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 04
I	14:49:43.847	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-0A
E	14:49:43.847	[DFU] Remote DFU error: OPERATION FAILED
V	14:49:43.847	[DFU] Disconnecting...
D	14:49:43.849	[DFU] gatt.disconnect()
I	14:49:43.865	[DFU] Disconnected
D	14:49:43.865	[DFU] gatt.refresh() (hidden)
D	14:49:43.866	[DFU] gatt.close()
D	14:49:43.877	[DFU] wait(600)
I	14:49:43.987	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
I	14:49:44.177	Connection parameters updated (interval: 16.25ms, latency: 0, timeout: 4000ms)
I	14:49:44.598	Connection parameters updated (interval: 16.25ms, latency: 0, timeout: 4000ms)
```

philips77, 2019-10-29:
Which SDK are you using? 12?

HansKrywaa, 2019-10-29:
Which SDK do you mean?

philips77, 2019-10-29:
I meant from which Nordic SDK the DFU bootloader is. For me it looks like Secure DFU from SDK 12.x.

philips77, 2019-10-30:
It's on the device. The firmware must have been forked from some version of the SDK.

HansKrywaa, 2019-11-04:
Closed as it was because of the device. Thanks for help!!

closed 2019-11-04 by HansKrywaa