nordicsemi/Android-DFU-Library #206: "(error 5): INVALID OBJECT" after executing init packet

eliotstock, 2019-11-07:
Hi Aleksander. I have DFU working end to end when using nRF Connect for Android to perform the DFU, but when performing it from my own app, it fails. Here's the logcat.

```
2019-11-07 22:11:58.538 22340-22340/cc.biketracker.android D/DfuActivity: onCreate() 20149302
2019-11-07 22:11:58.581 22340-22340/cc.biketracker.android D/DfuActivity: onResume() 20149302
2019-11-07 22:11:58.589 22340-22389/cc.biketracker.android D/FA: Logging event (FE): screen_view(_vs), Bundle[{firebase_event_origin(_o)=auto, firebase_previous_class(_pc)=SettingsActivity, firebase_previous_id(_pi)=-7184870573564434593, firebase_screen_class(_sc)=DfuActivity, firebase_screen_id(_si)=-7184870573564434592}]
2019-11-07 22:11:58.617 22340-22340/cc.biketracker.android D/DfuActivity: Network state: CONNECTED
2019-11-07 22:11:58.617 22340-22340/cc.biketracker.android D/DfuActivity: Online
2019-11-07 22:11:58.623 22340-22340/cc.biketracker.android I/DfuBaseService: DFU service created. Version: 1.9.0
2019-11-07 22:11:58.637 630-652/? W/SurfaceFlinger: Attempting to set client state on removed layer: cc.biketracker.android/cc.biketracker.android.settings.SettingsActivity#1
2019-11-07 22:11:58.637 630-652/? W/SurfaceFlinger: Attempting to destroy on removed layer: cc.biketracker.android/cc.biketracker.android.settings.SettingsActivity#1
2019-11-07 22:11:58.699 22340-22515/cc.biketracker.android I/DfuBaseService: Connecting to the device...
2019-11-07 22:11:58.700 22340-22515/cc.biketracker.android D/BluetoothGatt: connect() - device: C7:56:60:F0:56:58, auto: false
2019-11-07 22:11:58.700 22340-22515/cc.biketracker.android D/BluetoothGatt: registerApp()
2019-11-07 22:11:58.700 22340-22515/cc.biketracker.android D/BluetoothGatt: registerApp() - UUID=797d7195-c707-4a5c-929d-b458dea9b946
2019-11-07 22:11:58.702 22340-22475/cc.biketracker.android D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
2019-11-07 22:11:58.703 22340-22475/cc.biketracker.android D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=C7:56:60:F0:56:58
2019-11-07 22:11:58.703 22340-22475/cc.biketracker.android I/DfuBaseService: Connected to GATT server
2019-11-07 22:11:58.703 22340-22475/cc.biketracker.android I/DfuBaseService: Waiting 1600 ms for a possible Service Changed indication...
2019-11-07 22:11:58.827 22340-22340/cc.biketracker.android D/BluetoothGatt: readRssi() - device: C7:56:60:F0:56:58
2019-11-07 22:11:59.834 22340-22340/cc.biketracker.android D/BluetoothGatt: readRssi() - device: C7:56:60:F0:56:58
2019-11-07 22:12:00.305 22340-22475/cc.biketracker.android D/BluetoothGatt: discoverServices() - device: C7:56:60:F0:56:58
2019-11-07 22:12:00.309 22340-22475/cc.biketracker.android I/DfuBaseService: Attempting to start service discovery... succeed
2019-11-07 22:12:00.320 22340-22475/cc.biketracker.android D/BluetoothGatt: onSearchComplete() = Device=C7:56:60:F0:56:58 Status=0
2019-11-07 22:12:00.321 22340-22475/cc.biketracker.android I/DfuBaseService: Services discovered
2019-11-07 22:12:00.329 22340-22515/cc.biketracker.android I/DfuImpl: Reading Service Changed CCCD value...
2019-11-07 22:12:00.407 22340-22515/cc.biketracker.android I/DfuImpl: Enabling indications...
2019-11-07 22:12:00.410 22340-22515/cc.biketracker.android D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
2019-11-07 22:12:00.496 22340-22515/cc.biketracker.android I/DfuImpl: Buttonless service with bond sharing found -> SDK 14 or newer
2019-11-07 22:12:00.840 22340-22340/cc.biketracker.android D/BluetoothGatt: readRssi() - device: C7:56:60:F0:56:58
2019-11-07 22:12:01.517 22340-22515/cc.biketracker.android I/DfuImpl: Enabling indications...
2019-11-07 22:12:01.519 22340-22515/cc.biketracker.android D/BluetoothGatt: setCharacteristicNotification() - uuid: 8ec90004-f315-4f60-9fb8-838830daea50 enable: true
2019-11-07 22:12:01.848 22340-22340/cc.biketracker.android D/BluetoothGatt: readRssi() - device: C7:56:60:F0:56:58
2019-11-07 22:12:02.644 22340-22515/cc.biketracker.android I/DfuImpl: Sending Enter Bootloader (Op Code = 1)
2019-11-07 22:12:02.790 22340-22515/cc.biketracker.android I/DfuImpl: Response received (Op Code = 1, Status = 1)
2019-11-07 22:12:02.856 22340-22340/cc.biketracker.android D/BluetoothGatt: readRssi() - device: C7:56:60:F0:56:58
2019-11-07 22:12:03.861 22340-22340/cc.biketracker.android D/BluetoothGatt: readRssi() - device: C7:56:60:F0:56:58
2019-11-07 22:12:04.870 22340-22340/cc.biketracker.android I/chatty: uid=10160(cc.biketracker.android) identical 1 line
2019-11-07 22:12:05.880 22340-22340/cc.biketracker.android D/BluetoothGatt: readRssi() - device: C7:56:60:F0:56:58
2019-11-07 22:12:06.895 22340-22340/cc.biketracker.android D/BluetoothGatt: readRssi() - device: C7:56:60:F0:56:58
2019-11-07 22:12:07.847 22340-22358/cc.biketracker.android D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=7 device=C7:56:60:F0:56:58
2019-11-07 22:12:07.847 22340-22475/cc.biketracker.android D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=8 device=C7:56:60:F0:56:58
2019-11-07 22:12:07.848 22340-22475/cc.biketracker.android W/DfuBaseService: Target device disconnected with status: 8
2019-11-07 22:12:07.850 22340-22515/cc.biketracker.android I/DfuBaseService: Cleaning up...
2019-11-07 22:12:07.851 22340-22515/cc.biketracker.android D/BluetoothGatt: close()
2019-11-07 22:12:07.851 22340-22515/cc.biketracker.android D/BluetoothGatt: unregisterApp() - mClientIf=8
2019-11-07 22:12:07.855 22340-22515/cc.biketracker.android I/DfuImpl: Restarting to bootloader mode
2019-11-07 22:12:07.861 22340-22340/cc.biketracker.android D/BluetoothGatt: close()
2019-11-07 22:12:07.862 22340-22340/cc.biketracker.android D/BluetoothGatt: unregisterApp() - mClientIf=7
2019-11-07 22:12:07.864 22340-22340/cc.biketracker.android D/BikeTrackerManager: onDeviceDisconnected()
2019-11-07 22:12:07.865 22340-22340/cc.biketracker.android D/BikeTrackerManager: onDeviceDisconnected()
2019-11-07 22:12:07.865 22340-22340/cc.biketracker.android D/BikeTrackerManager: Ignoring device disconnection while DFU is in progress.
2019-11-07 22:12:07.866 22340-22340/cc.biketracker.android I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2019-11-07 22:12:09.910 22340-22515/cc.biketracker.android I/DfuBaseService: Connecting to the device...
2019-11-07 22:12:09.912 22340-22515/cc.biketracker.android D/BluetoothGatt: connect() - device: C7:56:60:F0:56:58, auto: false
2019-11-07 22:12:09.912 22340-22515/cc.biketracker.android D/BluetoothGatt: registerApp()
2019-11-07 22:12:09.913 22340-22515/cc.biketracker.android D/BluetoothGatt: registerApp() - UUID=256ff0d9-d031-451b-8313-0af9cc9803c5
2019-11-07 22:12:09.918 22340-22475/cc.biketracker.android D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2019-11-07 22:12:09.995 22340-22475/cc.biketracker.android D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=C7:56:60:F0:56:58
2019-11-07 22:12:09.996 22340-22475/cc.biketracker.android I/DfuBaseService: Connected to GATT server
2019-11-07 22:12:09.998 22340-22340/cc.biketracker.android I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_CONNECTED
2019-11-07 22:12:09.999 22340-22475/cc.biketracker.android I/DfuBaseService: Waiting 1600 ms for a possible Service Changed indication...
2019-11-07 22:12:11.603 22340-22475/cc.biketracker.android D/BluetoothGatt: discoverServices() - device: C7:56:60:F0:56:58
2019-11-07 22:12:11.606 22340-22475/cc.biketracker.android I/DfuBaseService: Attempting to start service discovery... succeed
2019-11-07 22:12:11.607 22340-22475/cc.biketracker.android D/BluetoothGatt: onPhyUpdate() - status=0 address=C7:56:60:F0:56:58 txPhy=2 rxPhy=2
2019-11-07 22:12:11.618 22340-22475/cc.biketracker.android W/BluetoothGatt: Unhandled exception in callback
    java.lang.NullPointerException: Attempt to invoke virtual method 'void no.nordicsemi.android.dfu.DfuBaseService.sendLogBroadcast(int, java.lang.String)' on a null object reference
        at no.nordicsemi.android.dfu.BaseDfuImpl$BaseBluetoothGattCallback.onPhyUpdate(BaseDfuImpl.java:207)
        at no.nordicsemi.android.dfu.DfuBaseService$5.onPhyUpdate(DfuBaseService.java:983)
        at android.bluetooth.BluetoothGatt$1$2.run(BluetoothGatt.java:209)
[... 35 lines cut ...]
2019-11-07 22:12:14.948 22340-22340/cc.biketracker.android I/DfuBaseService: DFU service destroyed

```

I'm using the same zip file in each case.

[... cut ...]
/Users/eliot/Library/Python/2.7/bin/nrfutil pkg generate --application _build/app.bin --application-version 0x02 --sd-req 0xB6 --hw-version 8 --key-file ../biketracker_dfu.pem _build/biketracker_dfu.zip
Zip created at _build/biketracker_dfu.zip
```

I'm using SDK 15.3.0 and my bootloader is the secure buttonless example one from the SDK. My soft device is s132 v6.1.1.

I can't currently get any logging from the bootloader, because it won't fit in flash when I build it with the logging in. I might be able to step through it if I have to, but it'll take some work on my gdb setup.

Any idea what might be going on? Should I post the zip file here?

philips77, 2019-11-13:
The reason for Invalid Object after executing Init packet may be Signature check fails or protobuf decoding fails. I managed to decode your init packet using **nrf util** and it seems fine. Are you sure you're sending the right fw to the right device, perhaps the signature is incorrect (bootloader might have been flashed with a different bootloader).
``` 
      |- op_code: INIT
      |- signature_type: ECDSA_P256_SHA256
      |- signature (little-endian): a6ee6cd91d3820f0a98ac0d9aeca81f887f6d98d8c82caf15a76fafbc563beb034923102b1fb991b0cbdc8269c0c97a6f2c9c385db2146a5cbc1033fd8552665
      |
      |- fw_version: 0x00000002 (2)
      |- hw_version 0x00000008 (8)
      |- sd_req: 0xB6
      |- type: APPLICATION
      |- sd_size: 0
      |- bl_size: 0
      |- app_size: 180012
      |
      |- hash_type: SHA256
      |- hash (little-endian): f91b1c4c75d798a30fb60ad3a8df27530bea2b30cd419120fb2ae30def54892b
      |
      |- boot_validation_type: ['VALIDATE_GENERATED_CRC']
      |- boot_validation_signature (little-endian): ['']
      |
      |- is_debug: False
```

eliotstock, 2019-11-21:
Hmm. I'm not that aware of all the checks the bootloader is doing on the firmware. I can't step through the bootloader code in gdb or SES and I can't spare enough flash space to build the bootloader with the RTT logging included. I'll have to find some more flash space somewhere and get back to you.

philips77, 2020-06-24:
Hello, I'm closing the issue now. Feel free to reopen it, or open a new one if you have another problem!

closed 2020-06-24 by philips77