nordicsemi/Android-DFU-Library #210: DFU mode connect not work

Hugo-Hyenatek, 2019-11-28:
Hi, some phones have DFU update problem, device stopped at connection moment.

Tested smartphone:
1. Huawei Y7 Prime 2018, Android OS 8.0.0
2. ASUS Z00UD, Android OS 6.0.1

DfuBaseService logs:
I/DfuBaseService: DFU service created. Version: 1.9.0
W/DfuBaseService: Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application. Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
I/DfuBaseService: Connecting to the device...

wait few minutes nothing happened!

philips77, 2019-11-28:
Make sure the devices are advertising and are connectable. Try DFU them using nRF Connect on those phones and send us the log from this app.

Hugo-Hyenatek, 2020-01-21:
Sorry, this time to reply.

Every time open the nRF Connect app for the first time, still stuck during connection.
**log only:**[Server] server started

After disconnection, then it can be used normally work after the second connection.

closed 2020-07-03 by philips77