nordicsemi/Android-DFU-Library #212: DFU CHARACTERISTICS NOT FOUND

vamshik113, 2019-12-18:
I'm trying to do DFU using ToolBox app and it is getting uploaded successfully. But in my app build I'm getting error DFU CHARACTERISTICS NOT FOUND. 

I have given below Configs.
final DfuServiceInitiator starter = new DfuServiceInitiator(device.getAddress())
                .setDeviceName(device.getName())
                .setKeepBond(false)
                .setForceDfu(false)
                .setPacketsReceiptNotificationsEnabled(true)
                .setPacketsReceiptNotificationsValue(5)
                .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)
                .setDisableNotification(true);

Android Version is 9.0

philips77, 2019-12-18:
If you're connecting to the same device it may be the cache. Before connecting in your app, connect with nRF Connect for Android and choose "Refresh services" from the top-right menu.

philips77, 2019-12-18:
Cache is the mechanism that phones and other devices do to avoid performing service discovery on each connection. On second and next connections the services are just taken from the phone's memory. However, it may happen that the services have changed on your device, and the phone is not aware of it. In DFU, the app and bootloader use different set of services. For non-bonded devices we use a different MAC address, so even if the devices are cached, it is not a problem, because from the phone's perspective app and bootloader are 2 different devices. on bonded devices the Service Changed indication is used to inform the client that services have changed. But when you develop or test multiple firmwares on one device and connect to it from a phone, the returned services may be of the previous sample.
nRF Connect allows to force-clear the cache (Android only).

vamshik113, 2019-12-19:
@philips77 Thanks for your reply. I have found the issue was with DFU library version used. I updated to latest version and DFU is working

closed 2019-12-19 by vamshik113

HerrNiklasRaab, 2020-01-09:
Hello @philips77, 

Same problem here, using the flutter plugin. Clearing Cache is not solving the problem. Updating firmware through the nRF Toolbox app is working properly.
@vamshik113 Which version did you use before? Which version are you using now?

HerrNiklasRaab, 2020-01-09:
This Experimental Buttonless DFU Service is detectable in the nRF Connect app: 8e400001-f315-4f60-9fb8-838830daea50

HerrNiklasRaab, 2020-01-09:
Fixed by setting because there was the Experimental Buttonless DFU Service on the device: 
`starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);`