nordicsemi/Android-DFU-Library #213: Perform simultaneous DFU to multiple devices

JongChern, 2020-01-23:
Is this library able to handle simultaneous DFU to multiple devices?

philips77, 2020-01-23:
Only one Intent Service with given name can be started. The following calls will be queued, as far as I know. But you could have multiple services, each extension `DfuBaseService` and this could work.
In other words: we didn't try this. Give it a shot and in case of a problem, check the hint above.

closed 2020-01-24 by JongChern

davidgyoung, 2021-02-11:
I have confirmed it is possible to simultaneously perform DFU on multiple devices (I tested four at a time successfully) by:

1. Make multiple identical copies of your Service class which slightly different names.  My four were identical to the sample service shown in this repo's instructions and named:

* DfuService
* DfuService2
* DfuService3
* DfuService4

2. Register all four of these in your AndroidManifest.xml as so:

```        
        <service
            android:name=".ble.DfuService">
        </service>
        <service
            android:name=".ble.DfuService2">
        </service>
        <service
            android:name=".ble.DfuService3">
        </service>
        <service
            android:name=".ble.DfuService4">
        </service>
```

3. In the code that starts your DFU, you need to keep track of which of the above four classes are currently in use, and always pick one that is not in use to start the next DFU.  I did something like this (Kotlin):

```
val dfuServiceClasses = arrayOf(DfuService::class.java, DfuService2::class.java,DfuService3::class.java,DfuService4::class.java)
val unusedServiceClass = getUnusedServiceClass(); // custom code to pick one of the above that is not in use
if (unusedServiceClass == null) {
  Log.e(TAG, "Cannot perform DFU -- all service classes in use")
  return
}
val starter = DfuServiceInitiator(device.getAddress())
            .setDeviceName(device.getName())
            .setKeepBond(false)
starter.setZip(null, dfuAttempt.localPathString)
starter.start(context,unusedServiceClass)
markServiceClassUsed(unusedServiceClass) // custom code to mark the service class as in use.  Clear this on completion.
```